UpstreamData/pyasic #8: Add Unit Tests, Comments, and Documentation/Docstrings

UpstreamData, 2022-05-19:
Most files need unit tests, comments, and documentation/typing added to them.

Current unit tests are located in `.tests`.

- Docstrings in `API` are mostly done, these are all base level implementations of the miner APIs, although some standardization across files and possibly improved type hinting would be useful.
- Unit tests for `network` have been written, but could always be better.
- Generally all files need some level of comments added, in my opinion it is a good idea to be as descriptive as possible, as an end goal is for even novice programmers to be able to design a simple program with this repo.

### TODO:
- #### Add unit tests:
    - [ ]  `API`  - This may be tough, these rely on external connections heavily.
    - [ ]  `config`
    - [ ]  `miners`  - Same issue as `API`
    - [x] `network`
    - [ ] `settings` - Possibly, although this probably wont be super hard/useful
- #### Add docstrings/documentation:
    - [x] `API`  - Maybe slight improvements to standardization.
    - [ ]  `config`
    - [ ]  `miners`
    - [ ] `network`
- #### Additional Comments:
    - Ongoing...

jaysharma0909, 2024-04-07:
@UpstreamData Can I contribute to this issue?

b-rowan, 2024-04-07:
Yeah, go for it.  There is still a lot of tests missing, and documentation is good, but could still use some more information.  Happy to provide specific is you need them, but this checklist here isn't really accurate anymore.

pavankalyan224847, 2024-04-09:
hello is this issue up for grabs can i help with this issue ?

b-rowan, 2024-04-09:
> hello is this issue up for grabs can i help with this issue ? 

Sure, this is just sort of an ongoing improvement issue.

pavankalyan224847, 2024-04-09:
can you please update checklist so that it'll be helpfull to look into what all things should be implemented and we can start working on this issue

PR #121 (2024-04-19): docs: Add documentation for `config`
Partial fix #8 
Fixes #124
Add documentation for `config`
PR #128 (2024-04-22): Add tests in rpc_tests
Partial fixes #8 
Added tests in rpc_tests for TestGCMinerRPCAPI
PR #131 (2024-04-27): Add tests in config_tests
Partial Fixes #8 
Added tests in config_tests
closed 2024-04-19 by UpstreamData