UpstreamData/pyasic #22: BTMiner `self.get_data()` not including errors in data (M30)

UpstreamData, 2022-11-01:
### Discussed in https://github.com/UpstreamData/pyasic/discussions/21

<div type='discussions-op-text'>

<sup>Originally posted by **JustinHowes01** November  1, 2022</sup>
I'm trying to write a script that will collect miner data over time on Whatsminer M30s machines, but when I try to print the error codes, they don't show up. I've tested this with miners that I have confirmed have errors such as 300, 530, 410 (missing a hashboard) and so on, and they still don't show. I've checked the Whatsminer.py file in the library and it includes all the errors I've tested.

Here is my code:

```
import pyasic
import asyncio
import pyperclip
import time

async def gather_miner_data():
    miner = await pyasic.MinerFactory().get_miner(miner_ip)
    miner_data = await miner.get_data()

    return miner_data.hashrate, miner_data.wattage, miner_data.errors, miner_data.env_temp

miner_ip = pyperclip.paste()

while True:
    hashrate, wattage, errors, temp = asyncio.run(gather_miner_data())

    print('Hashrate: %s\nWattage: %s\nErrors: %s\nTemperature: %s\n' % (hashrate, wattage, errors, temp))
    time.sleep(1)
```

The script uses the IP address copied to the clipboard to scan a miner, since we have very many machines running.

After running the script, it will give continous readings from the miner that look like this:

```
Hashrate: 72.79
Wattage: 3211
Errors: []
Temperature: 58.5
```

This miner in particular has a 275 power overheat error, and yet the error is not reported here.</div>

UpstreamData, 2022-11-01:
Moving this to an issue.  I just pushed up a new version, v0.18.6, can you try that? (`pip install pyasic=0.18.61)

UpstreamData, 2022-11-01:
Tested with M31S+ running FW version 2021-09-07, everything seems to be working with the above update.  I'm going to hop back a few versions and see if anything changes.

When you get a chance, need the info mentioned in the discussion - 
Miner(s) firmware version,
Pyasic off github or pip,
and the results of `await miner.api.summary()`.

UpstreamData, 2022-11-01:
@crc32 Makes me wonder if its getting reported in a different place on these miners? [Latest API docs for reference](https://aws-microbt-com-bucket.s3.us-west-2.amazonaws.com/WhatsminerAPI%20V2.0.4.pdf)

UpstreamData, 2022-11-01:
Looks like it might be the `get_error_code`  command?  I'll update some miners and give it a test run.

crc32, 2022-11-01:
That seems to be correct.

'echo '{"cmd":"get_error_code"}' | nc 10.10.xx.xx 4028'

returns:

> {"STATUS":"S","When":1667327194,"Code":131,"Msg":{"error_code":["302":"2022-10-31 03:42:26","532":"1970-01-02 08:00:02","412":"1970-01-02 08:00:01"]},"Description":""}

These are the expected errors.

That's a huge change from V2.0.2 to V2.0.4 of their API...

UpstreamData, 2022-11-01:
Fixed with v0.19.0.  Cause was a breaking change from the whatsminer API in v2.0.4 that moved error codes from the `summary` command to the `get_error_code` command.

closed 2022-11-01 by UpstreamData

JustinHowes01, 2022-11-02:
'''
{'STATUS': [{'STATUS': 'S', 'Msg': 'Summary'}],
 'SUMMARY': [{'Elapsed': 42709,
              'MHS av': 60420128.78,
              'MHS 5s': 69202822.25,
              'MHS 1m': 61117321.6,
              'MHS 5m': 60375400.49,
              'MHS 15m': 60397787.38,
              'HS RT': 60375400.49,
              'Accepted': 2354,
              'Rejected': 8,
              'Total MH': 2580490525209.0,
              'Temperature': 55.69,
              'freq_avg': 1041,
              'Fan Speed In': 0,
              'Fan Speed Out': 0,
              'Power': 2631,
              'Power Rate': 43.58,
              'Pool Rejected%': 0.3443,
              'Pool Stale%': 0.0,
              'Uptime': 449363,
              'Security Mode': 0,
              'Hash Stable': True,
              'Hash Stable Cost Seconds': 255,
              'Hash Deviation%': 0.0367,
              'Target Freq': 861,
              'Target MHS': 50771448,
              'Env Temp': 54.44,
              'Power Mode': 'Normal',
              'Factory GHS': 55525,
              'Power Limit': 2635,
              'Chip Temp Min': 83.25,
              'Chip Temp Max': 101.25,
              'Chip Temp Avg': 92.27,
              'Debug': '',
              'Btminer Fast Boot': 'disable'}],
 'id': 1}

Process finished with exit code 0
'''
This is the result from the command 'miner.api.summary()'. The miner is running firmware version 2022.09.09, and I'm using Pycharm, so the version of Pyasic that it downloaded is a clone from Github I believe.

UpstreamData, 2022-11-02:
Awesome.  The recent versions should be fixed with the patches made earlier, so if you pull the latest version you should be good to go.

I also just received a brand new firmware update direct from the Whatsminer support team, as they had some bad JSON returned in the new command for error handling.  It appears they haven't put it up on their website yet, so I have uploaded it to google drive if you would like to update with it - https://drive.google.com/drive/folders/1-KMW_ZWrgtWPt0Y1f5kvRzTG8nosqBjZ

JustinHowes01, 2022-11-02:
{'STATUS': [{'STATUS': 'S', 'Msg': 'Summary'}],
 'SUMMARY': [{'Elapsed': 42709,
              'MHS av': 60420128.78,
              'MHS 5s': 69202822.25,
              'MHS 1m': 61117321.6,
              'MHS 5m': 60375400.49,
              'MHS 15m': 60397787.38,
              'HS RT': 60375400.49,
              'Accepted': 2354,
              'Rejected': 8,
              'Total MH': 2580490525209.0,
              'Temperature': 55.69,
              'freq_avg': 1041,
              'Fan Speed In': 0,
              'Fan Speed Out': 0,
              'Power': 2631,
              'Power Rate': 43.58,
              'Pool Rejected%': 0.3443,
              'Pool Stale%': 0.0,
              'Uptime': 449363,
              'Security Mode': 0,
              'Hash Stable': True,
              'Hash Stable Cost Seconds': 255,
              'Hash Deviation%': 0.0367,
              'Target Freq': 861,
              'Target MHS': 50771448,
              'Env Temp': 54.44,
              'Power Mode': 'Normal',
              'Factory GHS': 55525,
              'Power Limit': 2635,
              'Chip Temp Min': 83.25,
              'Chip Temp Max': 101.25,
              'Chip Temp Avg': 92.27,
              'Debug': '',
              'Btminer Fast Boot': 'disable'}],
 'id': 1}

Process finished with exit code 0

This is the result from the command 'miner.api.summary()'. The miner is running firmware is a custom version, and I'm using Pycharm, so the version of Pyasic that it downloaded is a clone from Github I believe.

{'STATUS': [{'STATUS': 'S',
             'When': 1667356702,
             'Code': 11,
             'Msg': 'Summary',
             'Description': 'btminer'}],
 'SUMMARY': [{'Elapsed': 60745,
              'MHS av': 83176419.43,
              'MHS 5s': 101787926.66,
              'MHS 1m': 84789670.3,
              'MHS 5m': 83508184.53,
              'MHS 15m': 83280385.3,
              'HS RT': 83200331.15,
              'Found Blocks': 0,
              'Getworks': 2069,
              'Accepted': 4532,
              'Rejected': 12,
              'Hardware Errors': 0,
              'Utility': 4.48,
              'Discarded': 0,
              'Stale': 0,
              'Get Failures': 0,
              'Local Work': 954095201,
              'Remote Failures': 0,
              'Network Blocks': 110,
              'Total MH': 5052552825562.0,
              'Work Utility': 0.0,
              'Difficulty Accepted': 1183776768.0,
              'Difficulty Rejected': 3014656.0,
              'Difficulty Stale': 0.0,
              'Best Share': 20917622580,
              'Temperature': 50.5,
              'freq_avg': 950,
              'Fan Speed In': 0,
              'Fan Speed Out': 0,
              'Voltage': 1372,
              'Power': 3584,
              'Power_RT': 3596,
              'Power Rate': 43.08,
              'Device Hardware%': 0.0,
              'Device Rejected%': 0.0,
              'Pool Rejected%': 0.254,
              'Pool Stale%': 0.0,
              'Last getwork': 0,
              'Uptime': 61363,
              'Power Current': 246.193736,
              'Power Fanspeed': 0,
              'Error Code 0': 5030,
              'Error 0 Time': '',
              'Error Code 1': 5031,
              'Error 1 Time': '',
              'Error Code 2': 5032,
              'Error 2 Time': '',
              'Error Code Count': 3,
              'Factory Error Code Count': 0,
              'Security Mode': 0,
              'Liquid Cooling': True,
              'Hash Stable': True,
              'Hash Stable Cost Seconds': 1818,
              'Hash Deviation%': 0.0296,
              'Target Freq': 861,
              'Target MHS': 76157172,
              'Env Temp': 47.5,
              'Power Mode': 'Normal',
              'Firmware Version': "'20211101.22.REL'",
              'MAC': '00:00:00:00:00:00',
              'Factory GHS': 81920,
              'Power Limit': 3600,
              'Chip Temp Min': 79.25,
              'Chip Temp Max': 105.0,
              'Chip Temp Avg': 91.85,
              'Debug': '00.0_100.0_672',
              'Btminer Fast Boot': 'disable'}],
 'id': 1}

Process finished with exit code 0

This is a result from a similar miner but with an older version of firmware, 20211101.22.REL.

UpstreamData, 2022-11-02:
I was testing with the 2021 versions originally, but just happened to have some in the shop I could update and test with, it seems in some recent version they changed the way the API works with a breaking change (see above).  Either way it's fixed now.

UpstreamData, 2022-11-02:
I assume if you're running custom you're using chipless, I haven't tried the API with it yet so idk, but it uses 99% stock components so I assume it just uses the stock API (albeit from a new-ish version).

commit 82f6d2f2 (2022-11-01): version: bump version number
resolves #22