UpstreamData/pyasic #28: Thanks for this project guys. It is awesome.

Awote, 2023-01-11:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Awote, 2023-01-13:
thanks for help. i take mac
I have a question, will pyasic be able to detect the miner if the miner is connected via a switch -> router and pyasic directly to the router? I know the IP address of the miner.

UpstreamData, 2023-01-13:
Yep, as long as the miner and the computer you are using pyasic on are connected to the same network, this will work perfectly fine.  Just run the above script with 
```python
asyncio.run(get_miner_mac(IP_ADDR_OF_MINER))
```

UpstreamData, 2023-01-13:
For reference, if you need any more data from your miner, you can use the above script, but instead of `return data.mac` you can get any piece of data that is shown [here](https://pyasic.readthedocs.io/en/latest/data/miner_data/) by using the same format of `data.attribute`.

UpstreamData, 2023-01-22:
Closing this for now, seems like it might have been resolved.

@Awote if you have any more questions feel free to re-open this or to create a new discussion in the "Discussions" tab.

closed 2023-01-22 by UpstreamData