UpstreamData/pyasic #29: different time on scan_network_for_miners() and get_miner()

Awote, 2023-02-03:
Hi, i tested speed of functions scan_network_for_miners() and get_miner().  Why scan_network_for_miners() took 4-5 seconds on action (it didn't matter if i put IP or IP_RANGE) and get_miner() took  less then 0.5 seconds on same action?

UpstreamData, 2023-02-03:
`scan_network_for_miners()` will nearly never just scan 1 address.  When using it with just 1 IP, it assumes that is a network on that subnet with a subnet mask or /24, meaning if you enter 192.168.1.x, it will search 192.168.1.[1-254].  This is intended as a simplification feature for people less experienced with the way networking works, they may be able to find the IP address of their own computer, but maybe won't understand how their network is structured, so pyasic does the simplest and most common thing for them. 

It also attempts a connection beforehand to check if the miner is located at that address, so it opens then immediately closes connections on port 4028, meaning that it will always be slower than `get_miner()`, as that function is called as part of it but only after the initial connection is accepted.

UpstreamData, 2023-02-03:
I can't be sure what your goal is, but a good rule of thumb is that `get_miner` will always be quicker as scanning is doing more actions then calling `get_miner`.

If you're writing a program that needs to check beforehand where each miner is in the network, but still needs to be super fast, you can do 2 things:

- `MinerNetwork` exposes an async generator with the `scan_network_generator` method, you can use that with an async for loop to get miners as soon as the program knows they exist, and then use them immediately.

- `get_miner` should not error, only return `UnknownMiner`, so you can just call it on every address you want to check (you can actually use a custom method and a `MinerNetwork`, just call `list(MinerNetwork().get_network().hosts())` and loop through that), then make a check against `if not isinstance(miner, UnknownMiner):`.

Awote, 2023-02-03:
[image]
i found that timeout for scan_network_for_miner() is 3 seconds, maybe this timeout make scan_network_for_miner() too slow if he ping not a miner and wait 3 second ?

Awote, 2023-02-03:
is it bad idea if i decrease this timeout (1 not 3 seconds)?

UpstreamData, 2023-02-03:
Nope, decreasing this timeout is fine, that's why its there.  Come to think of it I have to come up with a better way to handle that, but for now, you can update those settings at the start of your program by doing:
```python3
from pyasic.settings import PyasicSettings

PyasicSettings().ping_timeout = 1
PyasicSettings().network_ping_retries = 1
```

These default values were there for robustness in the scan, so its possible depending on your miners that the scan could now miss the miners, but for sure tweak and find what works best.  I find Whatsminer models are extremely responsive to network requests, so if you have them they will likely not need a large timeout at all.

Awote, 2023-02-09:
Thanks for help. 
I how another problem, i can't get my miner's  `wattage` value when i use 'get_scan'  or 'MinerNetwork'. I allways get '-1' . I tested it on one miner and on many miners.
My code for one miner. 
'async def getMinerData(ip:str):
    print("test")
    h = time()
    miner = await pyasic.get_miner(ip)
    task = miner.get_data()
    # print(task)
    data = await asyncio.gather(task)
    print(time() - h)
    print(data)
    print("MINER ONE")
    return data.as_json()'
What i get :
{'ip': '10.1.50.50', 'datetime': '1675537482', 'mac': '00:00:00:00:00:00', 'model': 'S19j Pro', 'make': 'AntMiner', 'api_ver': '3.7', 'fw_ver': '4.11.1', 'hostname': 'Unknown', 'hashrate': 136.73, 'nominal_hashrate': 136.0, 'hashboards': [{'slot': 0, 'hashrate': 46.62, 'temp': 47, 'chip_temp': 32, 'chips': 126, 'expected_chips': 126, 'missing': False}, {'slot': 1, 'hashrate': 44.88, 'temp': 47, 'chip_temp': 32, 'chips': 126, 'expected_chips': 126, 'missing': False}, {'slot': 2, 'hashrate': 45.22, 'temp': 46, 'chip_temp': 31, 'chips': 126, 'expected_chips': 126, 'missing': False}], 'ideal_hashboards': 3, 'left_board_hashrate': 46.62, 'center_board_hashrate': 44.88, 'right_board_hashrate': 45.22, 'temperature_avg': 47, 'env_temp': -1.0, 'left_board_temp': 47, 'left_board_chip_temp': 32, 'center_board_temp': 47, 'center_board_chip_temp': 32, 'right_board_temp': 46, 'right_board_chip_temp': 31, 'wattage': -1, 'wattage_limit': -1, 'fan_1': 0, 'fan_2': 0, 'fan_3': 0, 'fan_4': 0, 'fan_psu': -1, 'left_chips': 126, 'center_chips': 126, 'right_chips': 126, 'total_chips': 378, 'ideal_chips': 378, 'percent_ideal': 100, 'nominal': True, 'pool_split': '0', 'pool_1_url': 'btc.foundryusapool.com:3333', 'pool_1_user': 'soldierlakeimmersion.3ca3081ff75e', 'pool_2_url': 'btc.foundryusapool.com:25', 'pool_2_user': 'soldierlakeimmersion.3ca3081ff75e', 'errors': [], 'fault_light': False, 'efficiency': 0}

UpstreamData, 2023-02-09:
Sadly, Antminer models don't provide any wattage measurements, so I have no way of figuring out what the wattage is, and I don't have an algorithm to guess what it might be approximately.  Your best bet would be to install BraiinsOS on your miner if it supported, and they have an algorithm that provides a guess at that information.  The other option is you could use some sort of smart plug or PDU that would tell you the wattage directly at the wall, and communicate with that, which will give a much more accurate measurement.

Just curious, does wattage show up on the web interface of the miner?  If they added it for whatever reason in the newest version or something I could definitely add it, I just have yet to see any way to read that info.

Awote, 2023-02-13:
Hi , i check antminer's GUI and don't find any solution for take miners wattage (((

closed 2023-02-13 by Awote