UpstreamData/pyasic #30: Will you add command assign static ip on miner?

Awote, 2023-02-13:
I can't find any information about how assign static ip on miner. what command should i use to do this?

Awote, 2023-02-13:
Or if it already existe in pyasic could you write a guide how can i do this ?

UpstreamData, 2023-02-13:
There is currently no way to do this via pyasic, it could very well be added, for now I assume you want to use it on X19 models?

Personal opinion on static IP addressing for miners?  I don't see a point, I find it would be much simpler (and could mess up less on the networking side) to just leave the IP set to DHCP and track locations of miners by MAC address, that way it is also consistent across networks (EG pulling a miner from site for repair then having issues identifying it or not having it show up at all on the repair network).

If you really feel that static IPs are a necessity, I will add it, but I strongly advise against it from personal experience.  Let me know.

crc32, 2023-02-13:
Not sure if Pyasic is the proper place to add static IPs? That's more something that should be handled by one's router. Specifically, always have the miner take in a DHCP, but if you want a static, have the router issue a static IP linked to the miners MAC. That way, if you move the miner, or otherwise change your network configuration, you don't lose quick access to the miner.

Of course, if it's a settable element, maybe pyasic should be able to set it (even if it's not the best idea)?

UpstreamData, 2023-02-13:
I think there are cases for both @crc32, in my case with a relatively closed off router in a mass of mines, I would much rather set them on the miner side.  That being said, it is always preferable to set that stuff on the network side locally, not publicly on the miner for a multitude of reasons, as long as its possible.

That being said, I agree, pyasic should support it strictly on principle, therefore, added in v0.27.5.

UpstreamData, 2023-02-13:
As of now, this is only supported for X19 models, if you need it for more miners please re-open, but I will look into the setting for other types of miners for now.

The function is accessible at `X19SubclassMiner.set_static_ip(ip: str, dns: str, gateway: str, subnet_mask: str = "255.255.255.0", hostname: str = None)` and can be unset with `X19SubclassMiner.set_dhcp(hostname: str = None)`.

UpstreamData, 2023-02-13:
Added for whatsminers in v0.27.6.

Awote, 2023-02-14:
It is good reason use MAC for miner's unification value but sometimes i didn't get MAC value for miner, that why i start this issue
'''
{
"ip": "10.1.50.50", "datetime": "1675537482", "mac": "00:00:00:00:00:00", "model": "S19j Pro", "make": "AntMiner", "api_ver": "3.7", "fw_ver": "4.11.1", "hostname": "Unknown", "hashrate": 136.73, "nominal_hashrate": 136.0, "hashboards": [{"slot": 0, "hashrate": 46.62, "temp": 47, "chip_temp": 32, "chips": 126, "expected_chips": 126, "missing": false}, {"slot": 1, "hashrate": 44.88, "temp": 47, "chip_temp": 32, "chips": 126, "expected_chips": 126, "missing": false}, {"slot": 2, "hashrate": 45.22, "temp": 46, "chip_temp": 31, "chips": 126, "expected_chips": 126, "missing": false}], "ideal_hashboards": 3, "left_board_hashrate": 46.62, "center_board_hashrate": 44.88, "right_board_hashrate": 45.22, "temperature_avg": 47, "env_temp": -1.0, "left_board_temp": 47, "left_board_chip_temp": 32, "center_board_temp": 47, "center_board_chip_temp": 32, "right_board_temp": 46, "right_board_chip_temp": 31, "wattage": -1, "wattage_limit": -1, "fan_1": 0, "fan_2": 0, "fan_3": 0, "fan_4": 0, "fan_psu": -1, "left_chips": 126, "center_chips": 126, "right_chips": 126, "total_chips": 378, "ideal_chips": 378, "percent_ideal": 100, "nominal": true, "pool_split": "0", "errors": [], "fault_light": false, "efficiency": 0},
'''

UpstreamData, 2023-02-14:
I'll try to create a fix for that this morning, there are 2 places you can get the MAC address on X19 models, I'll implement the second as I only found out about it recently.

Awote, 2023-02-14:
thanks for help !

Awote, 2023-02-14:
If it 's important i'll test ```pyasic``` on AntMiner S9 and it also didn't give ```MAC``` address

UpstreamData, 2023-02-14:
Are your miners by any chance running VNish firmware?

Awote, 2023-02-14:
if it is important they will be but not tomorrow

UpstreamData, 2023-02-14:
Added VNish MAC address support in v0.28.0.  Still have to run some tests on stock.

UpstreamData, 2023-02-14:
Added backup MAC address check for X19 stock in v0.28.1.

UpstreamData, 2023-02-14:
> if it is important they will be but not tomorrow

FYI only miners I've seen with the versioning scheme you sent above were running VNish, so not sure if that's just a recent update thing or what.

UpstreamData, 2023-02-14:
Stock S9 MAC fixed in v0.28.2.

closed 2023-02-14 by UpstreamData

Awote, 2023-02-14:
> Стандартный MAC-адрес S9 исправлен в версию 0.28.2.

thanks, i 'll test this on my miner tomorrow!

Awote, 2023-02-15:
thank now i take my s9 miner's MAC!