UpstreamData/pyasic #31: Question about pools in pyasic/data

Awote, 2023-02-18:
Hello, i have several questions. 
1) For what you read first and second pools?
2) Does they show that miner use both pools ?
3) Why i got ```pool_split: "0"``` ? i really don't understand what it should give me :((
4) If i got only one pool, does it mean that second pool don't work?
5) Will you add in ```pyasic\data``` third pool or you think that it's useless ?
I'm sorry i new in mining theme and don't know a lot .

UpstreamData, 2023-02-18:
Not sure what you mean by your first question, but I'll try to give a general overview of the pools part of `get_data`, and the command that supports it, `get_pools`.

The `get_pools` command gathers information on what I view to be the most important parts of the pool information on modern asics, that being 2 pools and 2 users, as well as a pool quota split if one exists.    `pool_1_url` and `pool_1_user` should always be the first pool on the miner.  `pool_2_url` and `pool_2_user` will usually be the second pool on the miner, unless you are running firmware such as BraiinsOS+ which supports hashrate splitting to multiple users, in which case pyasic will try to represent the seconds user in the pool split with the pool 2 data.  This means that `pool_split` is only used in cases where the hashrate is being split, and it represents a how many parts of the hashrate are going to each user, such as 1/1 which would be 50%/50%, or 4/1 which would be 80%/20%.  This means that `pool_split` being 0 just means there is no hashrate splitting on the device.

Regarding general pool related questions, the second (and third) pools are backup pools in case the connection to the first pool fails, but pyasic does no checks to see which pool is active.  The only goal of `get_pools` and `get_data` is to expose the most vital information to the user, while also being as minimal and unbloated as possible.

Regarding adding the 3rd pool to `get_data`, I don't currently have an plans to, I'm not sure it actually makes anything less confusing.  Sure most machines get the 3rd pool, but the edge cases where there are say 2 groups and 1/1 quota running 6 pools?  Then which pool do I pick as the 3rd one?  The problem is that pool data in `get_data` needs to extend theoretically forever, which I don't think is a good idea.

That all being said, if you really want to get all the pool data, pyasic definitely won't stop you, I would just use `get_config` as it does the exact infinite extension thing I was talking about above, and will gather data on every pool on the miner, or you can call `AnyMiner().api.pools()` and parse that out, as that is the source of the pool data when gathering data.

Awote, 2023-02-18:
Thanks i 'll test

Awote, 2023-02-18:
Could you tell me , how to remove miner's cache when ```pyasic``` find miner . I can't find func which do it. 
I need it because sometimes my miners turn off, but i see it after a while.

UpstreamData, 2023-02-18:
You should be able to 

```from pyasic.miners.miner_factory import MinerFactory

MinerFactory(). clear_cached_miners()```

closed 2023-02-25 by UpstreamData