UpstreamData/pyasic #32: Unable to run build

alwaysaugust, 2023-02-23:
**Describe the bug**
We are trying to install Pyasic but we are getting errors.

**To Reproduce**
Steps to reproduce the behavior:
tried poetry install --with dev
Installing dependencies from lock file
Get a log error from the Lock file.
Warning: poetry.lock is not consistent with pyproject.toml. You may be getting improper dependencies. Run `poetry lock [--no-update]` to fix it.
Because pyasic depends on isort (^5.10.1) which doesn't match any versions, version solving failed.

**Expected behavior**
Poetry should be installing the dependencies but it fails to read de pyproject.toml

**Screenshots**
This is the main bug. This should install dependencies and get it ready to build but is not downloading anything bcs of a mismatch
[image]

The screenshot shows we are trying to run poetry lock [--no-update]
[image]

**Desktop (please complete the following information):**
 - OS: Windows / WSL
 - Version Windows 11 / Ubuntu 22
 
**Additional context**
Do I need to run this only on ubuntu?

UpstreamData, 2023-02-23:
Looks like something is wrong with my dependencies.  This is probably due to the way I actually work on the project, its done via [pyasic-super](https://github.com/UpstreamData/pyasic-super) which consolidates all my projects related to pyasic into 1 project.  If you are looking to contribute to pyasic I highly suggest using that.  That being said, this is definitely on my end, ill look into it and get a fix out ASAP.  I assume you don't need it as part of a release, since you are trying to build it from source?

That aside, is there a reason not to just `pip install pyasic`?  That might be your simplest course of action.

UpstreamData, 2023-02-23:
Should be fixed as of 5ff0074.

UpstreamData, 2023-02-23:
Let me know how it goes, you will want to run `poetry install  --sync --with dev`,

alwaysaugust, 2023-03-01:
@UpstreamData Sorry I haven't had the chance to test it yet -- I'll report back once I do!

Thanks.

UpstreamData, 2023-03-01:
No problem, I'll re-open then until we're sure it works.  Had a few people just disappear after asking questions.

alwaysaugust, 2023-03-15:
This worked, thank you.

UpstreamData, 2023-03-15:
Sounds good.  Feel free to reach out if you have any more issues/questions.

closed 2023-03-15 by UpstreamData