UpstreamData/pyasic #35: Cannot removepool from antminer S9

Awote, 2023-03-08:
I try remove pool from my miner antminer S9 (BMMiner).
I get this error ```pyasic.errors.APIError: Access denied to 'removepool' command```.
My code:
```async def FindPool(ip: str):

    miner = await pyasic.get_miner(ip)
    try:
        t = await miner.api.removepool(n=0)
        miner_pools = await miner.api.pools()
    except:
        print(traceback.format_exc())
        miner_pools = []
    return miner_pools```

UpstreamData, 2023-03-08:
That's a miner side issue, technically with CGMiner's configuration.  You have to change the /etc/cgminer.conf file via ssh to be able to fix it, and I won't get into the specifics as you can see the CGMiner RPC API docs here - https://github.com/ckolivas/cgminer/blob/master/API-README

Generally, I wouldn't use the removepool API command.  It would be better to use `miner.get_config()` and `miner.send_config()`, and just replace the pools in that, as it will be more consistent across miners, but I'm not certain it's implemented for stock S9's yet.

Main reason it hasn't been implemented is because I haven't touched a stock S9 for years, as I see no reason to not just run some aftermarket firmware on them, and there are massive advantages to running custom firmware both at large and small scale.  I personally recommend BraiinsOS+, but that would be totally up to you.

That all being said I'm away from my computer for the next week or so, so I will make sure get and send config are implemented for stock S9s when I return.

UpstreamData, 2023-03-08:
I also can't remember if S9s actually use BMMiner or just modified CGMiner of some kind, but you can follow the same instructions for it, they are essentially the same thing on the API side, just replace CGMiner with BMMiner.

Awote, 2023-03-09:
I tried use ```miner.get_config()``` and then i override pool's info and then use ```miner.send_config()``` and got None. I research class which call func ```send_config``` and find soe interesting thing.
[image]

Awote, 2023-03-09:
How can i replace BMMiner to CGMiner ?

UpstreamData, 2023-03-09:
Just replace /etc/cgminer.conf with /etc/bmminer.conf.

Awote, 2023-03-15:
Hi.
I tried replace bmminer.conf to cgminer.cong but it didn't help me. I got same BMMiner Model in pyasic. I installed the VNISH Firmware, but got that Pyasic can't take VNISH on S9. :(

UpstreamData, 2023-03-15:
Try [BraiinsOS+](https://braiins.com/os/plus), support for it is much better right now since it's what I personally use and have the most experience with.  VNish is partially supported but mostly for X19 model.

Awote, 2023-03-15:
When i use ```pyasic.get_miner(ip:str)``` with my ip, then i write ```miner.api.get_config() , config has same structer for all miner's model ?  I mean that pyasic's config structer

UpstreamData, 2023-03-15:
Correct, it should.  There are miners that aren't supported by some functions though.  Also, `pyasic.config.MinerConfig` is the mining config of the miner, not the API config or other configs for the miner.

closed 2023-05-12 by UpstreamData