UpstreamData/pyasic #36: CGMiner: send_ssh_command hides exceptions raised by _get_ssh_connection

Resolude, 2023-05-12:
**Describe the bug**
pyasic.miners.backends.cgminer.CGMiner's send_ssh_command catches exceptions (asyncssh.Error, OSError) and returns None. This causes functions like stop_mining and resume mining, which lack an isinstance check, that call send_ssh_command to not have an exception raised to them. With the way stop_mining is written, the function will return True as long as no exception was raised, which makes it hard to detect SSH failures.

**To Reproduce**
Steps to reproduce the behavior:
1. Close or filter the SSH port on a cgminer
2. Try and call miner.stop_mining()

**Expected behavior**
stop_mining() should return false when there are SSH connection errors.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Miner Information (If applicable):**
 - Manufacturer: Bitmain
 - Type: Z15
 - Firmware Type: Stock
 - Firmware Version: CGMiner 4.9.0

UpstreamData, 2023-05-12:
Give it a test in v0.33.11?

Resolude, 2023-05-12:
Looks good now. Thank you. stop_mining and resume_mining return False as expected.

UpstreamData, 2023-05-12:
Awesome.  Thanks for the feedback, I'm relatively new to software development like this, so these things will likely happen quite a bit as I develop my software architecture abilities.  Let me know if you run into any more issues.

closed 2023-05-12 by UpstreamData