UpstreamData/pyasic #38: pyasic don't work on WIndows 7, python 3.8.10

Awote, 2023-05-21:
I install all dependents and wrote some code for test
```import asyncio
import pyasic

from pyasic.network import MinerNetwork

async def get_data(ip_range):
	h = time()
	print(h)
	net = MinerNetwork(ip_range)
	miners = await net.scan_network_for_miners()
	tasks = [miner.get_data() for miner in miners]
	data_task = await asynvio.gather(*tasks)
	print(data_task)

asyncio.run(get_data("192.168.10.10-192.168.10.250"))```
And i got this error

```(venv) C:\Users\Davis\Desktop\asyc_w7>python testW7.py
Traceback (most recent call last):
  File "testW7.py", line 3, in <module>
    import pyasic
  File "C:\Users\Davis\Desktop\asyc_w7\venv\lib\site-packages\pyasic\__init__.py
", line 30, in <module>
    from pyasic.miners import get_miner
  File "C:\Users\Davis\Desktop\asyc_w7\venv\lib\site-packages\pyasic\miners\__in
it__.py", line 21, in <module>
    from pyasic.miners.miner_factory import MinerFactory
  File "C:\Users\Davis\Desktop\asyc_w7\venv\lib\site-packages\pyasic\miners\mine
r_factory.py", line 576, in <module>
    class MinerFactory(metaclass=Singleton):
  File "C:\Users\Davis\Desktop\asyc_w7\venv\lib\site-packages\pyasic\miners\mine
r_factory.py", line 584, in MinerFactory
    ) -> AsyncIterable[AnyMiner]:
TypeError: 'ABCMeta' object is not subscriptable```
Could u tell me what i did wrong?

UpstreamData, 2023-05-21:
For some reason it seems AsyncIterable isn't being type hinted properly, I'll have to look into this, as it seems to be a python 3.8 issue.

UpstreamData, 2023-05-23:
Fixed in 0.33.15.  For some reason I had imported `AsyncIterable` from `collections.abc` instead of `typing`.