UpstreamData/pyasic #39: Braiins OS issue

mmitech, 2023-05-29:
After upgrading my ASIC's firmware I am unable to get the miner data

**This is the miner data from an Antminer with stock firmware:**

`
MinerData(ip='10.10.0.101', datetime=datetime.datetime(2023, 5, 29, 18, 15, 5, 343988, tzinfo=datetime.timezone(datetime.timedelta(seconds=7200), 'CEST')), mac='02:F0:33:33:DB:7C', model='S19', make='AntMiner', api_ver='3.1', fw_ver='Thu Jun 9 13:06:22 CST 2022', hostname='S1982T-01', hashrate=80.48, _hashrate=81.39, nominal_hashrate=82.0, hashboards=[HashBoard(slot=0, hashrate=26.54, temp=68, chip_temp=63, chips=88, expected_chips=76, missing=False), HashBoard(slot=1, hashrate=26.56, temp=69, chip_temp=64, chips=88, expected_chips=76, missing=False), HashBoard(slot=2, hashrate=27.38, temp=69, chip_temp=64, chips=88, expected_chips=76, missing=False)], ideal_hashboards=3, left_board_hashrate=26.54, center_board_hashrate=26.56, right_board_hashrate=27.38, temperature_avg=69, env_temp=-1.0, left_board_temp=68, left_board_chip_temp=63, center_board_temp=69, center_board_chip_temp=64, right_board_temp=69, right_board_chip_temp=64, wattage=-1, wattage_limit=-1, fans=[Fan(speed=5760), Fan(speed=5700), Fan(speed=6000), Fan(speed=5790)], fan_1=5760, fan_2=5700, fan_3=6000, fan_4=5790, fan_psu=-1, left_chips=88, center_chips=88, right_chips=88, total_chips=264, ideal_chips=228, percent_ideal_chips=116, percent_ideal_hashrate=98, percent_ideal_wattage=100, nominal=False, pool_split='0', pool_1_url='sha256asicboost.auto.nicehash.com:9200', pool_1_user='SOMETHING.S1982T-01', pool_2_url='', pool_2_user='', errors=[], fault_light=False, efficiency=0)
INFO:src.logger:MinerData(ip='10.10.0.101', datetime=datetime.datetime(2023, 5, 29, 18, 15, 5, 343988, tzinfo=datetime.timezone(datetime.timedelta(seconds=7200), 'CEST')), mac='02:F0:33:33:DB:7C', model='S19', make='AntMiner', api_ver='3.1', fw_ver='Thu Jun 9 13:06:22 CST 2022', hostname='S1982T-01', hashrate=80.48, _hashrate=81.39, nominal_hashrate=82.0, hashboards=[HashBoard(slot=0, hashrate=26.54, temp=68, chip_temp=63, chips=88, expected_chips=76, missing=False), HashBoard(slot=1, hashrate=26.56, temp=69, chip_temp=64, chips=88, expected_chips=76, missing=False), HashBoard(slot=2, hashrate=27.38, temp=69, chip_temp=64, chips=88, expected_chips=76, missing=False)], ideal_hashboards=3, left_board_hashrate=26.54, center_board_hashrate=26.56, right_board_hashrate=27.38, temperature_avg=69, env_temp=-1.0, left_board_temp=68, left_board_chip_temp=63, center_board_temp=69, center_board_chip_temp=64, right_board_temp=69, right_board_chip_temp=64, wattage=-1, wattage_limit=-1, fans=[Fan(speed=5760), Fan(speed=5700), Fan(speed=6000), Fan(speed=5790)], fan_1=5760, fan_2=5700, fan_3=6000, fan_4=5790, fan_psu=-1, left_chips=88, center_chips=88, right_chips=88, total_chips=264, ideal_chips=228, percent_ideal_chips=116, percent_ideal_hashrate=98, percent_ideal_wattage=100, nominal=False, pool_split='0', pool_1_url='sha256asicboost.auto.nicehash.com:9200', pool_1_user='SOMETHING.S1982T-01', pool_2_url='', pool_2_user='', errors=[], fault_light=False, efficiency=0)
`

**And this is from the Braiins OS firmware:**

`MinerData(ip='10.10.0.102', datetime=datetime.datetime(2023, 5, 29, 18, 15, 5, 344579, tzinfo=datetime.timezone(datetime.timedelta(seconds=7200), 'CEST')), mac='00:00:00:00:00:00', model='Unknown', make='Unknown', api_ver='Unknown', fw_ver='Unknown', hostname='Unknown', hashrate=0, _hashrate=0, nominal_hashrate=0, hashboards=[], ideal_hashboards=1, left_board_hashrate=0, center_board_hashrate=0, right_board_hashrate=0, temperature_avg=0, env_temp=-1.0, left_board_temp=0, left_board_chip_temp=0, center_board_temp=0, center_board_chip_temp=0, right_board_temp=0, right_board_chip_temp=0, wattage=-1, wattage_limit=-1, fans=[], fan_1=None, fan_2=None, fan_3=None, fan_4=None, fan_psu=-1, left_chips=0, center_chips=0, right_chips=0, total_chips=0, ideal_chips=1, percent_ideal_chips=0, percent_ideal_hashrate=0, percent_ideal_wattage=100, nominal=False, pool_split='0', pool_1_url='Unknown', pool_1_user='Unknown', pool_2_url='', pool_2_user='', errors=[], fault_light=None, efficiency=0)
INFO:src.logger:MinerData(ip='10.10.0.102', datetime=datetime.datetime(2023, 5, 29, 18, 15, 5, 344579, tzinfo=datetime.timezone(datetime.timedelta(seconds=7200), 'CEST')), mac='00:00:00:00:00:00', model='Unknown', make='Unknown', api_ver='Unknown', fw_ver='Unknown', hostname='Unknown', hashrate=0, _hashrate=0, nominal_hashrate=0, hashboards=[], ideal_hashboards=1, left_board_hashrate=0, center_board_hashrate=0, right_board_hashrate=0, temperature_avg=0, env_temp=-1.0, left_board_temp=0, left_board_chip_temp=0, center_board_temp=0, center_board_chip_temp=0, right_board_temp=0, right_board_chip_temp=0, wattage=-1, wattage_limit=-1, fans=[], fan_1=None, fan_2=None, fan_3=None, fan_4=None, fan_psu=-1, left_chips=0, center_chips=0, right_chips=0, total_chips=0, ideal_chips=1, percent_ideal_chips=0, percent_ideal_hashrate=0, percent_ideal_wattage=100, nominal=False, pool_split='0', pool_1_url='Unknown', pool_1_user='Unknown', pool_2_url='', pool_2_user='', errors=[], fault_light=None, efficiency=0)`

I was using 0.33.11 and I upgraded to the latest version (0.33.16)  and the issue persist,

This is the code I am using (based on docs):

```
all_miner_data = await asyncio.gather(*[miner.get_data() for miner in miners])
      for miner_data in all_miner_data:
          try:
              logger.logger.info(f"{miner_data}")
              # logger.logger.info(f"{miner_data.hostname}: {miner_data.hashrate}TH @ {miner_data.temperature_avg} ˚C")
          except Exception as e:
              logger.logger.info(f"failed with error {e}")
              
```

UpstreamData, 2023-05-29:
Looking at this, I see a few issues already that pyasic doesn't seem to be handling.

In the first set of data, the miner is being identified as an S19 with 76 chips, but it seems the miner you have is an S19 with 88.

Secondly, the miner isn't being identified properly as a Braiins OS S19, so the data being reported by the miner must have changed locations.

Could you try calling the `miner.api.devdetails()` and `miner.api.version()` commands?

mmitech, 2023-05-29:
```
{'STATUS': [{'STATUS': 'S', 'When': 1685379235, 'Code': 69, 'Msg': 'Device Details', 'Description': 'BOSer boser-buildroot 0.1.0-9b223345'}], 'DEVDETAILS': [{'Chips': 88, 'Cores': 45232, 'DEVDETAILS': 0, 'Device Path': '', 'Driver': '', 'Frequency': 573, 'ID': 1, 'Kernel': '', 'Model': 'Antminer S19J88NoPic', 'Name': 'HashchainManager 1', 'Voltage': 14.8192878}, {'Chips': 88, 'Cores': 45232, 'DEVDETAILS': 1, 'Device Path': '', 'Driver': '', 'Frequency': 592, 'ID': 2, 'Kernel': '', 'Model': 'Antminer S19J88NoPic', 'Name': 'HashchainManager 2', 'Voltage': 14.8192878}, {'Chips': 88, 'Cores': 45232, 'DEVDETAILS': 2, 'Device Path': '', 'Driver': '', 'Frequency': 577, 'ID': 3, 'Kernel': '', 'Model': 'Antminer S19J88NoPic', 'Name': 'HashchainManager 3', 'Voltage': 14.8192878}], 'id': 1}
```

UpstreamData, 2023-05-29:
I'll get this fixed on BOS.  In the meantime, would you mind sending those same outputs on stock?  I'm curious if I can identify it as an 88 chip S19 model on stock...

mmitech, 2023-05-29:
The api.devdetails() fails with this error `API Command Error: devdetails: Invalid command`

but the api.version() returns:

`{'STATUS': [{'STATUS': 'S', 'When': 1685381221, 'Code': 22, 'Msg': 'CGMiner versions', 'Description': 'cgminer 1.0.0'}], 'VERSION': [{'BMMiner': '1.0.0', 'API': '3.1', 'Miner': 'uart_trans.1.3', 'CompileTime': 'Thu Jun 9 13:06:22 CST 2022', 'Type': 'Antminer S19'}], 'id': 1}`

UpstreamData, 2023-05-29:
Alright, so with stock its just an S19.  That's not great.

Just doing some quick tests with the updated info, then it should work.

Does it show up any different on stock calling `miner.api.get_system_info()`?

UpstreamData, 2023-05-29:
Should be fixed in `v0.33.17`

mmitech, 2023-05-29:
not sure if I am doing anything wrong but the miner.api.get_system_info() returns ` failed with error 'UnknownAPI' object has no attribute 'get_system_info'`

UpstreamData, 2023-05-29:
You have to do it on stock, that wont work on braiins or a mis-identified miner.

mmitech, 2023-05-29:
OK I can confirm the issue is fixed on v0.33.17

closed 2023-05-29 by UpstreamData