UpstreamData/pyasic #40: resume_mining doesn't work

mmitech, 2023-05-31:
resume_mining doesn't work on both stock firmware and BrainOS, calling it returns false on Braiins OS, it does return true in stock firmware but the miner never resumes mining

UpstreamData, 2023-06-02:
Which version of BOS+?  Its possible they may have changed something there, that command just calls the API pause command, so to test that you can run `miner.api.send_command("pause", ignore_errors=True)`

I will do some testing and research on X19, but I know the way I have it set up does work on a bunch of X19 models.

mmitech, 2023-06-02:
pausing from the terminal using `echo '{"command":"pause"}' | nc "IP" 4028` works just fine
BOS+ `Version: 2023-05-29-0-1daef48b-23.03.2-plus`

BTW, pause doesn't work on the stock firmware as well.

UpstreamData, 2023-06-02:
Ok, absolute latest version.  There's a good chance they changed something about the result type in that version.  Can you send me the result of that command, either via pyasic or the terminal, as I don't have any S19 to put on BOS+, and S9s cant get that version.

Regarding stock, that was what I meant when I said I would do some research, but if you want, you can check the sleep mode value by:
-  Going to the configuration page on your miner,
-  Opening devtools (CTRL+SHIFT+I) and going to the network tab,
- Changing the mode to sleep mode,
- Submitting,
- And checking the value of the data sent to `set_miner_conf.cgi`

UpstreamData, 2023-06-02:
The results of those commands seem the same.  Was this done from CLI or via pyasic?

When I hard code those values into the api commands, the `stop_mining` and `resume_mining` commands both return True.

Stock looks normal to me, I will see if I have to adjust the order of the data or something.

UpstreamData, 2023-06-02:
I did that re-order with the stock config, that change is in `v0.33.21`.

mmitech, 2023-06-02:
> The results of those commands seem the same. Was this done from CLI or via pyasic?
> 
> When I hard code those values into the api commands, the `stop_mining` and `resume_mining` commands both return True.
> 
> Stock looks normal to me, I will see if I have to adjust the order of the data or something.

it was done from CLI

mmitech, 2023-06-02:
OK the stock firmware works now

mmitech, 2023-06-03:
BTW, the pause command work, only resumes doesn't work on BOS+ which is strange

UpstreamData, 2023-06-05:
Please see this PR (#43), it is still very much incomplete, but with some testing it will hopefully fix this issue. Any suggestions can be added there.

I believe this PR should fix this issue, I think what's happening is the miner isn't being identified properly.  If you could test by checking the output of `print(await pyasic.get_miner(ip))` on a paused miner running Braiins OS+?

mmitech, 2023-06-05:
yes, you are right, the miner is unknown when paused.

`DEBUG [BMMiner S19: 10.10.0.101, Unknown: 10.10.0.102, Unknown: 10.10.0.103, Unknown: 10.10.0.104, Unknown: 10.10.0.105, Unknown: 10.10.0.106, Unknown: 10.10.0.107, Unknown: 10.10.0.108, Unknown: 10.10.0.109, Unknown: 10.10.0.110, Unknown: 10.10.0.111, Unknown: 10.10.0.112, Unknown: 10.10.0.113, Unknown: 10.10.0.114, Unknown: 10.10.0.115, Unknown: 10.10.0.116, Unknown: 10.10.0.117, Unknown: 10.10.0.118, Unknown: 10.10.0.119, Unknown: 10.10.0.120, Unknown: 10.10.0.121, Unknown: 10.10.0.122, Unknown: 10.10.0.123, Unknown: 10.10.0.124, Unknown: 10.10.0.125, Unknown: 10.10.0.126, Unknown: 10.10.0.127, Unknown: 10.10.0.128, Unknown: 10.10.0.129, BOSMiner S19j No PIC: 10.10.0.130, BOSMiner S19j No PIC: 10.10.0.131, BOSMiner S19j No PIC: 10.10.0.132, BOSMiner S19j No PIC: 10.10.0.133, BMMiner S19: 10.10.0.134, BOSMiner S19j No PIC: 10.10.0.135, BOSMiner S19j No PIC: 10.10.0.136, BOSMiner S19j No PIC: 10.10.0.138, BOSMiner S19j No PIC: 10.10.0.139, BOSMiner S19j No PIC: 10.10.0.140, BOSMiner S19j No PIC: 10.10.0.141, BOSMiner S19j No PIC: 10.10.0.142, BOSMiner S19j No PIC: 10.10.0.143, BOSMiner S19j No PIC: 10.10.0.144, BOSMiner S19j No PIC: 10.10.0.145, BOSMiner S19j No PIC: 10.10.0.146, BOSMiner S19j No PIC: 10.10.0.147, BOSMiner S19j No PIC: 10.10.0.148, BOSMiner S19j No PIC: 10.10.0.149, BOSMiner S19j No PIC: 10.10.0.150]`