UpstreamData/pyasic #41: support for avalon 1166Pro and 1246

gav1111, 2023-06-03:
**Is your feature request related to a problem? Please describe.**
Support for Avalon 1166Pro and 1246. 

**Describe the solution you'd like**
For these to be recognized. 

**Describe alternatives you've considered**
Running excel spreadsheet and importing csv exported by BTCtools

UpstreamData, 2023-06-04:
Can you send the result of `await pyasic.miners.miner_factory.MinerFactory()._get_miner_type(IP)`?  This will help me figure out what data I need to be looking for to identify the miner.

gav1111, 2023-06-04:
`('A', 'CGMiner', 'ALONMINER 1166Pro', '3.7')`
`('A', 'CGMiner', 'ALONMINER 1246', '3.7')`

UpstreamData, 2023-06-07:
Working on this in #43.  If you could get me the results of the test script from [this comment](https://github.com/UpstreamData/pyasic/pull/43#issuecomment-1581251223) I can get that PR updated and testable for this.

To run these you will have to create a test script inside of a downloaded version of this repo on that branch, the version from pip cannot be used.