UpstreamData/pyasic #42: s17 pro not recognized in some instances

gav1111, 2023-06-04:
**Describe the bug**
Some s17 pro miners are not recognized and thus cannot be identified or queried as powered on miners. 
For these miners Socket connect failed: Connection refused is written on their webgui. 

**To Reproduce**
Seems to be inherent to the miner rather than certain steps. 

**Expected behavior**
To return miner information saying that hash is 0 etc etc

**Screenshots**
Here is btctools screenshot for one of miners in question and webgui screenshot
[image]
[image]

**Miner Information (If applicable):**
 - Antminer
 - S17 Pro
 - Firmware Type: Stock
 - Firmware Version: "Hardware Version	Socket connect failed: Connection refused"

**Additional context**
For S17+ models running Vnish 2.0.4 which also have socket connect failed messages on the webgui, they are still recognized by pyasic (even though data is not queried correctly). 

results of `await pyasic.miners.miner_factory.MinerFactory()._get_miner_type` for s17 pro

`WARNING:root:172.16.1.65 - Command devdetails+version: [Errno 111] Connect call failed ('172.16.1.65', 4028)
WARNING:root:172.16.1.65 - Command devdetails: [Errno 111] Connect call failed ('172.16.1.65', 4028)
WARNING:root:172.16.1.65 - Command version: [Errno 111] Connect call failed ('172.16.1.65', 4028)
WARNING:root:172.16.1.65 - Command get_version: [Errno 111] Connect call failed ('172.16.1.65', 4028)
WARNING:root:172.16.1.65: API Command Error: No API data.
(None, None, None, None)`

UpstreamData, 2023-06-04:
In consideration of this, #41, #37, and #40, which are all related to the way miners are identified, obviously something is going to have to be updated on that side.  I have been working on trying to find a way to update this for a while now, but it might be time to just make a branch for it for people to test it.

Hopefully have something to test on Monday.

gav1111, 2023-06-04:
Wonderful, let me know if I can help, I have a little bit of programming experience but with python its a bit of a struggle for me.

closed 2023-06-12 by UpstreamData