UpstreamData/pyasic #44: _hashrate and hashrate are not equal when using as_influxdb()

gav1111, 2023-06-12:
**Describe the bug**
I am running .get_data() on a list of ips returned by .scan_network_for_miners() and then exporting to influxdb by using .as_influxdb(). 

For some miners fields _hashrate and hashrate are not equal, for more popular miners these fields are equal. For example
S17 Pro AP 2.0.3:
`ip=.152,mac=00:00:00:00:00:00,model=Unknown,hostname=Unknown api_ver="0.0.0",fw_ver="Unknown",hashrate=42.07,_hashrate=0,nominal_hashrate=43.52`

S17+ (Vnish 2.0.4)
`ip=.177,mac=00:00:00:00:00:00,model=Unknown,hostname=Unknown api_ver="0.0.0",fw_ver="Unknown",hashrate=16.94,_hashrate=0,nominal_hashrate=17.47`

and then the opposite for avalons:
`ip=.164,mac=B4:A2:...,model=Avalon\ 1166,hostname=Unknown make="AvalonMiner",api_ver="0.0.0",fw_ver="4.11.1",hashrate=0.0,_hashrate=50.08,nominal_hashrate=0`

**Expected behavior**
I would expect them to be equal, and just wondering how exactly these fields are calculated. 

**Miner Information (If applicable):**
 - Manufacturer: Antminer
 - Type: S17+
 - Firmware Type: Vnish 2.0.4

 - Manufacturer: Antminer
 - Type: S17 Pro
 - Firmware Type: A.P. 2.0.3

**Additional context**
When querying data from influxdb I cant use hashrate or _hashrate because neither is accurate.

UpstreamData, 2023-06-12:
I might need to modify that function to ignore _hashrate.  To clarify, this is actually related to the way pyasic gets data in the first place.

_hashrate is a backup, it used to be the main hashrate before hashboards was added, and since each hashboard has its own hashrate, now hashrate is a property which adds up the hashrate from each board.  _hashrate is a backup in cases where hashboards can't be found or are explicitly excluded, and is the old way of getting hashrate, which is just a data point exposed by the miner's API.

That's why the could differ slightly, but usually shouldn't deviate by more than ~1%, not more than a rounding error.

Regarding your specific case - that has to do with incomplete compatibility with VNish specifically, as it is relatively unfinished and untested.  Avalons I believe do not expose per board hashrate at all (I could be mistaken).  It is possible that the stats command contains this under PVT_T{Board_Num}, but I would have to have more in depth information to check that out.

I can attempt to fix this, I'll provide a script for gathering all API data so I can test with it.

Please note I am gone all this week starting tomorrow, so it's very likely this won't get much attention until next Monday.

UpstreamData, 2023-06-12:
```python3
import pyasic
import asyncio
import pprint

IP = "192.168.1.1"

async def main():
    miner = await pyasic.get_miner(IP)
    for c in miner.api.commands:
        pprint.pprint(await miner.api.send_command(c, ignore_errors=True))
    
if __name__ == '__main__':
    asyncio.run(main())
```

Please send this in a text file, as it will be a rather long output.

gav1111, 2023-06-12:
I am getting the following errors, should I have some special git checkout?
`Traceback (most recent call last):
  File "/home/sf/temp/pyasic/hashr_request1.py", line 13, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/home/sf/temp/pyasic/hashr_request1.py", line 9, in main
    for c in miner.api.commands:
AttributeError: 'NoneType' object has no attribute 'api'`

UpstreamData, 2023-06-12:
You'll want to update to latest, then make sure you change the IP.  Pyasic isn't picking up a miner at the IP it's getting.

`pip install -U pyasic`

gav1111, 2023-06-12:
avalon 1166
[avalon1166.txt](https://github.com/UpstreamData/pyasic/files/11728867/avalon1166.txt)

UpstreamData, 2023-06-12:
Looks like it exposes it as MGHS in the stats command.

UpstreamData, 2023-06-16:
See how the `dev-get_miner` branch works for you?  `pip install pyasic@git+https://github.com/UpstreamData/pyasic@dev-miner_data`

gav1111, 2023-06-16:
getting some errors 
`[WARNING][06/16/23 11:00:43](root) - 172.16.1.160: Ping And Get Miner Exception: name 'url' is not defined
/usr/lib/python3.9/asyncio/base_events.py:1738 in default_exception_handler
[ERROR][06/16/23 11:01:07](asyncio) - unhandled exception during asyncio.run() shutdown
task: <Task finished name='Task-7269' coro=<BaseMiner.get_data() done, defined at /home/sf/.local/lib/python3.9/site-packages/pyasic/miners/base.py:453> exception=RuntimeError('The connection pool was closed while 1 HTTP requests/responses were still in-flight.')>
Traceback (most recent call last):
  File "/home/sf/.local/lib/python3.9/site-packages/pyasic/web/antminer.py", line 46, in send_command
    data = await client.get(url, auth=auth)
  File "/home/sf/.local/lib/python3.9/site-packages/httpx/_client.py", line 1757, in get
    return await self.request(
  File "/home/sf/.local/lib/python3.9/site-packages/httpx/_client.py", line 1530, in request
    return await self.send(request, auth=auth, follow_redirects=follow_redirects)
  File "/home/sf/.local/lib/python3.9/site-packages/httpx/_client.py", line 1631, in send
    raise exc
  File "/home/sf/.local/lib/python3.9/site-packages/httpx/_client.py", line 1625, in send
    await response.aread()
  File "/home/sf/.local/lib/python3.9/site-packages/httpx/_models.py", line 909, in aread
    self._content = b"".join([part async for part in self.aiter_bytes()])
  File "/home/sf/.local/lib/python3.9/site-packages/httpx/_models.py", line 909, in <listcomp>
    self._content = b"".join([part async for part in self.aiter_bytes()])
  File "/home/sf/.local/lib/python3.9/site-packages/httpx/_models.py", line 927, in aiter_bytes
    async for raw_bytes in self.aiter_raw():
  File "/home/sf/.local/lib/python3.9/site-packages/httpx/_models.py", line 993, in aiter_raw
    await self.aclose()
  File "/home/sf/.local/lib/python3.9/site-packages/httpx/_models.py", line 1006, in aclose
    await self.stream.aclose()
  File "/home/sf/.local/lib/python3.9/site-packages/httpx/_client.py", line 152, in aclose
    await self._stream.aclose()
  File "/home/sf/.local/lib/python3.9/site-packages/httpx/_transports/default.py", line 244, in aclose
    await self._httpcore_stream.aclose()
  File "/home/sf/.local/lib/python3.9/site-packages/httpcore/_async/connection_pool.py", line 354, in aclose
    await self._pool.response_closed(self._status)
  File "/home/sf/.local/lib/python3.9/site-packages/httpcore/_async/connection_pool.py", line 277, in response_closed
    async with self._pool_lock:
  File "/home/sf/.local/lib/python3.9/site-packages/httpcore/_synchronization.py", line 15, in __aenter__
    await self._lock.acquire()
  File "/home/sf/.local/lib/python3.9/site-packages/anyio/_core/_synchronization.py", line 142, in acquire
    await cancel_shielded_checkpoint()
  File "/home/sf/.local/lib/python3.9/site-packages/anyio/lowlevel.py", line 57, in cancel_shielded_checkpoint
    await get_asynclib().cancel_shielded_checkpoint()
  File "/home/sf/.local/lib/python3.9/site-packages/anyio/_backends/_asyncio.py", line 539, in cancel_shielded_checkpoint
    await sleep(0)
  File "/usr/lib/python3.9/asyncio/tasks.py", line 639, in sleep
    await __sleep0()
  File "/usr/lib/python3.9/asyncio/tasks.py", line 633, in __sleep0
    yield
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/sf/.local/lib/python3.9/site-packages/pyasic/miners/base.py", line 476, in get_data
    gathered_data = await self._get_data(allow_warning, data_to_get=data_to_get)
  File "/home/sf/.local/lib/python3.9/site-packages/pyasic/miners/base.py", line 429, in _get_data
    miner_data[data_name] = await function(**args_to_send)
  File "/home/sf/.local/lib/python3.9/site-packages/pyasic/miners/backends/antminer.py", line 126, in get_mac
    data = await self.web.get_system_info()
  File "/home/sf/.local/lib/python3.9/site-packages/pyasic/web/antminer.py", line 93, in get_system_info
    return await self.send_command("get_system_info")
  File "/home/sf/.local/lib/python3.9/site-packages/pyasic/web/antminer.py", line 46, in send_command
    data = await client.get(url, auth=auth)
  File "/home/sf/.local/lib/python3.9/site-packages/httpx/_client.py", line 2003, in __aexit__
    await self._transport.__aexit__(exc_type, exc_value, traceback)
  File "/home/sf/.local/lib/python3.9/site-packages/httpx/_transports/default.py", line 332, in __aexit__
    await self._pool.__aexit__(exc_type, exc_value, traceback)
  File "/home/sf/.local/lib/python3.9/site-packages/httpcore/_async/connection_pool.py", line 326, in __aexit__
    await self.aclose()
  File "/home/sf/.local/lib/python3.9/site-packages/httpcore/_async/connection_pool.py", line 312, in aclose
    raise RuntimeError(
RuntimeError: The connection pool was closed while 1 HTTP requests/responses were still in-flight.
Traceback (most recent call last):
  File "/home/sf/temp/pyasic/production_center.py", line 66, in <module>
    asyncio.run(scan_and_get_data())
  File "/usr/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/home/sf/temp/pyasic/production_center.py", line 28, in scan_and_get_data
    data = await asyncio.gather(*[m.get_data(data_to_get = [
  File "/home/sf/.local/lib/python3.9/site-packages/pyasic/miners/base.py", line 476, in get_data
    gathered_data = await self._get_data(allow_warning, data_to_get=data_to_get)
  File "/home/sf/.local/lib/python3.9/site-packages/pyasic/miners/base.py", line 429, in _get_data
    miner_data[data_name] = await function(**args_to_send)
  File "/home/sf/.local/lib/python3.9/site-packages/pyasic/miners/backends/bosminer.py", line 548, in get_hashboards
    board = hashboards[_id]
IndexError: list index out of range`

UpstreamData, 2023-06-16:
What's the script you're running?  Could you send it here please?  Sorry if you've sent it before, I probably have it, but I have a lot of stuff to keep track of.

gav1111, 2023-06-16:
here is the script 
`import asyncio

from pyasic.network import MinerNetwork
#from pyasic.settings import PyasicSettings

#PyasicSettings().network_ping_timeout = 30

import influxdb_client, os, time
from influxdb_client import InfluxDBClient, Point, WritePrecision
from influxdb_client.client.write_api import SYNCHRONOUS

# define asynchronous function to scan for miners
async def scan_and_get_data():
    # Define network range to be used for scanning
    # This can take a list of IPs, a constructor string, or an IP and subnet mask
    # The standard mask is /24 (x.x.x.0-255), and you can pass any IP address in the subnet
    net = MinerNetwork("172.16.1.1", mask=21)
    # Scan the network for miners
    # This function returns a list of miners of the correct type as a class
    miners: list = await net.scan_network_for_miners()

    # We can now get data from any of these miners
    # To do them all we have to create a list of tasks and gather them
    # tasks = [miner.get_data() for miner in miners]
    # Gather all tasks asynchronously and run them
    data = await asyncio.gather(*[m.get_data(data_to_get = [
                "mac",
                "model",
                "api_ver",
                "fw_ver",
                "hashrate",
                "nominal_hashrate",
                "hashboards",
                "env_temp",
                "wattage",
                "wattage_limit",
                "fans",
                "fan_psu",
                "errors",
                "fault_light",
                "pools",
            ]) for m in miners])
    data = [d.as_influxdb() for d in data]

    # Data is now a list of MinerData, and we can reference any part of that
    # Print out all data for now
    for item in data:
     print(item)

    #initialize influxdb
    token = "xxxx=="
    org = "xxxx"
    url = "http://10.10.10.1:8086"

    write_client = influxdb_client.InfluxDBClient(url=url, token=token, org=org)

    bucket="xxxx"

    write_api = write_client.write_api(write_options=SYNCHRONOUS)
    for item in data:
     write_api.write(bucket=bucket, org="xxxx", record=item)

if __name__ == "__main__":
	asyncio.run(scan_and_get_data())`

UpstreamData, 2023-06-19:
I think I got that fixed, let me know how it works.

gav1111, 2023-06-20:
I am getting the following error
```
  warnings.warn(
/usr/lib/python3.9/asyncio/base_events.py:1738 in default_exception_handler
[ERROR][06/20/23 09:13:34](asyncio) - unhandled exception during asyncio.run() shutdown
task: <Task finished name='Task-7239' coro=<BaseMiner.get_data() done, defined at /home/sf/.local/lib/python3.9/site-packages/pyasic/miners/base.py:453> exception=RuntimeError('The connection pool was closed while 1 HTTP requests/responses were still in-flight.')>
Traceback (most recent call last):
  File "/home/sf/.local/lib/python3.9/site-packages/pyasic/web/antminer.py", line 46, in send_command
    data = await client.get(url, auth=auth)
  File "/home/sf/.local/lib/python3.9/site-packages/httpx/_client.py", line 1757, in get
    return await self.request(
  File "/home/sf/.local/lib/python3.9/site-packages/httpx/_client.py", line 1530, in request
    return await self.send(request, auth=auth, follow_redirects=follow_redirects)
  File "/home/sf/.local/lib/python3.9/site-packages/httpx/_client.py", line 1631, in send
    raise exc
  File "/home/sf/.local/lib/python3.9/site-packages/httpx/_client.py", line 1625, in send
    await response.aread()
  File "/home/sf/.local/lib/python3.9/site-packages/httpx/_models.py", line 909, in aread
    self._content = b"".join([part async for part in self.aiter_bytes()])
  File "/home/sf/.local/lib/python3.9/site-packages/httpx/_models.py", line 909, in <listcomp>
    self._content = b"".join([part async for part in self.aiter_bytes()])
  File "/home/sf/.local/lib/python3.9/site-packages/httpx/_models.py", line 927, in aiter_bytes
    async for raw_bytes in self.aiter_raw():
  File "/home/sf/.local/lib/python3.9/site-packages/httpx/_models.py", line 993, in aiter_raw
    await self.aclose()
  File "/home/sf/.local/lib/python3.9/site-packages/httpx/_models.py", line 1006, in aclose
    await self.stream.aclose()
  File "/home/sf/.local/lib/python3.9/site-packages/httpx/_client.py", line 152, in aclose
    await self._stream.aclose()
  File "/home/sf/.local/lib/python3.9/site-packages/httpx/_transports/default.py", line 244, in aclose
    await self._httpcore_stream.aclose()
  File "/home/sf/.local/lib/python3.9/site-packages/httpcore/_async/connection_pool.py", line 354, in aclose
    await self._pool.response_closed(self._status)
  File "/home/sf/.local/lib/python3.9/site-packages/httpcore/_async/connection_pool.py", line 277, in response_closed
    async with self._pool_lock:
  File "/home/sf/.local/lib/python3.9/site-packages/httpcore/_synchronization.py", line 15, in __aenter__
    await self._lock.acquire()
  File "/home/sf/.local/lib/python3.9/site-packages/anyio/_core/_synchronization.py", line 142, in acquire
    await cancel_shielded_checkpoint()
  File "/home/sf/.local/lib/python3.9/site-packages/anyio/lowlevel.py", line 57, in cancel_shielded_checkpoint
    await get_asynclib().cancel_shielded_checkpoint()
  File "/home/sf/.local/lib/python3.9/site-packages/anyio/_backends/_asyncio.py", line 539, in cancel_shielded_checkpoint
    await sleep(0)
  File "/usr/lib/python3.9/asyncio/tasks.py", line 639, in sleep
    await __sleep0()
  File "/usr/lib/python3.9/asyncio/tasks.py", line 633, in __sleep0
    yield
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/sf/.local/lib/python3.9/site-packages/pyasic/miners/base.py", line 476, in get_data
    gathered_data = await self._get_data(allow_warning, data_to_get=data_to_get)
  File "/home/sf/.local/lib/python3.9/site-packages/pyasic/miners/base.py", line 429, in _get_data
    miner_data[data_name] = await function(**args_to_send)
  File "/home/sf/.local/lib/python3.9/site-packages/pyasic/miners/backends/antminer.py", line 126, in get_mac
    data = await self.web.get_system_info()
  File "/home/sf/.local/lib/python3.9/site-packages/pyasic/web/antminer.py", line 93, in get_system_info
    return await self.send_command("get_system_info")
  File "/home/sf/.local/lib/python3.9/site-packages/pyasic/web/antminer.py", line 46, in send_command
    data = await client.get(url, auth=auth)
  File "/home/sf/.local/lib/python3.9/site-packages/httpx/_client.py", line 2003, in __aexit__
    await self._transport.__aexit__(exc_type, exc_value, traceback)
  File "/home/sf/.local/lib/python3.9/site-packages/httpx/_transports/default.py", line 332, in __aexit__
    await self._pool.__aexit__(exc_type, exc_value, traceback)
  File "/home/sf/.local/lib/python3.9/site-packages/httpcore/_async/connection_pool.py", line 326, in __aexit__
    await self.aclose()
  File "/home/sf/.local/lib/python3.9/site-packages/httpcore/_async/connection_pool.py", line 312, in aclose
    raise RuntimeError(
RuntimeError: The connection pool was closed while 1 HTTP requests/responses were still in-flight.
Traceback (most recent call last):
  File "/home/sf/temp/pyasic/production_center.py", line 66, in <module>
    asyncio.run(scan_and_get_data())
  File "/usr/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/home/sf/temp/pyasic/production_center.py", line 28, in scan_and_get_data
    data = await asyncio.gather(*[m.get_data(data_to_get = [
  File "/home/sf/.local/lib/python3.9/site-packages/pyasic/miners/base.py", line 476, in get_data
    gathered_data = await self._get_data(allow_warning, data_to_get=data_to_get)
  File "/home/sf/.local/lib/python3.9/site-packages/pyasic/miners/base.py", line 429, in _get_data
    miner_data[data_name] = await function(**args_to_send)
  File "/home/sf/.local/lib/python3.9/site-packages/pyasic/miners/backends/bosminer.py", line 548, in get_hashboards
    board = hashboards[_id]
IndexError: list index out of range
```

gav1111, 2023-06-20:
getting some errors still
```
  warnings.warn(
/usr/lib/python3.9/asyncio/base_events.py:1738 in default_exception_handler
[ERROR][06/20/23 09:31:37](asyncio) - unhandled exception during asyncio.run() shutdown
task: <Task finished name='Task-7216' coro=<BaseMiner.get_data() done, defined at /home/sf/.local/lib/python3.9/site-packages/pyasic/miners/base.py:453> exception=RuntimeError('The connection pool was closed while 1 HTTP requests/responses were still in-flight.')>
Traceback (most recent call last):
  File "/home/sf/.local/lib/python3.9/site-packages/pyasic/web/antminer.py", line 46, in send_command
    data = await client.get(url, auth=auth)
  File "/home/sf/.local/lib/python3.9/site-packages/httpx/_client.py", line 1757, in get
    return await self.request(
  File "/home/sf/.local/lib/python3.9/site-packages/httpx/_client.py", line 1530, in request
    return await self.send(request, auth=auth, follow_redirects=follow_redirects)
  File "/home/sf/.local/lib/python3.9/site-packages/httpx/_client.py", line 1631, in send
    raise exc
  File "/home/sf/.local/lib/python3.9/site-packages/httpx/_client.py", line 1625, in send
    await response.aread()
  File "/home/sf/.local/lib/python3.9/site-packages/httpx/_models.py", line 909, in aread
    self._content = b"".join([part async for part in self.aiter_bytes()])
  File "/home/sf/.local/lib/python3.9/site-packages/httpx/_models.py", line 909, in <listcomp>
    self._content = b"".join([part async for part in self.aiter_bytes()])
  File "/home/sf/.local/lib/python3.9/site-packages/httpx/_models.py", line 927, in aiter_bytes
    async for raw_bytes in self.aiter_raw():
  File "/home/sf/.local/lib/python3.9/site-packages/httpx/_models.py", line 993, in aiter_raw
    await self.aclose()
  File "/home/sf/.local/lib/python3.9/site-packages/httpx/_models.py", line 1006, in aclose
    await self.stream.aclose()
  File "/home/sf/.local/lib/python3.9/site-packages/httpx/_client.py", line 152, in aclose
    await self._stream.aclose()
  File "/home/sf/.local/lib/python3.9/site-packages/httpx/_transports/default.py", line 244, in aclose
    await self._httpcore_stream.aclose()
  File "/home/sf/.local/lib/python3.9/site-packages/httpcore/_async/connection_pool.py", line 354, in aclose
    await self._pool.response_closed(self._status)
  File "/home/sf/.local/lib/python3.9/site-packages/httpcore/_async/connection_pool.py", line 277, in response_closed
    async with self._pool_lock:
  File "/home/sf/.local/lib/python3.9/site-packages/httpcore/_synchronization.py", line 15, in __aenter__
    await self._lock.acquire()
  File "/home/sf/.local/lib/python3.9/site-packages/anyio/_core/_synchronization.py", line 142, in acquire
    await cancel_shielded_checkpoint()
  File "/home/sf/.local/lib/python3.9/site-packages/anyio/lowlevel.py", line 57, in cancel_shielded_checkpoint
    await get_asynclib().cancel_shielded_checkpoint()
  File "/home/sf/.local/lib/python3.9/site-packages/anyio/_backends/_asyncio.py", line 539, in cancel_shielded_checkpoint
    await sleep(0)
  File "/usr/lib/python3.9/asyncio/tasks.py", line 639, in sleep
    await __sleep0()
  File "/usr/lib/python3.9/asyncio/tasks.py", line 633, in __sleep0
    yield
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/sf/.local/lib/python3.9/site-packages/pyasic/miners/base.py", line 476, in get_data
    gathered_data = await self._get_data(allow_warning, data_to_get=data_to_get)
  File "/home/sf/.local/lib/python3.9/site-packages/pyasic/miners/base.py", line 429, in _get_data
    miner_data[data_name] = await function(**args_to_send)
  File "/home/sf/.local/lib/python3.9/site-packages/pyasic/miners/backends/antminer.py", line 126, in get_mac
    data = await self.web.get_system_info()
  File "/home/sf/.local/lib/python3.9/site-packages/pyasic/web/antminer.py", line 93, in get_system_info
    return await self.send_command("get_system_info")
  File "/home/sf/.local/lib/python3.9/site-packages/pyasic/web/antminer.py", line 46, in send_command
    data = await client.get(url, auth=auth)
  File "/home/sf/.local/lib/python3.9/site-packages/httpx/_client.py", line 2003, in __aexit__
    await self._transport.__aexit__(exc_type, exc_value, traceback)
  File "/home/sf/.local/lib/python3.9/site-packages/httpx/_transports/default.py", line 332, in __aexit__
    await self._pool.__aexit__(exc_type, exc_value, traceback)
  File "/home/sf/.local/lib/python3.9/site-packages/httpcore/_async/connection_pool.py", line 326, in __aexit__
    await self.aclose()
  File "/home/sf/.local/lib/python3.9/site-packages/httpcore/_async/connection_pool.py", line 312, in aclose
    raise RuntimeError(
RuntimeError: The connection pool was closed while 1 HTTP requests/responses were still in-flight.
Traceback (most recent call last):
  File "/home/sf/temp/pyasic/production_center_1.py", line 66, in <module>
    asyncio.run(scan_and_get_data())
  File "/usr/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/home/sf/temp/pyasic/production_center_1.py", line 28, in scan_and_get_data
    data = await asyncio.gather(*[m.get_data(data_to_get = [
  File "/home/sf/.local/lib/python3.9/site-packages/pyasic/miners/base.py", line 476, in get_data
    gathered_data = await self._get_data(allow_warning, data_to_get=data_to_get)
  File "/home/sf/.local/lib/python3.9/site-packages/pyasic/miners/base.py", line 429, in _get_data
    miner_data[data_name] = await function(**args_to_send)
  File "/home/sf/.local/lib/python3.9/site-packages/pyasic/miners/backends/bosminer.py", line 548, in get_hashboards
    board = hashboards[_id]
IndexError: list index out of range
```

UpstreamData, 2023-06-20:
Interesting.  Seems like the index might be really weird on that machine.  What does this script output?

```python3
import pyasic
import asyncio

IP = "192.168.1.1"

async def main():
    m = await pyasic.get_miner(IP)
    data = await m.web.send_command(
        {
            "bosminer": {
                "info": {
                    "workSolver": {
                        "childSolvers": {
                            "name": None,
                            "realHashrate": {"mhs1M"},
                            "hwDetails": {"chips"},
                            "temperatures": {"degreesC"},
                        }
                    }
                }
            }
        },
    )
    print(data)

asyncio.run(main())
```

gav1111, 2023-06-20:
Can you help me narrow down which one it may be? I ran a couple of them and they returned okish looking results, but not sure what I am looking for, here is a sample:

.242
`{'data': {'bosminer': {'info': {'workSolver': {'childSolvers': [{'hwDetails': {'chips': 76}, 'name': '3', 'realHashrate': {'mhs1M': 24872865.493889533}, 'temperatures': [{'degreesC': 45.0}, {'degreesC': 60.0}]}]}}}}}`
.101
`{'data': {'bosminer': {'info': {'workSolver': {'childSolvers': [{'hwDetails': {'chips': 76}, 'name': '1', 'realHashrate': {'mhs1M': 25070177.770518556}, 'temperatures': [{'degreesC': 45.5}, {'degreesC': 60.5}]}, {'hwDetails': {'chips': 76}, 'name': '2', 'realHashrate': {'mhs1M': 25546688.236001726}, 'temperatures': [{'degreesC': 45.5}, {'degreesC': 60.5}]}, {'hwDetails': None, 'name': '3', 'realHashrate': {'mhs1M': 0.0}, 'temperatures': []}]}}}}}`

maybe .203 this one which only has 2 fans?
`{'data': {'bosminer': {'info': {'workSolver': {'childSolvers': [{'hwDetails': {'chips': 76}, 'name': '1', 'realHashrate': {'mhs1M': 20380184.314650767}, 'temperatures': [{'degreesC': 43.5}, {'degreesC': 58.5}]}, {'hwDetails': {'chips': 76}, 'name': '2', 'realHashrate': {'mhs1M': 21504185.801899213}, 'temperatures': [{'degreesC': 44.5}, {'degreesC': 59.5}]}, {'hwDetails': {'chips': 76}, 'name': '3', 'realHashrate': {'mhs1M': 22970786.527520243}, 'temperatures': [{'degreesC': 45.0}, {'degreesC': 60.0}]}]}}}}}`

UpstreamData, 2023-06-20:
Seems like they start indexing from 1.  Damn.  Ill see what I can do.

UpstreamData, 2023-06-20:
To clarify, the issue is actually with a board offset, EG, S9s start at 6, and do 6, 7, 8, S17 appear to do 1, 2, and 3, and S19s, do 0, 1, and 2.  I have to set up an offset to handle this, but its a bit of a pain.

UpstreamData, 2023-06-20:
Can you update from github again, then retry this script?  I might have found a workaround.

https://github.com/UpstreamData/pyasic/issues/44#issuecomment-1595096128

gav1111, 2023-06-20:
The script works, I dont think results are as expected though, here they are:
S17 Pro AP 2.0.3 (both _hashrate and hashrate fields are populated, great):
`ip=.152,mac=00:00:00:00:00:00,model=Unknown,hostname=Unknown api_ver="0.0.0",fw_ver="Unknown",hashrate=36.83,nominal_hashrate=43.52,hashboard_1_hashrate=11.89,hashboard_1_temperature=72,hashboard_1_chip_temperature=65,hashboard_1_chips=48,hashboard_1_expected_chips=0,hashboard_2_hashrate=12.13,hashboard_2_temp`

S17+ (Vnish 2.0.4) (both _hashrate and hashrate fields are populated, great):
`ip=.177,mac=00:00:00:00:00:00,model=Unknown,hostname=Unknown api_ver="0.0.0",fw_ver="Unknown",hashrate=12.41,nominal_hashrate=17.47,hashboard_1_hashrate=12.41,hashboard_1_temperature=75,hashboard_1_chip_temperature=64,hashboard_1_chips=65,hashbo`

Avalon 1166 (missing _hashrate field, hashrate fields is 0. Before _hashrate was showing correct hashrate):
`ip=.164,mac=B4:A2:..,model=Avalon\ 1166\ Pro,hostname=Unknown make="AvalonMiner",api_ver="0.0.0",fw_ver="4.11.1",hashrate=0.0,nominal_hashrate=0,hashboard_1_hashrate=0.0,hashboard_1_temperature=-1,hashboard_1_chip_temperature=-1,hashboard_1_chips=0,hashboard_1_expected_chips=120,hashboard_2_hashrate=0.0,hashboard_2_temperature=-1,hashboard_2_chip_temperature=-1,hashboard_2_chips=0,hashboard_2_expected_chips=120,hashboard_3_hashrate=0.0,hashboard_3_temperature=-1,hashboard_3_chip_temperature=-1,hashboard_3_chips=0,hashboard_3_expected_chips=120,ideal_hashboards=3,temperature_avg=0,env_temp=-1.0,wattage=-1,wattage_limit=-1,fan_1="-1",fan_2="-1",fan_3="-1",fan_4="-1",fan_psu=-1,total_chips=0,ideal_chips=360,percent_ideal_chips=0,percent_ideal_hashrate=0,percent_ideal_wattage=100,nominal=false,pool_split="0",pool_1_url="`

UpstreamData, 2023-06-20:
Both showing unknown model. Might not have the specific names of them set up properly or something.  Here's another script.  

```python3
import pyasic
import asyncio

async def main():
    miner = await pyasic.get_miner("192.168.1.152") # S17AP
    print(await miner.api.version())
    print(await miner.api.devdetails())

    miner = await pyasic.get_miner("192.168.1.152") # S17AP
    print(await miner.api.stats())

    miner = await pyasic.get_miner("192.168.1.164") # A1166 Pro
    print(await miner.get_data())

asyncio.run(main())
```

gav1111, 2023-06-20:
S17AP:
```
{'STATUS': [{'STATUS': 'S', 'When': 1687300219, 'Code': 22, 'Msg': 'CGMiner versions', 'Description': 'cgminer 4.11.1'}], 'VERSION': [{'CGMiner': '4.11.1', 'API': '3.7'}], 'id': 1}
/home/sf/.local/lib/python3.9/site-packages/pyasic/API/__init__.py:84 in send_command
[WARNING][06/20/23 15:30:20](root) - 172.16.1.152: API Command Error: devdetails: Access denied to 'devdetails' command
Traceback (most recent call last):
  File "/home/sf/temp/pyasic/620test2.py", line 15, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/home/sf/temp/pyasic/620test2.py", line 7, in main
    print(await miner.api.devdetails())
  File "/home/sf/.local/lib/python3.9/site-packages/pyasic/API/bmminer.py", line 390, in devdetails
    return await self.send_command("devdetails")
  File "/home/sf/.local/lib/python3.9/site-packages/pyasic/API/__init__.py", line 87, in send_command
    raise APIError(validation[1])
pyasic.errors.APIError: Access denied to 'devdetails' command
```

S17 VNISH
`{'STATUS': [{'STATUS': 'S', 'When': 1687300299, 'Code': 70, 'Msg': 'CGMiner stats', 'Description': 'cgminer 4.11.1'}], 'STATS': [{'Cgminer': '4.11.1', 'Miner': '35.0.1.3', 'CompileTime': 'Mon Nov 29 21:55:02 CST 2021', 'Type': 'Antminer S17+ (vnish 2.0.4)'}, {'STATS': 0, 'ID': 'BTM_S17P_0', 'Elapsed': 38976, 'Calls': 0, 'Wait': 0.0, 'Max': 0.0, 'Min': 99999999.0, 'GHS 5s': '11755.80', 'GHS av': 12388.15, 'miner_count': 3, 'total_acn': 65, 'frequency': 400, 'total_freqavg': 400.0, 'total_rateideal': 17472.0, 'total_rate': 11755.8, 'state': 'mining', 'fan_num': 4, 'fan_mode': 0, 'fan_pwm': 16, 'fan1': 1560, 'fan2': 1920, 'fan3': 1560, 'fan4': 1920, 'temp_num': 3, 'temp1': 65, 'temp2': 0, 'temp3': 0, 'temp2_1': 75, 'temp2_2': 0, 'temp2_3': 0, 'temp3_1': 75, 'temp3_2': 0, 'temp3_3': 0, 'temp_pcb1': '41-65-40-63', 'temp_pcb2': '0-0-0-0', 'temp_pcb3': '0-0-0-0', 'temp_chip1': '52-76-52-73', 'temp_chip2': '0-0-0-0', 'temp_chip3': '0-0-0-0', 'chain_acn1': 65, 'chain_acn2': 0, 'chain_acn3': 0, 'chain_vol1': 17700, 'chain_vol2': 17700, 'chain_vol3': 17700, 'freq_avg1': 400.0, 'freq_avg2': 0.0, 'freq_avg3': 0.0, 'chain_rateideal1': 17472.0, 'chain_rateideal2': 0.0, 'chain_rateideal3': 0.0, 'chain_rate1': 11755.8, 'chain_rate2': 0.0, 'chain_rate3': 0.0, 'chain_hw1': 1, 'chain_hw2': 0, 'chain_hw3': 0, 'chain_acs1': 'ooooo ooooo ooooo ooooo ooooo ooooo ooooo ooooo oooox ooooo ooooo xoooo oooox', 'chain_acs2': 'Failed to detect ASIC chips', 'chain_acs3': 'Failed to detect ASIC chips', 'chain_consumption1': 562, 'chain_consumption2': 5, 'chain_consumption3': 5, 'miner_version': '35.0.1.3', 'build_version': '2.0.3'}], 'id': 1}`

A1166Pro
```
Traceback (most recent call last):
  File "/home/sf/temp/pyasic/620test2.py", line 8, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/home/sf/temp/pyasic/620test2.py", line 6, in main
    print(await miner.get_data())
  File "/home/sf/.local/lib/python3.9/site-packages/pyasic/miners/base.py", line 476, in get_data
    gathered_data = await self._get_data(allow_warning, data_to_get=data_to_get)
  File "/home/sf/.local/lib/python3.9/site-packages/pyasic/miners/base.py", line 396, in _get_data
    web_command_data = await self.web.multicommand(
AttributeError: 'NoneType' object has no attribute 'multicommand'
```

UpstreamData, 2023-06-21:
VNish S17+ should be supported now, had to create a new type for it, although it seems it partially identified it as a VNish machine, so this will only add the model.

The S17AP doesn't appear to be passing its model there, not sure why, usually for antminers its there.

A1166 is missing a web attribute for some reason.  I remember implementing it at one point, but it appears it was removed somewhere.  It is completely unnecessary to have it for the core functionality of pyasic, so I have just removed references to it and moved back to just the RPC api, but in the future I will see if I can find a miner to test with to add it.

The only remaining miner there that _should_ be having issues is the S17AP, I just need to figure out where that data is.

Can you run the following script on it, so I can see where it passes the type?  Should be something like `Antminer S17 Pro`.

```python3
import pyasic
import asyncio
import pprint

IP = "192.168.1.1"

async def main():
    miner = await pyasic.get_miner(IP)
    for c in miner.api.commands:
        pprint.pprint(await miner.api.send_command(c, ignore_errors=True))
    
if __name__ == '__main__':
    asyncio.run(main())
```

gav1111, 2023-06-21:
here you go 
```
{'STATUS': [{'Code': 45,
             'Description': 'cgminer 4.11.1',
             'Msg': "Access denied to 'addpool' command",
             'STATUS': 'E',
             'When': 1687360169}],
 'id': 1}
{'STATUS': [{'Code': 45,
             'Description': 'cgminer 4.11.1',
             'Msg': "Access denied to 'asc' command",
             'STATUS': 'E',
             'When': 1687360169}],
 'id': 1}
{'STATUS': [{'Code': 45,
             'Description': 'cgminer 4.11.1',
             'Msg': "Access denied to 'asccount' command",
             'STATUS': 'E',
             'When': 1687360169}],
 'id': 1}
{'STATUS': [{'Code': 45,
             'Description': 'cgminer 4.11.1',
             'Msg': "Access denied to 'ascdisable' command",
             'STATUS': 'E',
             'When': 1687360170}],
 'id': 1}
{'STATUS': [{'Code': 45,
             'Description': 'cgminer 4.11.1',
             'Msg': "Access denied to 'ascenable' command",
             'STATUS': 'E',
             'When': 1687360170}],
 'id': 1}
{'STATUS': [{'Code': 45,
             'Description': 'cgminer 4.11.1',
             'Msg': "Access denied to 'ascidentify' command",
             'STATUS': 'E',
             'When': 1687360170}],
 'id': 1}
{'STATUS': [{'Code': 45,
             'Description': 'cgminer 4.11.1',
             'Msg': "Access denied to 'ascset' command",
             'STATUS': 'E',
             'When': 1687360170}],
 'id': 1}
{'STATUS': [{'Code': 45,
             'Description': 'cgminer 4.11.1',
             'Msg': "Access denied to 'check' command",
             'STATUS': 'E',
             'When': 1687360170}],
 'id': 1}
{'STATUS': [{'Code': 45,
             'Description': 'cgminer 4.11.1',
             'Msg': "Access denied to 'coin' command",
             'STATUS': 'E',
             'When': 1687360170}],
 'id': 1}
{'STATUS': [{'Code': 45,
             'Description': 'cgminer 4.11.1',
             'Msg': "Access denied to 'config' command",
             'STATUS': 'E',
             'When': 1687360170}],
 'id': 1}
{'STATUS': [{'Code': 45,
             'Description': 'cgminer 4.11.1',
             'Msg': "Access denied to 'debug' command",
             'STATUS': 'E',
             'When': 1687360170}],
 'id': 1}
{'STATUS': [{'Code': 45,
             'Description': 'cgminer 4.11.1',
[... 431 lines cut ...]
             'Description': 'cgminer 4.11.1',
             'Msg': 'Missing pool id parameter',
             'STATUS': 'E',
             'When': 1687360173}],
 'id': 1}
{'STATUS': [{'Code': 45,
             'Description': 'cgminer 4.11.1',
             'Msg': "Access denied to 'usbstats' command",
             'STATUS': 'E',
             'When': 1687360173}],
 'id': 1}
{'STATUS': [{'Code': 22,
             'Description': 'cgminer 4.11.1',
             'Msg': 'CGMiner versions',
             'STATUS': 'S',
             'When': 1687360173}],
 'VERSION': [{'API': '3.7', 'CGMiner': '4.11.1'}],
 'id': 1}
{'STATUS': [{'Code': 45,
             'Description': 'cgminer 4.11.1',
             'Msg': "Access denied to 'zero' command",
             'STATUS': 'E',
             'When': 1687360173}],
 'id': 1}
```

UpstreamData, 2023-06-21:
Got it, stats.  Out of curiosity, is this running some firmware other than stock?  Or is it just passing back weird data on stock?  I don't usually see anything after the model name on stock, but VNish has a similar method of setting type to this, I.E. `Antminer S17 Pro (A.P. 2.0.3)` and `Antminer S17 Pro (vnish 2.0.4)`.

gav1111, 2023-06-21:
I dont think its stock, I am not really an expert, so I took this screenshot to help
[image]

gav1111, 2023-06-21:
and 
```
Traceback (most recent call last):
  File "/home/misha/temp/pyasic/621test2.py", line 9, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/home/misha/temp/pyasic/621test2.py", line 7, in main
    print(await miner_factory._web_ping(IP))
TypeError: _web_ping() missing 1 required positional argument: 'url'
```

gav1111, 2023-06-21:
and for _web_ping its missing url

UpstreamData, 2023-06-21:
```python3
import aiohttp

from pyasic.miners.miner_factory import miner_factory
import asyncio

IP = "192.168.1.1"

async def main():
    urls = [f"http://{ip}/", f"https://{ip}/"]
    async with aiohttp.ClientSession() as session:
        tasks = [asyncio.create_task(miner_factory._web_ping(session, url)) for url in urls]
        data = await asyncio.gather(*tasks)
    print(data)
        

asyncio.run(main())
```

gav1111, 2023-06-21:
here you go
```
[('<?xml version="1.0" encoding="iso-8859-1"?>\n<!DOCTYPE html PUBLIC "-//W3C//DTD XHTML 1.0 Transitional//EN"\n         "http://www.w3.org/TR/xhtml1/DTD/xhtml1-transitional.dtd">\n<html xmlns="http://www.w3.org/1999/xhtml" xml:lang="en" lang="en">\n <head>\n  <title>401 - Unauthorized</title>\n </head>\n <body>\n  <h1>401 - Unauthorized</h1>\n </body>\n</html>\n', <ClientResponse(http://.152/) [401 Unauthorized]>
<CIMultiDictProxy('WWW-Authenticate': 'Digest realm="antMiner Configuration", nonce="97f57ddd457d696a3f947b62c7acd1e9", qop="auth"', 'Content-Type': 'text/html', 'Content-Length': '351', 'Date': 'Wed, 21 Jun 2023 19:04:24 GMT', 'Server': 'lighttpd/1.4.32')>
), (None, None)]
```

UpstreamData, 2023-06-21:
Maybe I'm wrong, that might not be VNish, it seems to identify in every way as a stock antminer.  It must just be using the same framework or something.  Jeez.  Ill see what I can do.

gav1111, 2023-06-21:
you are right its something else, I did a google image search on the logo, and found nothing
[image]

UpstreamData, 2023-06-21:
Usually they brand VNish with a white label, but this could just be outdated.  In any case, the latest commit adds a secondary check for antminers in the stats command, this should now identify as a stock antminer S17 Pro, because that seems to be what it identifies itself as for the most part.  Give it a shot, and see if those 3 miners are working for you now.

gav1111, 2023-06-21:
Yes they are identified now. Avalons are returning hashrate=0, before hashrate also was = 0 but _hashrate was also returned that wasnt zero.

UpstreamData, 2023-06-22:
Seems like I might not have fixed the per-board hashrate for avalons yet.  I'll put some work into that tomorrow.

UpstreamData, 2023-06-22:
I think it should be up and running now.  Fingers crossed.  Oh, and I was able to fix a few more of the functions in the avalonminer implementation based on that info you sent.  Appreciate it!

Let me know if it works for you and ill merge it to master.

gav1111, 2023-06-22:
still returning as strings `fan_1="3480",fan_2="3480",fan_3="3540"`

UpstreamData, 2023-06-22:
Got it.  Added some accidental quotes in the data parser.  Should be good now.

gav1111, 2023-06-22:
Ah, I see something else, an s17 is returning the following

`,fan_1=0,fan_2=0,fan_3=None,fan_4=None`

I think before, it would return -1 to keep same type

UpstreamData, 2023-06-22:
Ah yeah thats an issue, I'll get that fixed.

UpstreamData, 2023-06-22:
Better now?  Replaced some of the `get(item)` with `get(item, 0)` so it will give 0 now.

gav1111, 2023-06-22:
Yep seems to be all good. Let me know when its commited please.

closed 2023-06-22 by UpstreamData

UpstreamData, 2023-06-22:
It’s up in version 0.35.0.

UpstreamData, 2023-06-22:
It's up in v0.35.0