UpstreamData/pyasic #46: Problem when i stoped miner and then try to connect to it.

Awote, 2023-06-19:
I wrote code for test.
```async def reboot_miners(ips:list, counter:int) -> None:
    net = MinerNetwork(ips)
    pyasic_miners = await net.scan_network_for_miners()
    data = await asyncio.gather(*[item.get_data() for item in pyasic_miners])
    await pyasic_miners[0].stop_mining()```
Then i tried get miner's data 
   ```async def reboot_miners(ips:list, counter:int) -> None:
    net = MinerNetwork(ips)
    pyasic_miners = await net.scan_network_for_miners()
    data = await asyncio.gather(*[item.get_data() for item in pyasic_miners])```
 And got this warning 
 ```192.168.1.49: Ping And Get Miner Exception: name 'url' is not defined```
 After that i can't take miner's information

UpstreamData, 2023-06-19:
What type of miner is this?  I fixed this for BraiinsOS+ in the `dev-miner_data` branch, as there were a few issues with data gathering and getting miner types that were showing up related to that.

You can test this with that version, installed with `pip install pyasic@git+https://github.com/UpstreamData/pyasic@dev-miner_data`.

Additionally, I wouldn't use `scan_network_for_miners`, you probably want to do that once, then store the IPs and use `pyasic.get_miner`, or just store the miner objects.  pyasic has issues with identifying miners that are shut down, which I have been working to resolve, but its very tough since a lot of miners do weird things when shut down.

Awote, 2023-06-20:
i tried to use other branch , bit when i use miner.get_data() , i get this error
 ```File "c:\Users\aipro\OneDrive\Desktop\manager\venv\lib\site-packages\pyasic\miners\base.py", line 476, in get_data
    gathered_data = await self._get_data(allow_warning, data_to_get=data_to_get)
  File "c:\Users\sam\OneDrive\Desktop\manager\venv\lib\site-packages\pyasic\miners\base.py", line 431, in _get_data
    pools_data = await function(**args_to_send)
    data_object = await controle_object.get_data()
  File "c:\Users\sam\OneDrive\Desktop\manager\venv\lib\site-packages\pyasic\miners\base.py", line 476, in get_data
    gathered_data = await self._get_data(allow_warning, data_to_get=data_to_get)
  File "c:\Users\sam\OneDrive\Desktop\manager\venv\lib\site-packages\pyasic\miners\base.py", line 431, in _get_data
    pools_data = await function(**args_to_send)
  File "c:\Users\sam\OneDrive\Desktop\manager\venv\lib\site-packages\pyasic\miners\backends\bosminer.py", line 770, in get_pools
    pools = {"quota": group["strategy"]["quota"]}
TypeError: 'NoneType' object is not subscriptable```

Awote, 2023-06-20:
When i try use pyasic.get_miner with this miner i can take pools, config but can't take data of miner by ```get_data()``` . I get error 
```Exception has occurred: TypeError
'NoneType' object is not subscriptable
  File "C:\Users\aipro\OneDrive\Desktop\AIPROMAP_manager\core\asic_funcs.py", line 50, in getMinerData
    miner_from_task = await asyncio.gather(miner.get_data())
  File "C:\Users\aipro\OneDrive\Desktop\AIPROMAP_manager\services\queue_connecter\tags\miners\miner_add.py", line 12, in miner_add```

UpstreamData, 2023-06-20:
Try updating that branch again and see if it works.  I added a check for a `TypeError` there.

gav1111, 2023-06-23:
looks like same error that I was getting that is now fixed?

UpstreamData, 2023-06-23:
Yeah same error.  Give it a test now and see?

Awote, 2023-06-27:
I test and this fix work for me, thanks!

closed 2023-06-27 by Awote