UpstreamData/pyasic #48: script hangs when t19 bosminer hangs

gav1111, 2023-06-23:
**Describe the bug**
I am running a script every minute. On one occasion a T19 running BOS+ 23.03.2+ hung and the whole script failed to collect any data for about 15 minutes until T19 was reset. The error given by the script after about 3 minutes was:

'''
Traceback (most recent call last):
  File "/home/sf/temp/pyasic/production_test.py", line 66, in <module>
    asyncio.run(scan_and_get_data())
  File "/usr/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/home/sf/temp/pyasic/production_test.py", line 28, in scan_and_get_data
    data = await asyncio.gather(*[m.get_data(data_to_get = [
  File "/home/sf/.local/lib/python3.9/site-packages/pyasic/miners/base.py", line 476, in get_data
    gathered_data = await self._get_data(allow_warning, data_to_get=data_to_get)
  File "/home/sf/.local/lib/python3.9/site-packages/pyasic/miners/base.py", line 417, in _get_data
    if web_command_data.get("multicommand"):
AttributeError: 'NoneType' object has no attribute 'get'
```

here is the script
```
import asyncio

from pyasic.network import MinerNetwork
#from pyasic.settings import PyasicSettings

#PyasicSettings().network_ping_timeout = 30

import influxdb_client, os, time
from influxdb_client import InfluxDBClient, Point, WritePrecision
from influxdb_client.client.write_api import SYNCHRONOUS

# define asynchronous function to scan for miners
async def scan_and_get_data():
    # Define network range to be used for scanning
    # This can take a list of IPs, a constructor string, or an IP and subnet mask
    # The standard mask is /24 (x.x.x.0-255), and you can pass any IP address in the subnet
    net = MinerNetwork("172.16.1.1", mask=21)
    # Scan the network for miners
    # This function returns a list of miners of the correct type as a class
    miners: list = await net.scan_network_for_miners()

    # We can now get data from any of these miners
    # To do them all we have to create a list of tasks and gather them
    # tasks = [miner.get_data() for miner in miners]
    # Gather all tasks asynchronously and run them
    data = await asyncio.gather(*[m.get_data(data_to_get = [
                "mac",
                "model",
                "api_ver",
                "fw_ver",
                "hashrate",
                "nominal_hashrate",
                "hashboards",
                "env_temp",
                "wattage",
                "wattage_limit",
                "fans",
                "fan_psu",
                "errors",
                "fault_light",
                "pools",
            ]) for m in miners])
    data = [d.as_influxdb() for d in data]

    # Data is now a list of MinerData, and we can reference any part of that
    # Print out all data for now
    for item in data:
     print(item)

    #initialize influxdb
    token = "xxxx=="
    org = "xxxx"
    url = "http://10.10.10.1:8086"

    write_client = influxdb_client.InfluxDBClient(url=url, token=token, org=org)

    bucket="xxxx"

    write_api = write_client.write_api(write_options=SYNCHRONOUS)
    for item in data:
     write_api.write(bucket=bucket, org="xxxx", record=item)

if __name__ == "__main__":
	asyncio.run(scan_and_get_data())
```

UpstreamData, 2023-06-23:
I believe I fixed this exact issue earlier, I was having a similar issue.  Try with v0.36.0.

gav1111, 2023-06-24:
Before I wrote the message above, I did update to v.036.0 but also at the same time T19 became functioning, so I wasnt 100% if updating or T19 becoming responsive was the solution.

gav1111, 2023-06-25:
Had an issue again, seems that script was get heavily delayed, and because its ran every minute, bottlenecks and then whats written to the server gets delayed by 10-15 minutes. T19 BOS+ was causing it, here is the error
```
/home/sf/.local/lib/python3.9/site-packages/pyasic/miners/base.py:98 in _get_ssh_connection
[WARNING][06/25/23 04:54:09](root) - Connection refused: BOSMiner T19: 172.16.1.200
/home/sf/.local/lib/python3.9/site-packages/pyasic/miners/base.py:98 in _get_ssh_connection
[WARNING][06/25/23 04:54:24](root) - Connection refused: BOSMiner T19: 172.16.1.200
```

UpstreamData, 2023-06-25:
There's only a few places _get_ssh_connection can be called, one of them is get_fw_version.  Ssh has a tendency to cause issues and be pretty slow, so if you don't absolutely need fw_version it would just remove it from your list of data_to_get when calling get_data.

gav1111, 2023-06-25:
I am experiencing this right now with a another T19 or S19 BOS+ miner, the miner in the gui says `Ubus error: Command failed: Request timed out` when trying to look in through gui. I removed fw_ver from datapoints in my script, this part
```
data = await asyncio.gather(*[m.get_data(data_to_get = [
                "mac",
                "model",
                "api_ver",
                "fw_ver",
                "hashrate",
``` but script still hanging

UpstreamData, 2023-06-25:
Forgot MAC address, and it is forced to go through ssh, try removing that one?  If it's that, I'll remove references to send_ssh_command where possible and try to get graphql info on some of that, or maybe query luci or something...

gav1111, 2023-06-25:
It was s19 BOS+, it unfroze just right now, so it ll have to wait until next freeze to try the mac address. Pity. But the whole thing lasted just over 2h and although data was collected it was at irregular intervals and live ish status updates wont possible

UpstreamData, 2023-06-25:
Seems like I should just try to remove as much of the ssh command stuff as possible.

gav1111, 2023-06-25:
How does btctools get this info? It returns mac and everything and very fast too.

UpstreamData, 2023-06-25:
They're querying LUCI, see source here - https://github.com/btccom/libbtctools/blob/master/src/lua/scripts/scanner/BosHttpLuci.lua

gav1111, 2023-06-26:
Ok converting to it now, thank you

UpstreamData, 2023-06-27:
There was a bug, seems like later version of BOS+ only respond if you set BTCTools as your user agent header, it should be fixed in 0.36.2

gav1111, 2023-06-27:
ok, updated

gav1111, 2023-06-30:
t19 hung again, the every minute script stopped working because it gives the following error (t19 hang screenshotted)

```
200
[{"macaddr":"00xx"}]
200
[{"macaddr":"ecxx"}]
200
[{"macaddr":"14xx"}]
200
[{"macaddr":"a4xx"}]
200
[{"macaddr":"f4:xx"}]
Traceback (most recent call last):
  File "/home/sf/pyasic/production.py", line 65, in <module>
    asyncio.run(scan_and_get_data())
  File "/usr/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/home/sf/pyasic/production.py", line 28, in scan_and_get_data
    data = await asyncio.gather(*[m.get_data(data_to_get = [
  File "/home/sf/.local/lib/python3.9/site-packages/pyasic/miners/base.py", line 497, in get_data
    gathered_data = await self._get_data(allow_warning, data_to_get=data_to_get)
  File "/home/sf/.local/lib/python3.9/site-packages/pyasic/miners/base.py", line 416, in _get_data
    web_command_data = await self.web.multicommand(
  File "/home/sf/.local/lib/python3.9/site-packages/pyasic/web/bosminer.py", line 91, in multicommand
    data = dict(**luci_data, **gql_data)
TypeError: dict() argument after ** must be a mapping, not NoneType
```
[image]

UpstreamData, 2023-06-30:
I tried to release new versions yesterday, but working on a laptop I don't usually use, it didn't push my tags to publish the releases.  Can you try and see if it raises that error in 0.36.7?

gav1111, 2023-07-02:
Looks like didnt work, I have no data at night for about 2h, coincided with a bos+ going offline. I was sleeping so couldnt check what was happening

gav1111, 2023-07-04:
happening again, here is the error
```
Traceback (most recent call last):
  File "/home/sf/pyasic/production_print_only.py", line 62, in <module>
    asyncio.run(scan_and_get_data())
  File "/usr/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/home/sf/pyasic/production_print_only.py", line 28, in scan_and_get_data
    data = await asyncio.gather(*[m.get_data(data_to_get = [
  File "/home/sf/.local/lib/python3.9/site-packages/pyasic/miners/base.py", line 547, in get_data
    gathered_data = await self._get_data(allow_warning, data_to_get=data_to_get)
  File "/home/sf/.local/lib/python3.9/site-packages/pyasic/miners/base.py", line 500, in _get_data
    miner_data[data_name] = await function(**args_to_send)
  File "/home/sf/.local/lib/python3.9/site-packages/pyasic/miners/backends/bosminer.py", line 1041, in get_fault_light
    data = (
AttributeError: 'NoneType' object has no attribute 'strip'
```

UpstreamData, 2023-07-04:
Fixed that one in `0.36.10`.  That's ssh related, but should only pause for 10 seconds, and its happeneing when checking `fault_light` status, which you can just remove from the list.

gav1111, 2023-07-04:
ah ok I was running 0.36.9 when I got that error, updated and I ll keep monitoring

gav1111, 2023-07-06:
one t19 bos+ is hanging, script is giving the following error
```
Traceback (most recent call last):
  File "/home/sf/pyasic/production_print_only.py", line 62, in <module>
    asyncio.run(scan_and_get_data())
  File "/usr/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/home/sf/pyasic/production_print_only.py", line 49, in scan_and_get_data
    print(item)
NameError: name 'item' is not defined
```

UpstreamData, 2023-07-06:
That one appears to be in your script, not sure why it's happening but you might have a bad variable name or something.

gav1111, 2023-07-06:
you are right, the for loop was commented out when cycling through each item in the data array. ok we wait again for the next hang

closed 2023-11-29 by UpstreamData