UpstreamData/pyasic #50: return data on how long miner has been up and running and power mode its in

gav1111, 2023-06-24:
A nice feature would be to include in get_miner_data information on how long miner has been up and what power mode its in.

UpstreamData, 2023-06-24:
Run time is a good idea, something so can work on for sure.  I have a bit of a backlog of things first, like adding LuxOS support, but that would be a good thing to add to get_data.

LPM is tough (see my comments on #49), since power_limit is already in there, but for X19 and non tuning miners I think a good option would be to measure power used by LPM and then set the power_limit value to whatever that value is when it's in LPM, and the normal power level when it's not.  You can also call get_config and then compare to the X19MinerMode.

UpstreamData, 2023-06-27:
Added an uptime check for the more common models in 0.36.2, let me know if it has any issues.

gav1111, 2023-06-27:
uptime returns 0 even for new gen miners

UpstreamData, 2023-06-27:
Which type of miner?  Can you send the output of the api command stats and summary?  That's where it should show up...

gav1111, 2023-06-27:
S17 pro, S17+, Avalong, Whatsminer M30S, S19, T19, S19 Pro, S19j Pro. Here is api results for a S19 Pro

```
{'STATUS': [{'Code': 14,
             'Description': 'cgminer 1.0.0',
             'Msg': 'Invalid command',
             'STATUS': 'E',
             'When': 1687903253}],
 'id': 1}
{'STATUS': [{'Code': 14,
             'Description': 'cgminer 1.0.0',
             'Msg': 'Invalid command',
             'STATUS': 'E',
             'When': 1687903253}],
 'id': 1}
{'STATUS': [{'Code': 14,
             'Description': 'cgminer 1.0.0',
             'Msg': 'Invalid command',
             'STATUS': 'E',
             'When': 1687903253}],
 'id': 1}
{'STATUS': [{'Code': 14,
             'Description': 'cgminer 1.0.0',
             'Msg': 'Invalid command',
             'STATUS': 'E',
             'When': 1687903253}],
 'id': 1}
{'STATUS': [{'Code': 14,
             'Description': 'cgminer 1.0.0',
             'Msg': 'Invalid command',
             'STATUS': 'E',
             'When': 1687903253}],
 'id': 1}
{'STATUS': [{'Code': 14,
             'Description': 'cgminer 1.0.0',
             'Msg': 'Invalid command',
             'STATUS': 'E',
             'When': 1687903253}],
 'id': 1}
{'STATUS': [{'Code': 14,
             'Description': 'cgminer 1.0.0',
             'Msg': 'Invalid command',
             'STATUS': 'E',
             'When': 1687903253}],
 'id': 1}
{'STATUS': [{'Code': 14,
             'Description': 'cgminer 1.0.0',
             'Msg': 'Invalid command',
             'STATUS': 'E',
             'When': 1687903254}],
 'id': 1}
{'STATUS': [{'Code': 14,
             'Description': 'cgminer 1.0.0',
             'Msg': 'Invalid command',
             'STATUS': 'E',
             'When': 1687903254}],
 'id': 1}
{'STATUS': [{'Code': 14,
             'Description': 'cgminer 1.0.0',
             'Msg': 'Invalid command',
             'STATUS': 'E',
             'When': 1687903254}],
 'id': 1}
{'STATUS': [{'Code': 14,
             'Description': 'cgminer 1.0.0',
             'Msg': 'Invalid command',
             'STATUS': 'E',
             'When': 1687903254}],
 'id': 1}
{'STATUS': [{'Code': 14,
[... 442 lines cut ...]
             'When': 1687903253}],
 'id': 1}
{'STATUS': [{'Code': 14,
             'Description': 'cgminer 1.0.0',
             'Msg': 'Invalid command',
             'STATUS': 'E',
             'When': 1687903253}],
 'id': 1}
{'STATUS': [{'Code': 14,
             'Description': 'cgminer 1.0.0',
             'Msg': 'Invalid command',
             'STATUS': 'E',
             'When': 1687903253}],
 'id': 1}
{'STATUS': [{'Code': 14,
             'Description': 'cgminer 1.0.0',
             'Msg': 'Invalid command',
             'STATUS': 'E',
             'When': 1687903254}],
 'id': 1}
{'STATUS': [{'Code': 14,
             'Description': 'cgminer 1.0.0',
             'Msg': 'Invalid command',
             'STATUS': 'E',
```

UpstreamData, 2023-06-27:
Had to be at index 1 instead of 0, should be fixed in 0.36.4.

gav1111, 2023-06-28:
updated but still getting a zero:
`,model=S19,hostname=Unknown uptime=0,make="AntMiner",api_ver="0.0.0",fw_ver="Tue Dec  6 16:10:00 CST 2022",hashrate=74.63,nominal_hashrate=74.57,`

UpstreamData, 2023-06-28:
Is this using the script with the `data_to_get` list set?  If so you have to make sure to add `uptime` in there.

gav1111, 2023-06-28:
oh my bad

gav1111, 2023-06-30:
Is it possible to incorporate X19MinerMode in get_data() request?

UpstreamData, 2023-06-30:
I'm considering replacing the `is_mining` check with an enum that would have `mining`, `tuning`, `low_power`, `failed`, and `stopped`.  Then `is_mining` would check for values other than `stopped`, and you would also be able to check that mode through that attribute, maybe call it `miner_state`?

gav1111, 2023-06-30:
Yes that would be very useful I think. miner_state seems like the appropriate attribute.

UpstreamData, 2023-11-29:
Aiming to have this as part of #73 ideally, and just have it as part of the micro config.

UpstreamData, 2023-12-11:
This will be part of the config once #76 gets merged, it is a bit more to query but should work.

closed 2023-12-16 by b-rowan

UpstreamData, 2023-12-21:
#81 adds support for this directly into `MinerData`.