UpstreamData/pyasic #51: Could you add function set_static_ip in BOS.

Awote, 2023-06-27:
I see that you add function ```set_static_ip``` in antminer and bmminer backend. Could you add this function in BOS backend ?

UpstreamData, 2023-06-27:
Should be there in 0.36.3.

Awote, 2023-06-30:
I tested func for BOS and it does not work for me.
I wrote simple code for test.
```async def set_static(input_ips:str, gateway:str, subnet_mask:str, dns:str = "10.1.50.1, 8.8.8.8, 8.8.4.4"):
    miner = await pyasic.get_miner(input_ips)
    await miner.set_static_ip(miner.ip, dns, gateway, subnet_mask)
    await miner.reboot()
    print("PASS")```

```asyncio.run(set_static("10.1.50.229", "10.1.50.1", "255.255.255.0"))```
And after reboot i check BOS GUI.
![image](https://github.com/UpstreamData/pyasic/assets/37435132/b85d4a63-191a-4659-9255-cbd83cec886d)

Awote, 2023-06-30:
i checked your func and ```data``` has value 
"cat: can't open '/etc/config/network': No such file or directory\n"

UpstreamData, 2023-06-30:
Are you running a BBB X19?  Not sure if they still have LuCI in there or what.  Ill have to do some more testing to be able to do this it seems.

closed 2024-01-24 by Awote