UpstreamData/pyasic #52: `is_mining` on Braiins OS+ with S19J Pro returns `True` even not mining

Schnitzel, 2023-06-29:
**Describe the bug**
I got an S19J Pro with Braiins OS+, when i run `stop_mining()` the attribute `is_mining` always returns `True`, even though the miner actually stops mining.

**Expected behavior**
`is_mining` returns `False` when `stop_mining()` is sent

**Miner Information (If applicable):**
 - Manufacturer: Bitmain
 - Type: S19J Pro
 - Firmware Type: Braiins OS+
 - Firmware Version: `2023-03-05-0-8cbe158f-23.02-plus`

UpstreamData, 2023-06-29:
This seems to be related to some type of miners on braiins OS stopping the tuner when the pause command is sent, and others seem to not report that the tuner is not running.  I'll see what I can do.

UpstreamData, 2023-06-29:
Should be fixed in 0.36.5

closed 2023-06-29 by UpstreamData