UpstreamData/pyasic #56: TypeError when Miner Offline

Schnitzel, 2023-06-29:
**Describe the bug**
when the miner is offline (like no power) homeassistant fails with this error:

```
2023-06-29 23:12:27.357 ERROR (MainThread) [custom_components.miner.coordinator] Unexpected error fetching Antminer data: dict() argument after ** must be a mapping, not NoneType
Traceback (most recent call last):
  File "/home/vscode/.local/lib/python3.11/site-packages/homeassistant/helpers/update_coordinator.py", line 283, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/workspaces/miner/custom_components/miner/coordinator.py", line 49, in _async_update_data
    miner_data = await self.miner.get_data()
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/vscode/.local/lib/python3.11/site-packages/pyasic/miners/base.py", line 497, in get_data
    gathered_data = await self._get_data(allow_warning, data_to_get=data_to_get)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/vscode/.local/lib/python3.11/site-packages/pyasic/miners/base.py", line 416, in _get_data
    web_command_data = await self.web.multicommand(
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/vscode/.local/lib/python3.11/site-packages/pyasic/web/bosminer.py", line 91, in multicommand
    data = dict(**luci_data, **gql_data)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: dict() argument after ** must be a mapping, not NoneType
```

**Expected behavior**
`pyasic` returns an `pyasic.APIError` which can be caught by Homeassistant

**Miner Information (If applicable):**
 - Manufacturer: Bitmain
 - Type: S19J Pro
 - Firmware Type: Braiins OS+
 - Firmware Version: `2023-03-05-0-8cbe158f-23.02-plus`

UpstreamData, 2023-06-29:
I think I can fix this by checking if the data is `None` and then using an empty dict instead. I'll see what I can get done on this here shortly.

UpstreamData, 2023-06-30:
I believe I fixed this in 0.36.7.

closed 2023-06-30 by UpstreamData