UpstreamData/pyasic #57: Issues on versions > 0.36.4

Devinmara, 2023-07-06:
**Describe the bug**
I received different functionality with the same miners using 0.36.4 vs. 0.36.5.

**To Reproduce**
Steps to reproduce the behavior:
Install pyasic 0.36.5 via 
python3.8 -m pip install pyasic==0.36.5
Use pyasic in a script

Steps to reproduce the behavior:
Install pyasic 0.36.4 via 
python3.8 -m pip install pyasic==0.36.4
Use pyasic in a script

**Expected behavior**
Gives back appropriate data from the api, 0.36.5 I would assume works with more miners than older versions, however my script stops on 0.36.5 with key error, whereas 0.36.4 the same script will save to a database data relative to other miners.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 Ubuntu 18.04 using python3.8

**Additional context**
0.36.5 output:
```
/home/administrator/.local/lib/python3.8/site-packages/pyasic/miners/types/whatsminer/M5X/M56S.py:28: UserWarning: Unknown chip count for miner type M56S VH30, please open an issue on GitHub (https://github.com/UpstreamData/pyasic).
  warnings.warn(
/home/administrator/.local/lib/python3.8/site-packages/pyasic/API/bmminer.py:72 in _x19_multicommand
[WARNING][07/06/23 23:25:25](root) - BMMinerAPI: 10.1.1.145 - ([Hidden] X19 Multicommand) - API Command Error 0
Traceback (most recent call last):
  File "./main.py", line 33, in <module>
    miner_data = asyncio.run(get_miner(miner_ip))
  File "/usr/lib/python3.8/asyncio/runners.py", line 43, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.8/asyncio/base_events.py", line 608, in run_until_complete
    return future.result()
  File "./main.py", line 24, in get_miner
    miner_data = await miner.get_data()
  File "/home/administrator/.local/lib/python3.8/site-packages/pyasic/miners/base.py", line 501, in get_data
    gathered_data = await self._get_data(allow_warning, data_to_get=data_to_get)
  File "/home/administrator/.local/lib/python3.8/site-packages/pyasic/miners/base.py", line 454, in _get_data
    miner_data[data_name] = await function(**args_to_send)
  File "/home/administrator/.local/lib/python3.8/site-packages/pyasic/miners/backends/bmminer.py", line 172, in get_fw_ver
    api_version = await self.api.version()
  File "/home/administrator/.local/lib/python3.8/site-packages/pyasic/API/bmminer.py", line 86, in version
    return await self.send_command("version")
  File "/home/administrator/.local/lib/python3.8/site-packages/pyasic/API/__init__.py", line 86, in send_command
    validation = self._validate_command_output(data)
  File "/home/administrator/.local/lib/python3.8/site-packages/pyasic/API/__init__.py", line 246, in _validate_command_output
    elif data["STATUS"][0]["STATUS"] not in ("S", "I"):
KeyError: 0
```

0.36.4 output:
```
/home/administrator/.local/lib/python3.8/site-packages/pyasic/miners/types/whatsminer/M5X/M56S.py:28: UserWarning: Unknown chip count for miner type M56S VH30, please open an issue on GitHub (https://github.com/UpstreamData/pyasic).
  warnings.warn(
/home/administrator/.local/lib/python3.8/site-packages/pyasic/API/bmminer.py:72 in _x19_multicommand
[WARNING][07/06/23 23:26:52](root) - BMMinerAPI: 10.1.1.145 - ([Hidden] X19 Multicommand) - API Command Error 0
/home/administrator/.local/lib/python3.8/site-packages/pyasic/API/bmminer.py:72 in _x19_multicommand
[WARNING][07/06/23 23:26:56](root) - BMMinerAPI: 10.1.1.242 - ([Hidden] X19 Multicommand) - API Command Error 0
```

UpstreamData, 2023-07-06:
For some reason the data being returned from that specific miners API doesn't return the result status in the correct place.  I can try to handle for this, what type of miner is the one having the error (10.1.1.145)?

As an aside, `pyasic` is still very much in development, and breaking changes tend to happen a lot.  Semantic versioning information states that in 0.x.x versions, non backwards compatible API changes may also happen.  That being said, if you need it to be stable for any given environment, I highly recommend pinning your version to a specific version, such as `pyasic==0.36.10`.

Devinmara, 2023-07-07:
I have tracked down to 2 miners that are having the issues, I've been able to incorporate 0.36.10 using exception handling on the miner.get_data() function.

2 miners in question:
Name: Antminer S19j Pro (104TH)AMCB07
Firmware version: 2.2.9.543

Name: Antminer S19 XPA113D
Firmware version: 2.2.2.341

The error I'm receiving for these two miners as follows:
```
/home/administrator/.local/lib/python3.8/site-packages/pyasic/API/bmminer.py:72 in _x19_multicommand
[WARNING][07/07/23 00:08:13](root) - BMMinerAPI: 10.1.1.145 - ([Hidden] X19 Multicommand) - API Command Error 0
Exception: 0 on 10.1.1.145
/home/administrator/.local/lib/python3.8/site-packages/pyasic/miners/types/whatsminer/M5X/M56S.py:28: UserWarning: Unknown chip count for miner type M56S VH30, please open an issue on GitHub (https://github.com/UpstreamData/pyasic).
  warnings.warn(
/home/administrator/.local/lib/python3.8/site-packages/pyasic/API/bmminer.py:72 in _x19_multicommand
[WARNING][07/07/23 00:08:17](root) - BMMinerAPI: 10.1.1.242 - ([Hidden] X19 Multicommand) - API Command Error 0

```

Let me know any other debuggin information you need, this is on Ubuntu 18.04 with python3.8.0, Thanks.

Devinmara, 2023-07-07:
Here's the call stack and local variables while debugging and stuck on .145: 
[image]

UpstreamData, 2023-07-07:
Interesting.  That all looks basically correct.  Can you test a few things:
1. Does the miner type pyasic returns from `get_miner` match the real miner type?  You can call `print(await pyasic.get_miner(IP))` to check this.  I'm not certain if something is off about the AMLogic CB or possibly they're different somehow.
2. What is the result of the version command with errors off, and maybe some other commands too for reference?  These can be acquired with `print(await miner.api.send_command("version", ignore_errors=True))`.  If you could do this with the `"version"` command and the `"devdetails"` command that would help.

Devinmara, 2023-07-07:
1) Nah it's just returning {}
```
/home/administrator/.local/lib/python3.8/site-packages/pyasic/API/bmminer.py:72 in _x19_multicommand
[WARNING][07/07/23 00:49:33](root) - BMMinerAPI: 10.1.1.145 - ([Hidden] X19 Multicommand) - API Command Error 0
Exception: 0 on 10.1.1.145
{}
```

2) I'm unsure what miner object you are talking about, here is how my code looks
```
async def scan_miners():
    network = MinerNetwork("10.1.1.50")
    miners = await network.scan_network_for_miners()
    ##print(miners)
    return miners

async def get_miner(miner_ip):
    miner = await MinerFactory().get_miner(miner_ip)
    try:
        miner_data = await miner.get_data()
        return miner_data
    except Exception as ex:
        print(f"Exception: {ex} on {miner_ip}")
        return {}
    ##print(miner_data)
    

if __name__ == "__main__":
    miners = asyncio.run(scan_miners())
    miner_ips = re.findall(r'\b\d{1,3}\.\d{1,3}\.\d{1,3}\.\d{1,3}\b', format(miners))
    miners_data = []
    for miner_ip in miner_ips:
        ##print(asyncio.run(get_miner(miner_ip)))
        miner_data = asyncio.run(get_miner(miner_ip))
        if miner_data:
            miners_data.append(miner_data)
```

UpstreamData, 2023-07-07:
Here's a script that will test what I want to check:

```python3
import pyasic
import asyncio

NETWORK = pyasic.MinerNetwork("10.1.1.50")

ERR_MINERS = ["10.1.1.145", "10.1.1.242"]

async def main():
    print("[TEST]: SCAN & DATA")
    # test scanning and data
    miners = await NETWORK.scan_network_for_miners()
    await asyncio.gather(*[get_miner_data(m) for m in miners])
    await asyncio.gather(*[devdetails_and_version(m) for m in miners])

    print("[TEST]: DEVDETAILS & VERSION")
    # test erroring miners
    miners = await asyncio.gather(*[pyasic.get_miner(ip) for ip in ERR_MINERS])
    await asyncio.gather(*[get_miner_data(m) for m in miners])
    await asyncio.gather(*[devdetails_and_version(m) for m in miners])

async def get_miner_data(m: pyasic.AnyMiner):
    try:
        print(m, "[SUCCESS]:", await m.get_data())
    except Exception as e:
        print(m, "[ERROR]:", e)

async def devdetails_and_version(m: pyasic.AnyMiner):
    try:
        tasks = [m.api.send_command(cmd, ignore_errors=True) for cmd in ["devdetails", "version"]]
        print(m, "[SUCCESS]:", await asyncio.gather(*tasks))
    except Exception as e:
        print(m, "[ERROR]:", e)

if __name__ == '__main__':
    asyncio.run(main())
```

Devinmara, 2023-07-07:
Apologies for the issue, we are utilizing a proprietary firmware that doesn't match the API call, closing issue.

UpstreamData, 2023-07-07:
That's no problem.  I can work on adding support for it if you want, but we would likely need to work more closely on gathering information.

You are also welcome to extend pyasic if you prefer, just create a class that inherits from `BaseMiner`.

Devinmara, 2023-07-07:
Thank you @UpstreamData for the information, I believe we are going to pull our own version of Pyasic and incorporate our API into it, I greatly appreciate you giving me a leg up by recommending inheriting from BaseMiner.

UpstreamData, 2023-07-07:
No problem.  Feel free to hit me up if you have any questions or need help!

Devinmara, 2023-08-30:
Hey this commit ended up solving my issues with this request/issue:
https://github.com/UpstreamData/pyasic/compare/master...Devinmara:pyasic_ml:master
From testing it appears to work with other miners still as well.

UpstreamData, 2023-08-30:
Re-opening.

Looked at the code, it looks good, but I think it should be -
`if isinstance(data["STATUS"], dict):`

I'll see if I can implement this tonight.

Devinmara, 2023-08-30:
Sounds good man, i've been making modifications in my own branch but would love for this to be upstream so my codebase doesn't get as messy I can just keep updated pyasic :) thanks for your hard work!

UpstreamData, 2023-08-30:
> Hey this commit ended up solving my issues with this request/issue: [master...Devinmara:pyasic_ml:master](https://github.com/UpstreamData/pyasic/compare/master...Devinmara:pyasic_ml:master) From testing it appears to work with other miners still as well.

Done.  Pushing out version 0.38.1 with the changes.

Devinmara, 2023-08-31:
working as expected.

closed 2023-08-31 by Devinmara