UpstreamData/pyasic #60: Antminer S19 XP REST API

Devinmara, 2023-08-30:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
I'd like to be able to poll a miner I have that responds with Antminer S19 XP, however I believe because it's replying in a REST format pyasic is unable to handle it.
**Describe the solution you'd like**
A clear and concise description of what you want to happen.
I'm curious if there may be a REST solution created somewhere in the code I'm not seeing at the moment in order to implement the change.
**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
I've thought about creating a new class of miner, but I hope this miner can be captured with functionality already built into pyasic.
**Additional context**
Add any other context or screenshots about the feature request here.
any/all suggestions would be greatly appreciated! Thank you.

UpstreamData, 2023-08-30:
I'm slightly confused about this.  Can you try the `miner.web` attribute and send some restuls of commands from there?  Should be an instance of `AntminerModernWebAPI`, here in the code - https://github.com/UpstreamData/pyasic/blob/8c5503d0028e2d7f638c86236db8e0b869398883/pyasic/web/antminer.py#L26

Devinmara, 2023-08-31:
It doesn't appear to get picked up by my initial scan_network_for_miners(ip-ip2)

The miner is using this API:
https://app.swaggerhub.com/apis/ePIC-Blockchain/umc_api/0.30.6

UpstreamData, 2023-08-31:
Ah ok.  You have an ePIC board.  Ill see what I can do.

Devinmara, 2023-08-31:
Any possibility you'd have some hints/recommendations on where to start it? I wouldn't mind taking a crack at it!

UpstreamData, 2023-08-31:
Sure.  It would be much easier for you since you actually have one.  

You will have to do a few things - 
1. Create a new backend type (pyasic/miners/backends/epic.py or similar), with a new backend class.
2. Find a way of identifying the miner as that "backend" via the functionality and enum defined in the miner_factory (pyasic/miners/miner_factory.py)
3. Create subclasses of miner types (physical types) and the new backend you created in the corresponding files.
4. Add those subtypes to the MINER_CLASSES dict in pyasic/miners/miner_factory.py

Let me know if you have any questions, if you have telegram you can DM me there, username is b_rowan

UpstreamData, 2023-08-31:
Additionally, the backend type must inherit from BaseMiner, and implement the abstract methods shown in the docs [here](https://pyasic.readthedocs.io/en/latest/miners/base_miner/).

You will also likely want to implement a new web type in pyasic/web, which will handle all the commands from the schema as needed, you can look at any of the other web API types for this.

jpcomps, 2023-11-29:
This can hopefully be closed now. @Devinmara know its been awhile, but curious if the latest version works for you

UpstreamData, 2023-12-11:
Going to close this now, I think its extremely safe to assume it works.

closed 2023-12-11 by UpstreamData