UpstreamData/pyasic #63: I can't rewrite BosMiner pools

Awote, 2023-09-22:
I can't rewrite pools of BOSMiner S19j Pro.

```async def re_config_pool(ip: str, pools_info) -> bool:
    # Get the miner object associated with the given IP address
    miner = await pyasic.get_miner(ip)

    # Check if the miner is accessible and get its configuration
    if miner.api_type is not None:
        ConfigMiner = await miner.get_config()
        new_config_miner = rewrite_pool_config_by_worker(ConfigMiner, pools_info)

        # Send the updated configuration to the miner and reboot it
        await miner.send_config(new_config_miner)
        # if hasattr(miner, "restart_bosminer"):
        #     rez = await miner.restart_bosminer()
        #if hasattr(miner, "restart_backend"):
        #    rez = await miner.restart_backend()
        #else:
        #   rez = await miner.reboot()

        # Return True indicating success
    else:
        # Return False if the miner is not accessible
        return False```

After this code i get next data:
url:stratum+tcp://eu.stratum.braiins.com:3
User:braiinstest.dummy_miner
Status: Alive 
and etc.

UpstreamData, 2023-09-22:
Please try this updated code with the latest version of pyasic, let me know if it changes anything.   It's possible this is a BOS+ issue that isn't applying the config being sent over SSH.

```python3
from __future__ import annotations

import asyncio

from pyasic.config import _Pool, _PoolGroup, MinerConfig
import pyasic

async def update_miner_pools(ip: str, pools_info: list[dict]):
    # Get the miner object associated with the given IP address
    miner = await pyasic.get_miner(ip)

    # Check if the miner is accessible and get its configuration
    if miner is not None:
        miner_conf = await miner.get_config()
        new_config_miner = rewrite_pool_config_by_worker(miner_conf, pools_info)

        # Send the updated configuration to the miner and reboot it
        await miner.send_config(new_config_miner)
        
        # If you want to append the last octet of the IP to the user you can use user_suffix:
        # suffix = ip.split(".")[-1]
        # await miner.send_config(new_config_miner, user_suffix=suffix)

        return True
    else:
        return False

def rewrite_pool_config_by_worker(config: MinerConfig, pools_info: list[dict]):
    config.pool_groups = [
        _PoolGroup(
            pools=[
                _Pool(
                    url=pool.get("url", ""),
                    username=pool.get("user", ""),
                    password=pool.get("pwd", ""),
                )
                for pool in pools_info
            ]
        )
    ]
    return config

if __name__ == "__main__":
    asyncio.run(
        update_miner_pools(
            "192.168.1.100",
            [
                {
                    "url": "stratum+tcp://pool.url.real:3333",
                    "user": "pool_username.worker_name",
                    "pwd": "123",
                }
            ],
        )
    )
```

Awote, 2023-09-23:
I wrote functions like in your example 
```def rewrite_pool_config_by_worker(pools_config: MinerConfig, value_update: list[dict]):
    value_update = sorted(value_update, key=lambda x:x["number"])
    rewrite_pools = []
    for item in value_update:
        rewrite_pools.append(create_pool_structure(username=item["worker"], url=check_url_correct(item["url"]), password=item["password"]))
    pools_config.pool_groups[0].pools = rewrite_pools
    return pools_config``` 
 ```def create_pool_structure(username: str, url: str, password: str):
    return _Pool(username=username, url=url, password=password)```
I don't need to create new pool group so i only put new pools in pool group, but i got same pool info im miner GUI
url:stratum+tcp://eu.stratum.braiins.com:3
User:braiinstest.dummy_miner
Status: Alive 
and etc.

UpstreamData, 2023-09-23:
Ok, it seems possible that the bosminer.toml config is not being applied properly.  My guess is that with the new GRPC api they changed something about how the configuration is applied, and it may not be as simple as just changing the file out.

Can you confirm the firmware version running on this miner?  I'm assuming something newer than 23.01?

Awote, 2023-09-23:
Yes this is Version: 2023-04-20-0-0ce150e9-23.03-plus

Awote, 2023-09-23:
if i can help you fix this problem, i 'll try.

UpstreamData, 2023-09-23:
Alright.  I have no way to test this, I have a BCB on the way, but no miner to test it with.  I will work on trying to implement GRPC on a theoretical basis, and possibly test with the BCB if it works on its own, but it's likely there is no good solution to this until then sadly, unless there is something in the graphQL api that will work.

Awote, 2023-09-23:
This is what BOS + send from GUI to miner. First request 
```mutation ($groupID: ID!, $poolID: ID!, $enabled: Boolean!, $url: String!, $user: String!, $password: String!) {
  bosminer {
    config {
      group(id: $groupID) {
        ... on AttributeError {
          message
          __typename
        }
        ... on GroupConfigurator {
          pool(id: $poolID) {
            ... on PoolConfiguratorError {
              message
              __typename
            }
            ... on PoolConfigurator {
              update(enabled: $enabled, url: $url, user: $user, password: $password) {
                ... on PoolConfig {
                  ...PoolConfig
                  __typename
                }
                ... on PoolRecordError {
                  message
                  invalidUrl
                  invalidUser
                  __typename
                }
                __typename
              }
              __typename
            }
            __typename
          }
          __typename
        }
        __typename
      }
      __typename
    }
    __typename
  }
}

fragment PoolConfig on PoolConfig {
  __typename
  id
  enabled
  url
  user
  password
}
```

Awote, 2023-09-23:
This is second request 
```
{
  bosminer {
    config {
      ... on Error {
        message
        __typename
      }
      ... on BosminerConfig {
        groups {
          ...GroupConfig
          __typename
        }
        autotuning {
          ...Autotuning
          __typename
        }
        hashChainGlobal {
          ...HashChainGlobal
          __typename
        }
        hashChains {
          ...HashChains
          __typename
        }
        performanceScaling {
          ...PerformanceScaling
          __typename
        }
        tempControl {
          ...TempControl
          __typename
        }
        fanControl {
          ...FanControl
          __typename
        }
        __typename
      }
      __typename
    }
    __typename
  }
}

fragment GroupConfig on GroupConfig {
  __typename
  id
  name
  pools {
    ...PoolConfig
    __typename
  }
  strategy {
    __typename
    ... on QuotaStrategy {
      quota
      __typename
    }
    ... on FixedShareRatioStrategy {
      fixedShareRatio
      __typename
    }
  }
}

fragment PoolConfig on PoolConfig {
  __typename
  id
  enabled
  url
  user
  password
}

fragment Autotuning on Autotuning {
  __typename
  enabled
  mode
  powerTarget
  hashrateTarget
}

fragment PerformanceScaling on PerformanceScaling {
  __typename
  enabled
  powerStep
  shutdownEnabled
  shutdownDuration
  hashrateStep
  minPowerTarget
  minHashrateTarget
}

fragment HashChainGlobal on HashChainGlobal {
  __typename
  voltage
  frequency
  asicBoost
}

fragment HashChains on HashChain {
  __typename
  name
  enabled
  voltage
  frequency
}

fragment TempControl on TempControl {
  __typename
  mode
  targetTemp
  hotTemp
  dangerousTemp
}

fragment FanControl on FanControl {
  __typename
  speed
  minFans
  immersionModeEnabled
}
```

Awote, 2023-09-27:
if you need , i can remote give you miner with Bos 23.03.plus.
I need your tg for contact

UpstreamData, 2023-09-27:
TG is b_rowan.

I've been super busy this week, so I may not be super active, but I'll try to get around to it as soon as I have a chance.

closed 2023-11-28 by Awote