UpstreamData/pyasic #65: One miner don't give data when i use miner.get_data() and await about 200+ seconds

Awote, 2023-10-02:
I get this error

``Traceback (most recent call last):
  File "c:\Users\aipro\OneDrive\Desktop\SAM\core\asic_funcs.py", line 183, in get_miner_data 
    data_object = await controle_object.get_data()
  File "c:\Users\aipro\OneDrive\Desktop\SAM\venv\lib\site-packages\pyasic\miners\base.py", line 543, in get_data
    gathered_data = await self._get_data(
  File "c:\Users\aipro\OneDrive\Desktop\SAM\venv\lib\site-packages\pyasic\miners\base.py", line 460, in _get_data
    api_command_data = await api_command_task
  File "c:\Users\aipro\OneDrive\Desktop\SAM\venv\lib\site-packages\pyasic\API\bmminer.py", line 51, in multicommand
    data = await self.send_command(command, allow_warning=allow_warning)
  File "c:\Users\aipro\OneDrive\Desktop\SAM\venv\lib\site-packages\pyasic\API\__init__.py", line 76, in send_command
    data = await self._send_bytes(json.dumps(cmd).encode("utf-8"))
  File "c:\Users\aipro\OneDrive\Desktop\SAM\venv\lib\site-packages\pyasic\API\__init__.py", line 209, in _send_bytes
    ret_data = await asyncio.wait_for(reader.read(4096), timeout=timeout)
  File "C:\Users\aipro\AppData\Local\Programs\Python\Python310\lib\asyncio\tasks.py", line 445, in wait_for
    return fut.result()
  File "C:\Users\aipro\AppData\Local\Programs\Python\Python310\lib\asyncio\streams.py", line 668, in read 
    await self._wait_for_data('read')
  File "C:\Users\aipro\AppData\Local\Programs\Python\Python310\lib\asyncio\streams.py", line 501, in _wait_for_data
    await self._waiter
  File "C:\Users\aipro\AppData\Local\Programs\Python\Python310\lib\asyncio\proactor_events.py", line 282, in _loop_reading
    length = fut.result()
  File "C:\Users\aipro\AppData\Local\Programs\Python\Python310\lib\asyncio\windows_events.py", line 817, in _poll
    value = callback(transferred, key, ov)
  File "C:\Users\aipro\AppData\Local\Programs\Python\Python310\lib\asyncio\windows_events.py", line 485, in finish_recv
    return ov.getresult()
OSError: [WinError 121] The semaphore timeout period has expired``

I tested it in AntMiner S19j Pro with firmware BMMiner

UpstreamData, 2023-10-02:
Seems like the miner isn't responding on the API port.  Could you try simplifying the command calls you're doing, and see if possibly it's just as issue with the miner not responding?

```python3
import pyasic
import asyncio

IP = "192.168.1.10"

async def main():
    miner = await pyasic.get_miner(IP)
    res = await miner.api.devs()
    print(res)

asyncio.run(main())
```

If this hangs forever it's likely the miner isn't responding on port 4028.

Awote, 2023-10-15:
i finally tested this and got that this miner doesn't work (it can't give me GUI by miner's ip) but i have a question. Why timeout is too large , i mean you have in settings timeout for 3-5 second , but ithis miner has timeout aboute 20-3 seconds