UpstreamData/pyasic #66: V9 integration

epper202, 2023-10-12:
hi!
Is it possible to add the antMiner V9 to the supported devices?

Much appreciated thank you!

BR,
Egbert

UpstreamData, 2023-10-12:
Hey Egbert, the V9 is not technically supported, but it should be possible to add support for it, and it's very possible it already has some basic functionality.  You can test with the `pyasic.get_miner()` function, and see if it returns `None` or `UnknownMiner`, as those would mean it is wholly unsupported.  If it returns a base type (such as `BMMiner`), then I should just be able to create an information type for it and have it work.

epper202, 2023-10-14:
[image]

Got it to work, didn't change anything actually. However few details are missing and therefor no active power controllability :(

UpstreamData, 2023-10-14:
Ok, so the framework is there, but it's not reporting a first board.  There's very likely no way to do power control for it, as my bet is that this miner has no way of handling power control and won't report a wattage.  Most stock firmware miners won't allow you to do that.

I would like to get the model and some general miner information in pyasic, so nominal information can be calculated in case anyone wants to use a more full functionality set than the hass-miner integration provides, so if you are able to, could you send the results of the `stats`, `devs`, `devdetails`, and `version` RPC API commands?

UpstreamData, 2023-10-14:
Do you have access to a computer with python on it and know how to use it?  If so I'll send you a script that can do it with pyasic.

If not, I assume you're on windows, I will have to do some research on how to do the commands on windows, but I can send you a script for that too.

UpstreamData, 2023-10-14:
Done.  Here is a powershell script that will get the data I need.

To use this, make a new text file, paste this data in, and change the "server" variable at the top to match the IP of your miner.  Once that's done, change the file extension of the file to .ps1 (if you cant see this, its in file explore top menu, view, then a checkbox that says file name extensions, then you can rename the file).  Right click, select "Run with powershell", then send the result in a text file here.

```powershell
$server = "192.168.1.5"
$port = 4028
$commands = @("devs", "stats", "devdetails", "version")

try {
    foreach ($cmd in $commands) {
        $client = New-Object System.Net.Sockets.TcpClient
        $client.Connect($server, $port)

        $stream = $client.GetStream()

        $command = '{"command": "' + $cmd + '"}'
        $writer = [System.Text.Encoding]::UTF8.GetBytes($command)

        $stream.Write($writer, 0, $writer.Length)
        $stream.Flush()

        # Read response
        $reader = New-Object System.IO.StreamReader($stream)
        $response = $reader.ReadToEnd()
        $reader.Close()

        # Display the response
        Write-Host "Response for command '$cmd':"
        Write-Host "-------------------------------------------------------------------------------------"
        Write-Host $response
        Write-Host ""

        # Clear the stream for the next command
        $stream.Flush()
        # Close the connection
        $client.Close()
    }

    # Wait for a key press
    Read-Host "Press Enter to exit..."
}
catch {
    Write-Host "An error occurred: $_"
    Read-Host "Press Enter to exit..."
}
```

epper202, 2023-10-14:
Will do first thing in the morning!

epper202, 2023-10-15:
Well I ran it but the powershell closes after a few seconds without having any text displayed.

UpstreamData, 2023-10-15:
Ok, I had it working on my system, so it should work.  Make sure you replace the IP at the top of the file with your miners IP?  Then try running the file via any command line, just copy the file location, add a / to the end, then the file name, and see what it prints out.

epper202, 2023-11-05:
OK so 3 weeks in and tried multiple Home assistant integrations for the miner.
I tried your forked one, 0.6.
tried schnitzel 0.6 and 1.0.0.0 beta but gave only trouble. tried to downgrade in home assistant core and worked but got only 1 miner active. Then I read a topic with miners in parallel but now I just try to install the pyasic.

I am on Ubuntu but not experienced just yet. 
Tried to install pyasic through command line pip install pyasic.
installed pip first then it threw me "error: externally managed environment" hint: see PEP 668
`error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
`
PEP 668 is way too complicated for me right now so I dont understand it.

tried to install pyasic through pipx because the help text says so but ran into more errors.

`pipx install pyasic
Note: Dependent package 'rsa' contains 6 apps
  - pyrsa-decrypt
  - pyrsa-encrypt
  - pyrsa-keygen
  - pyrsa-priv2pub
  - pyrsa-sign
  - pyrsa-verify
Note: Dependent package 'charset-normalizer' contains 1 apps
  - normalizer
Note: Dependent package 'httpx' contains 1 apps
  - httpx
Note: Dependent package 'pyaml' contains 1 apps
  - pyaml

No apps associated with package pyasic. Try again with '--include-deps' to
include apps of dependent packages, which are listed above. If you are
attempting to install a library, pipx should not be used. Consider using pip
or a similar tool instead.
`

UpstreamData, 2023-11-05:
I would start with trying a virtual environment.  It shows in the error message from pop how to do this (sort of).

In the directory you want to create the venv, run `python3 -m venv path/to/venv` or `python3 -m venv ./venv`.  This will create a virtual environment in your folder inside another folder called venv.  From there you can run `./venv/Scripts/activate` to activate the virtual environment, then run `pip install pyasic` and it should load pyasic into the environment.

If you want to run any scripts that use pyasic, you have to make sure you activate the VENV first, as it is storing the library.

[ Read more here.](https://docs.python.org/3/library/venv.html)

epper202, 2023-11-05:
> I would start with trying a virtual environment. It shows in the error message from pip how to do this (sort of).
> 
> In the directory you want to create the venv, run `python3 -m venv path/to/venv` or `python3 -m venv ./venv`. This will create a virtual environment in your folder inside another folder called venv. From there you can run `./venv/Scripts/activate` to activate the virtual environment, then run `pip install pyasic` and it should load pyasic into the environment.
> 
> If you want to run any scripts that use pyasic, you have to make sure you activate the VENV first, as it is storing the library.
> 
> [ Read more here.](https://docs.python.org/3/library/venv.html)

And I was still investigating why the heck my quotes went bad.. Thanks, try it directly

UpstreamData, 2023-11-05:
> And I was still investigating why the heck my quotes went bad.. Thanks, try it directly

Try triple ` on each end.  You can also specify a language with ```python

epper202, 2023-11-05:
Thanks. So now I have installed pyasic in here:
`path/to/venv/bin/pip install pyasic`

UpstreamData, 2023-11-05:
Yep, you should be able to activate it and use pip/python as commands normally.

epper202, 2023-11-05:
Yes both of them work, not really sure how the activate works since it throws me no such file or directory when i try your command

epper202, 2023-11-05:
Ah it works. Source path/to/venv/bin/activate
what am i supposed to do now?

UpstreamData, 2023-11-05:
Now you should be able to create a python script and run it using python (`python3` when the venv is activated is linked to your venv python, so you can just run commands as you normally would)

epper202, 2023-11-05:
Many thanks so far!
I need to get started with Python so first things first.
I've never ran this before also no scripts

UpstreamData, 2023-11-05:
You should be able to make a new python file with `nano script_name.py`.  From there I would start with the examples in the README.md file at the root of this project, those shoupd be copy/pastable directly out of there into your script, and you should be able to save and run the file with `python3 script_name.py`.

epper202, 2023-11-05:
Yehesss it works, many thanks!
Now I need to proceed and do things with this information

epper202, 2023-11-05:
> You should be able to make a new python file with `nano script_name.py`. From there I would start with the examples in the README.md file at the root of this project, those should be copy/pastable directly out of there into your script, and you should be able to save and run the file with `python3 script_name.py`.

Do you know why this happens when integrating v1.0.0.beta in home assistant?:
```Unable to install package pyasic==0.36.9: ERROR: Cannot install pyasic==0.36.9 because these package versions have conflicting dependencies. ERROR: ResolutionImpossible: for help visit https://pip.pypa.io/en/latest/topics/dependency-resolution/#dealing-with-dependency-conflicts```

UpstreamData, 2023-11-05:
I recently updated the fork of HASS-miner to use the latest version, you may want to try that?

epper202, 2023-11-05:
Ok! added : ```https://github.com/UpstreamData/hass-miner``` in custom repo and installed, restarted. but then:

> Config flow could not be loaded: 500 Internal Server Error Server got itself in trouble
LOG:
> Unable to install package pyasic==0.40.0: ERROR: Cannot install pyasic==0.40.0 because these package versions have conflicting dependencies. ERROR: ResolutionImpossible: for help visit https://pip.pypa.io/en/latest/topics/dependency-resolution/#dealing-with-dependency-conflicts

>

Logger: aiohttp.server
Source: /usr/local/lib/python3.11/site-packages/aiohttp/web_protocol.py:403
First occurred: 9:59:56 PM (1 occurrences)
Last logged: 9:59:56 PM
Error handling request

Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/aiohttp/web_protocol.py", line 433, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/aiohttp/web_app.py", line 504, in _handle
    resp = await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 85, in security_filter_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 100, in forwarded_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 28, in request_context_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 80, in ban_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 236, in auth_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/headers.py", line 31, in headers_middleware
    response = await handler(request)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/view.py", line 148, in handle
    result = await handler(request, **request.match_info)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/decorators.py", line 63, in with_admin
    return await func(self, request, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 147, in post
    return await super().post(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 72, in wrapper
    result = await method(view, request, data, *args, **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 71, in post
    result = await self._flow_mgr.async_init(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 880, in async_init
    flow, result = await task
                   ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 898, in _async_init
    flow = await self.async_create_flow(handler, context=context, data=data)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 1000, in async_create_flow
    handler = await _async_get_flow_handler(
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2096, in _async_get_flow_handler
    await _load_integration(hass, domain, hass_config)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2072, in _load_integration
    await async_process_deps_reqs(hass, hass_config, integration)
  File "/usr/src/homeassistant/homeassistant/setup.py", line 420, in async_process_deps_reqs
    await requirements.async_get_integration_with_requirements(
  File "/usr/src/homeassistant/homeassistant/requirements.py", line 52, in async_get_integration_with_requirements
    return await manager.async_get_integration_with_requirements(domain)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/requirements.py", line 168, in async_get_integration_with_requirements
    await self._async_process_integration(integration, done)
  File "/usr/src/homeassistant/homeassistant/requirements.py", line 183, in _async_process_integration
    await self.async_process_requirements(
  File "/usr/src/homeassistant/homeassistant/requirements.py", line 248, in async_process_requirements
    await self._async_process_requirements(name, missing)
  File "/usr/src/homeassistant/homeassistant/requirements.py", line 286, in _async_process_requirements
    raise RequirementsNotFound(name, list(failures))
homeassistant.requirements.RequirementsNotFound: Requirements for miner not found: ['pyasic==0.40.0'].

So these are the error messages from the HASS log

epper202, 2023-11-06:
Changed pyasic version in code below
WAS:   "requirements": ["pyasic==0.40.0"],
IS:   "requirements": ["pyasic=<0.40.0"],

and solved the first error.
2nd remains with the long list

```{
  "domain": "miner",
  "name": "Miner",
  "codeowners": ["@Schnitzel"],
  "config_flow": true,
  "documentation": "https://github.com/Schnitzel/hass-miner",
  "homekit": {},
  "iot_class": "local_polling",
  "issue_tracker": "https://github.com/Schnitzel/hass-miner/issues",
  "requirements": ["pyasic==0.40.0"],
  "ssdp": [],
  "version": "0.7.0",
  "zeroconf": []
}```

epper202, 2023-11-06:
I actually removed everything before installing the forked version, too bad it doesnt work still. fighting it for too long without any luck

epper202, 2023-11-09:
> My assumption here is you had both of them installed and they were conflicting with one another maybe? In any case this will likely work better.

Removed everything restarted hass 2x 
installed your repo again
directly: 

> Config flow could not be loaded: 500 Internal Server Error Server got itself in trouble

UpstreamData, 2023-11-09:
Ok, so something is definitely not working.  I'll do some testing later today and see what I can find out.

I did just fix a really obscure Linux bug in 0.40.1 that happens on scan, I wonder if that could be the issue.  Are you able to send the logs for the integration?

epper202, 2023-11-09:
Of course! Give me 30 min.

epper202, 2023-11-09:
First: 
Version check: b3b2869
Repo installed: https://github.com/UpstreamData/hass-miner 

Starting integration:
Fault log:

Logger: aiohttp.server
Source: /usr/local/lib/python3.11/site-packages/aiohttp/web_protocol.py:403
First occurred: 3:12:34 PM (1 occurrences)
Last logged: 3:12:34 PM

Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/aiohttp/web_protocol.py", line 433, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/aiohttp/web_app.py", line 504, in _handle
    resp = await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 85, in security_filter_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 100, in forwarded_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 28, in request_context_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 80, in ban_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 236, in auth_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/headers.py", line 31, in headers_middleware
    response = await handler(request)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/view.py", line 148, in handle
    result = await handler(request, **request.match_info)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/decorators.py", line 63, in with_admin
    return await func(self, request, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 147, in post
    return await super().post(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 72, in wrapper
    result = await method(view, request, data, *args, **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 71, in post
    result = await self._flow_mgr.async_init(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 880, in async_init
    flow, result = await task
                   ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 898, in _async_init
    flow = await self.async_create_flow(handler, context=context, data=data)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 1000, in async_create_flow
    handler = await _async_get_flow_handler(
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2096, in _async_get_flow_handler
    await _load_integration(hass, domain, hass_config)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2072, in _load_integration
    await async_process_deps_reqs(hass, hass_config, integration)
  File "/usr/src/homeassistant/homeassistant/setup.py", line 420, in async_process_deps_reqs
    await requirements.async_get_integration_with_requirements(
  File "/usr/src/homeassistant/homeassistant/requirements.py", line 52, in async_get_integration_with_requirements
    return await manager.async_get_integration_with_requirements(domain)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/requirements.py", line 168, in async_get_integration_with_requirements
    await self._async_process_integration(integration, done)
  File "/usr/src/homeassistant/homeassistant/requirements.py", line 183, in _async_process_integration
    await self.async_process_requirements(
  File "/usr/src/homeassistant/homeassistant/requirements.py", line 242, in async_process_requirements
    self._raise_for_failed_requirements(name, missing)
  File "/usr/src/homeassistant/homeassistant/requirements.py", line 271, in _raise_for_failed_requirements
    raise RequirementsNotFound(integration, [req])
homeassistant.requirements.RequirementsNotFound: Requirements for miner not found: ['pyasic==0.40.0'].

epper202, 2023-11-09:
And: 

Logger: homeassistant.util.package
Source: util/package.py:102
First occurred: 3:15:37 PM (3 occurrences)
Last logged: 3:15:44 PM

Unable to install package pyasic==0.40.0: ERROR: Cannot install pyasic==0.40.0 because these package versions have conflicting dependencies. ERROR: ResolutionImpossible: for help visit https://pip.pypa.io/en/latest/topics/dependency-resolution/#dealing-with-dependency-conflicts

UpstreamData, 2023-11-09:
Still failing to install.  Not sure what exactly is conflicting?  I'll have to test on a fresh HASS instance and find out.

epper202, 2023-11-09:
So I ran a fresh hass in a virtualbox.
Got confused which repo I needed haha, on schnitzell's page it's referring to your repo and you are referring to his repo :').
I've installed your's, tried integration and it went fine!
Ran it on:
>  Home Assistant 2023.10.5 Supervisor 2023.11.0 Operating System 11.1 Frontend 20231005.0 - latest 

Now I did core update to:

>  Core 2023.11.1 Supervisor 2023.11.0 Operating System 11.1 Frontend 20231030.1

Tried to integrate.. boom, 

> Config flow could not be loaded: 500 Internal Server Error Server got itself in trouble

so it has to do with a core update (?)

UpstreamData, 2023-11-09:
> so it has to do with a core update (?)

Seems like it, yes.  Like I said, I'll have to do some testing, maybe it'll be a simple fix.

epper202, 2023-11-10:
OK! I just want to deliver as much as data as possible

b-rowan, 2023-12-23:
I finally figured this out.  It's an issue with dependencies on home assistant being incompatible with pyasic, specifically httpx.  Thats why the message is so cryptic, pyasic is incompatible with the homeassistant package.  For now, there isn't anything I can do until they update to httpx>0.25.2, but theyre going to 0.26.0 in the next update, so this will likely work in the new year.

Please see my PR on the hass-miner repo for more info and to watch status - https://github.com/Schnitzel/hass-miner/pull/252

closed 2024-01-26 by UpstreamData