UpstreamData/pyasic #69: IceRiver KS2

tomasbedrich, 2023-11-22:
Hello, is it possible to add support for IceRiver KS2 please?

I can provide some cooperation if necessary.

UpstreamData, 2023-11-22:
Absolutely.  Reach out to me on Telegram (b_rowan) and we can figure something out.  I haven't even seen an iceriver miner yet, so I don't know what theyre capable of or what will actually be required to support them, but if they implement a RPC api like most miners it should be no issues.

schwoi, 2023-12-18:
@UpstreamData I have started some investigation with the KS3M as well and have some of the base calls and responses if that would be some help.

UpstreamData, 2023-12-18:
> @UpstreamData I have started some investigation with the KS3M as well and have some of the base calls and responses if that would be some help.

Sure, that would be awesome.  We had a brief conversation about it, but never really got anywhere with it.

schwoi, 2023-12-27:
@UpstreamData I have put together an initial mock API from the KS3M. It isn't 100% complete but a start.
https://github.com/schwoi/ks3m-mockapi
There is also a postman collection in there as well and If this isn't of help let me know and we can work out a better way to get the details you need.

Also please excuse my python. It is not something I use so there was a fair amount of chatgpt and google involved to get the basics.

UpstreamData, 2023-12-27:
Looks good. Thank you!

This will help me be able to start on a framework for them, but the main issue right now is going to be identifying them as "IceRiver" miners.  I'll have to take a look back through our DMs and try to re-implement something, then I can make a branch for testing.

b-rowan, 2024-08-20:
Should be in a pretty good state here - https://github.com/UpstreamData/pyasic/tree/dev_iceriver