UpstreamData/pyasic #78: Support of Antminer S19k Pro NoPic

kamordi, 2023-12-16:
Would love to see the Antminer S19k Pro NoPic suported with the SW.  

Return of `await pyasic.miners.miner_factory._get_miner_type(IP)` 

MinerTypes.BRAIINS_OS

b-rowan, 2023-12-16:
Yeah, can do.  Can you confirm a few things for me?

I need the model string (the result of `miner_factory.get_miner_model_braiins_os(ip)`, and the chip count per board (`miner.get_hashboards` will tell you)

kamordi, 2023-12-16:
`miner_factory.get_miner_model_braiins_os(ip) ` -> Antminer S19K Pro NoPic

HashBoard(slot=0, hashrate=None, temp=None, chip_temp=None, chips=None, expected_chips=0, missing=True), HashBoard(slot=1, hashrate=27.48, temp=45, chip_temp=60, chips=77, expected_chips=0, missing=False), HashBoard(slot=2, hashrate=None, temp=None, chip_temp=None, chips=None, expected_chips=0, missing=True)

Each Board has 77 chips. Two Boards are currently disabled. Would it be also possible to en- and disable hashboards via pyasic?

b-rowan, 2023-12-16:
I believe it's possible, I think through BOS+s GRPC API, but I've never tried.  This will work on BOS+ only, so maybe make a check for `isinstance(miner, BOSMiner)` and `miner.web.grpc is not None`, but the command is `miner.web.grpc. enable_hashboards(["1", "2", "3"])` where 1, 2, and 3 are the IDs of the hashboards you want to turn on.

I'll get the types supported here shortly.

b-rowan, 2023-12-16:
New version with the support should be going up now (`v0.43.0`).  Just in case, IF you are using `MinerConfig`, be careful, as I accidentally pulled a huge amount of breaking changes for it into the master.  The config should still work if you were only using get and send config, but if you were modifying any parts of it that will no longer work as expected.

closed 2023-12-16 by b-rowan