UpstreamData/pyasic #79: problem with pyasic 0.43.0

Awote, 2023-12-17:
Hi, i updated ```pyasic``` and get errors.
 
 ```File "C:\Users\balak\scanner\core\asic_funcs.py", line 9, in <module>
    import pyasic
  File "c:\Users\balak\scanner\venv\Lib\site-packages\pyasic\__init__.py", line 22, in <module>
    from pyasic.config import MinerConfig
  File "c:\Users\balak\scanner\venv\Lib\site-packages\pyasic\config\__init__.py", line 26, in <module>
    @dataclass
     ^^^^^^^^^
  File "C:\Python311\Lib\dataclasses.py", line 1221, in dataclass
    return wrap(cls)
           ^^^^^^^^^
  File "C:\Python311\Lib\dataclasses.py", line 1211, in wrap
    return _process_class(cls, init, repr, eq, order, unsafe_hash,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Python311\Lib\dataclasses.py", line 959, in _process_class
    cls_fields.append(_get_field(cls, name, type, kw_only))
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Python311\Lib\dataclasses.py", line 816, in _get_field
    raise ValueError(f'mutable default {type(f.default)} for field '
ValueError: mutable default <class 'pyasic.config.pools.PoolConfig'> for field pools is not allowed: use default_factory

UpstreamData, 2023-12-17:
Let me look into this, seems like a python 3.11 change to dataclasses, hopefully shouldn't be too hard to fix.

UpstreamData, 2023-12-17:
Should be fixed in 0.43.1.  Give it a test and see if it's working for you, but I tested on 3.12 and no issues.

closed 2023-12-23 by b-rowan