UpstreamData/pyasic #80: implement MinerConfig scheme for ePIC UMC

jpcomps, 2023-12-17:
Now that the new MinerConfig scheme has been implemented, transition the ePIC UMC API to use the new config and give the ability to change settings. Just creating the feature request for tracking

May need some guidance @UpstreamData on your preferred way or thinking of how this should be implemented. My thoughts are to start with simply with changing the pool config which should be straightforward, then I can use that template/guidance to start implementing things like PerpetualTune (autotune settings).

UpstreamData, 2023-12-17:
It may be easiest to start with trying to go from a config on a miner to the MinerConfig by implementing the `from_epic` method on each part of the config.  It will also need to be added as a type hinting method to the classes in `config/base.py`.

I think there's going to be a similar issue with this and BOS+ GRPC, as well as slightly less severe with whatsminers, which is that config options are spread across different endpoints, so what I would suggest is that `MinerConfig.as_epic()` returns a dict with data for different commands to use.  For example, `as_wm()` returns a dict with `pools` and `tuning`, which is a dict with mode set to either normal, low, high, or power_tune, with the power_tune dict having the power limit added.

Hopefully this all makes sense.  Also, if you're on telegram, you can reach out to me directly there at b_rowan if you have questions or need help with any of that.

UpstreamData, 2023-12-17:
Fixing another issue with MinerConfig right now, so I'll add the base implementation as it's just passing anyway.

jpcomps, 2023-12-18:
Thanks for that, will take a look. I think what you proposed makes a lot of sense. Thanks for adding that. Will probably reach out on telegram as well but will give it a first go.

b-rowan, 2023-12-23:
#84 merged, part 1 of the support is done.

UpstreamData, 2024-01-26:
Finished in #101.

closed 2024-01-26 by UpstreamData