UpstreamData/pyasic #83: Add S19 Pro Hydro support

UpstreamData, 2023-12-22:
Add support for the S19 Pro Hydro as mentioned in https://github.com/Schnitzel/hass-miner/issues/271

UpstreamData, 2023-12-22:
@ors86

Chip count is on the web interface, just need the count of a working board.  I believe there is also only 3 boards, but if there is more please let me know.

Model string comes from the `/cgi-bin/get_system_info.cgi` endpoint (log into your miner, then enter `ip/cgi-bin/get_system_info.cgi`, such as `192.168.10.10/cgi-bin/get_system_info.cgi`) as "minertype".  Should show something like this `{"minertype":"Antminer S19j","nettype":"DHCP","netdevice":"eth0","macaddr":"A2:E1:2A:D0:11:A2","hostname":"Antminer","ipaddress":"10.0.5.164","netmask":"255.255.255.0","gateway":"","dnsservers":"","system_mode":"GNU/Linux","system_kernel_version":"Linux 4.6.0-xilinx-g03c746f7 #2 SMP PREEMPT Mon Sep 21 11:50:03 CST 2020","system_filesystem_version":"Fri Sep 15 14:39:20 CST 2023","firmware_type":"Release"}`, I just need the minertype string, in this case `Antminer S19j`.

ors86, 2023-12-22:
180 chips | 4 x hashing boards | "minertype":"Antminer S19 Pro Hyd."

b-rowan, 2023-12-22:
On it.  I will update the hass-miner branch after I get this released as a version.

b-rowan, 2023-12-22:
Can you confirm for me that the miner type is the same as shows up in `/cgi-bin/summary.cgi`?  That is a rather unconventional type string, usually they dont include periods...

ors86, 2023-12-23:
Here is the full output:

`{"STATUS": {"STATUS": "S", "when": 1703326030, "Msg": "summary", "api_version": "1.0.0"}, "INFO": {"miner_version": "49.0.1.3", "CompileTime": "Fri Aug 19 09:34:31 CST 2022", "type": "Antminer S19 Pro Hyd."}, "SUMMARY": [{"elapsed": 415020, "rate_5s": 186916.98, "rate_30m": 186484.4, "rate_avg": 186773.09, "rate_ideal": 184000.0, "rate_unit": "GH/s", "hw_all": 11000, "bestshare": 18058961468, "status": [{"type": "rate", "status": "s", "code": 0, "msg": ""}, {"type": "network", "status": "s", "code": 0, "msg": ""}, {"type": "fans", "status": "s", "code": 0, "msg": ""}, {"type": "temp", "status": "s", "code": 0, "msg": ""}]}]}`

b-rowan, 2023-12-23:
Awesome, this is done then.  Pyasic itself will work with that miner as of 0.44.1.

That being said there isn't anything I can do about home assistant right now (see the original issue), so this will  just have to wait until the new year when they update home assistant.  Thanks for your help!

closed 2023-12-23 by b-rowan