UpstreamData/pyasic #89: Could you add stop_mining in BMMiner?

PashaPlank, 2024-01-11:
I see that stop_mining func in BMMiner class can't really stop miner, could you add this?

UpstreamData, 2024-01-11:
Technically BMMiner on its own doesn't support shutdown or "sleep" mode, this is usually implemented by a downstream class (see `pyasic/miners/backends/antminer.py` under the `AntMinerModern` class).

If you have a way to shutdown BMMiner using only the API, I'm open to it, otherwise I will just have to implement it for whatever miner you're trying to shutdown.

closed 2024-01-26 by UpstreamData

s0kil, 2024-01-27:
The `power_off` command works well, and `power_on` to start mining again.

There will be a message in Miner Logs:
```
01-27 18:50:24 Suspend btminer as by whatsminer API
```

When powered off via API.

b-rowan, 2024-01-27:
@s0kil isnt that for BTMiner, not BMMiner?

s0kil, 2024-01-27:
Oh right, I've misread the title.

Well, if you want a safe way to ensure the machine won't hash. Set the pools to null, or empty. And It will force it not to hash.

s0kil, 2024-01-27:
@b-rowan BMMiner doesn't support sleep mode?