UpstreamData/pyasic #94: Unable to get_data for ePIC board on S19k Pro

musicman23, 2024-01-15:
**Describe the bug**
Unable to get data from ePIC board with latest (or outdated) firmware. Miner model is S19k Pro 120. 

**To Reproduce**
Steps to reproduce the behavior, run this script:
```
import asyncio
from pyasic import get_miner

async def gather_miner_data():
    miner = await get_miner("192.168.29.32")
    if miner is not None:
        miner_data = await miner.get_data()
        print(miner_data)  # all data from the dataclass
        print(miner_data.hashrate)  # hashrate of the miner in TH/s

if __name__ == "__main__":
    asyncio.run(gather_miner_data())
```
Gets this output:
```
Traceback (most recent call last):
  File "/Users/lee/git/minerstats/test.py", line 12, in <module>
    asyncio.run(gather_miner_data())
  File "/opt/homebrew/Cellar/python@3.11/3.11.6_1/Frameworks/Python.framework/Versions/3.11/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/opt/homebrew/Cellar/python@3.11/3.11.6_1/Frameworks/Python.framework/Versions/3.11/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/Cellar/python@3.11/3.11.6_1/Frameworks/Python.framework/Versions/3.11/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/Users/lee/git/minerstats/test.py", line 7, in gather_miner_data
    miner_data = await miner.get_data()
                 ^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/lee/git/minerstats/venv/lib/python3.11/site-packages/pyasic/miners/base.py", line 582, in get_data
    gathered_data = await self._get_data(
                    ^^^^^^^^^^^^^^^^^^^^^
  File "/Users/lee/git/minerstats/venv/lib/python3.11/site-packages/pyasic/miners/base.py", line 552, in _get_data
    miner_data[data_name] = await function(**args_to_send)
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: object NoneType can't be used in 'await' expression
```

**Expected behavior**
Miner data returned.

**Desktop (please complete the following information):**
 - OS: MacOS 
 - Version 14.1.2 Sonoma

**Miner Information (If applicable):**
 - Manufacturer: Bitmain
 - Type: S19k Pro 120
 - Firmware Type: ePIC Blockchain board
 - Firmware Version: 38.2 (also tested with 32.4)

**Additional context**
pyasic version 0.46.1

The miner is reachable. I verified the script works on an S19k Pro with factory board (Amlogic), and S19j Pro 104 with Braiins. I expect this is an issue with newer model with ePIC board.

UpstreamData, 2024-01-15:
Let me take a look into this, I know generally what's causing the issue, but ill see if I can identify the specifics.

UpstreamData, 2024-01-15:
Can you test with the newest version I just put out, `0.47.0`?  It wont fix the issue, but it will make the logging of what's actually happening more verbose.  Issue I'm running into right now is I can't identify which specific function is missing from the ePIC that is causing that.

Also, just be aware, 0.47.0 comes with a massive body of changes (mostly to BraiinsOS+), so if you're using the same venv for something you may want to separate them for now, just so nothing breaks.

musicman23, 2024-01-15:
Awesome, Thank you! Here is the updated output:

```
Traceback (most recent call last):
  File "/Users/lee/git/minerstats/venv/lib/python3.11/site-packages/pyasic/miners/base.py", line 609, in _get_data
    miner_data[data_name] = await function(**args_to_send)
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: object NoneType can't be used in 'await' expression

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/Users/lee/git/minerstats/test.py", line 12, in <module>
    asyncio.run(gather_miner_data())
  File "/opt/homebrew/Cellar/python@3.11/3.11.6_1/Frameworks/Python.framework/Versions/3.11/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/opt/homebrew/Cellar/python@3.11/3.11.6_1/Frameworks/Python.framework/Versions/3.11/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/Cellar/python@3.11/3.11.6_1/Frameworks/Python.framework/Versions/3.11/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/Users/lee/git/minerstats/test.py", line 7, in gather_miner_data
    miner_data = await miner.get_data()
                 ^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/lee/git/minerstats/venv/lib/python3.11/site-packages/pyasic/miners/base.py", line 644, in get_data
    gathered_data = await self._get_data(
                    ^^^^^^^^^^^^^^^^^^^^^
  File "/Users/lee/git/minerstats/venv/lib/python3.11/site-packages/pyasic/miners/base.py", line 611, in _get_data
    raise APIError(
pyasic.errors.APIError: Failed to call api_ver on ePIC S19k Pro (ePIC): 192.168.29.32 while getting data.
```

UpstreamData, 2024-01-15:
Should be fixed in `0.47.1`.  Had some abstract methods defined as regular, non-async functions, and that was messing it up.

closed 2024-01-15 by UpstreamData

musicman23, 2024-01-15:
Thank you so much!!! Working.