UpstreamData/pyasic #103: Bos miner.web.grpc.get_pool_groups() return OSError: [WinError 121]

Sam-SHanshin, 2024-01-28:
**Describe the bug**
When i try get miner's pools status by miner.api.pools() i got empty dict then i try to use 
``` miner.web.grpc.get_pool_groups() ```
i got 
```OSError: [WinError 121]```

**Screenshots**
Traceback (most recent call last):
```File "C:\Users\Оксана\Desktop\Sam\core\asic_funcs.py", line 200, in get_miner_data
    x = await controle_object.web.grpc.get_pool_groups()
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "c:\Users\Оксана\Desktop\Sam\venv\Lib\site-packages\pyasic\web\bosminer\__init__.py", line 525, in get_pool_groups
    return await self.send_command("get_pool_groups", GetPoolGroupsRequest())
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "c:\Users\Оксана\Desktop\Sam\venv\Lib\site-packages\pyasic\web\bosminer\__init__.py", line 319, in send_command
    metadata.append(("authorization", await self.auth()))
                                      ^^^^^^^^^^^^^^^^^
  File "c:\Users\Оксана\Desktop\Sam\venv\Lib\site-packages\pyasic\web\bosminer\__init__.py", line 333, in auth
    await self._get_auth()
  File "c:\Users\Оксана\Desktop\Sam\venv\Lib\site-packages\pyasic\web\bosminer\__init__.py", line 345, in _get_auth
    await stream.send_message(req, end=True)
  File "c:\Users\Оксана\Desktop\Sam\venv\Lib\site-packages\grpclib\client.py", line 241, in send_message
    await self.send_request()
  File "c:\Users\Оксана\Desktop\Sam\venv\Lib\site-packages\grpclib\client.py", line 173, in send_request
    protocol = await self._channel.__connect__()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "c:\Users\Оксана\Desktop\Sam\venv\Lib\site-packages\grpclib\client.py", line 748, in __connect__
    self._protocol = await self._create_connection()
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "c:\Users\Оксана\Desktop\Sam\venv\Lib\site-packages\grpclib\client.py", line 725, in _create_connection
    _, protocol = await self._loop.create_connection(
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Оксана\AppData\Local\Programs\Python\Python311\Lib\asyncio\base_events.py", line 1085, in create_connection
    raise exceptions[0]
  File "C:\Users\Оксана\AppData\Local\Programs\Python\Python311\Lib\asyncio\base_events.py", line 1069, in create_connection
    sock = await self._connect_sock(
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Оксана\AppData\Local\Programs\Python\Python311\Lib\asyncio\base_events.py", line 973, in _connect_sock
    await self.sock_connect(sock, address)
  File "C:\Users\Оксана\AppData\Local\Programs\Python\Python311\Lib\asyncio\proactor_events.py", line 726, in sock_connect
    return await self._proactor.connect(sock, address)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Оксана\AppData\Local\Programs\Python\Python311\Lib\asyncio\windows_events.py", line 846, in _poll
    value = callback(transferred, key, ov)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Оксана\AppData\Local\Programs\Python\Python311\Lib\asyncio\windows_events.py", line 633, in finish_connect
    ov.getresult()
OSError: [WinError 121]
```
**Desktop (please complete the following information):**
- python 3.11.4
- windows 10

**Miner Information (If applicable):**
 - Type: Antminer S19J Pro NoPic (BOS)
 - Firmware Type: [BraiinsOS]
 - Firmware Version: 2023-04-20-0-0ce150e9-23.03-plus

**Additional context**
i use pyasic 0.46.0

b-rowan, 2024-01-28:
OSError 121 is refused connection, you have to update your miner to the latest version, since older versions don't have gRPC support.

I would also recommend updating pyasic to the latest version, as that will fix some other bugs.

b-rowan, 2024-01-28:
Also, on recent versions, it will become `miner.web` instead of `miner.web.grpc`, but I would recommend using `get_config` instead if you just intend to get config information.

Sam-SHanshin, 2024-01-28:
What i BOSerWebAPI and BOSMinerRPCAPI. I mean that  why do you need to create two classes for Bos?

Sam-SHanshin, 2024-01-28:
I see that use ```get_config``` is good practic, but i need to get pools status (Are they Alive or Dead?)

b-rowan, 2024-01-28:
Regarding your first question; The reason for 2 BOS+ classes is that they dropped support for S9 (and technically X17) prior to implementing gRPC, so in order to maintain compatibility in a simple way with S9s, I split the classes out into something for S9 and something for newer models.  This is slightly buggy with X17 models, because they are updatable to new versions but have tuner issues, so it's troublesome to decide between implementations there.

Regarding active pool status, yes, that's the correct way to do that then.  I've considered trying to add that to miner data, but it seems complicated to structure that in a useful way, and it's not something I would want to put into miner config...

b-rowan, 2024-06-11:
> I see that use `get_config` is good practic, but i need to get pools status (Are they Alive or Dead?) `get_config` give me pools parameters but not statuses

Getting this added into miner data, some miners already have it, and more are getting it soon.

b-rowan, 2024-08-20:
This should be fixed in latest, give it a test and let me know if you still have issues.

closed 2024-08-20 by b-rowan