UpstreamData/pyasic #104: Braiins S17 Pro stopped working after 0.46.0

dcod3d, 2024-01-30:
**Describe the bug**
pyasic no longer detects miners after version `0.46.0` 
Braiins firmware on `23.01` [Last supported version for X17 series]

**To Reproduce**
install pyasic > `0.46.0` 
run sample data gathering code `miner.get_data()` from `pyasic-0.50.4`
```Traceback (most recent call last):
  File "/home/XXX/test.py", line 17, in <module>
    asyncio.run(gather_miner_data())
  File "/usr/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/home/XXX/test.py", line 11, in gather_miner_data
    all_miner_data = await asyncio.gather(*[miner.get_data() for miner in miners])
  File "XXX/lib/python3.9/site-packages/pyasic/miners/base.py", line 474, in get_data
    gathered_data = await self._get_data(
  File "XXX/lib/python3.9/site-packages/pyasic/miners/base.py", line 398, in _get_data
    self.web.multicommand(*web_multicommand, allow_warning=allow_warning)
TypeError: multicommand() got an unexpected keyword argument 'allow_warning'
```

**Desktop (please complete the following information):**
 - OS: debian

**Miner Information (If applicable):**
 - Manufacturer: Bitmain
 - Type: S17 Pro
 - Firmware Type: BraiinsOS
 - Firmware Version: 23.01

**Additional context**

UpstreamData, 2024-01-30:
Ill take a look at it here shortly, there are some issues with BOS+ on X17 models because they lack gRPC support, but ill just make them subclass BOSMiner instead of BOSer I suppose.

UpstreamData, 2024-01-30:
Can you try with pyasic verson 0.50.6?  `python3 -m pip install pyasic==0.50.6`

dcod3d, 2024-01-30:
Working with this update. Thanks!

closed 2024-01-30 by dcod3d