UpstreamData/pyasic #106: Some times i can't get miner's mac

Awote, 2024-02-07:
**Describe the bug**
i use 
```pyasic_miners = await MinerNetwork.from_address(ip_range).scan()
get_data_task =  [controle_object.get_data() for controle_object in pyasic_miners]
data = await asyncio.gather(*get_data_task)
```

And when i check debug i see that some miners don't have mac.
Also i set custom parameters for pyasic
```update("network_ping_retries", 5)
update("network_ping_timeout", 5)
update("factory_get_retries", 3)
update("get_data_retries", 6)
update("factory_get_timeout", 4)
update("api_function_timeout", 7)
update("socket_linger_time", 4000)
```

**Screenshots**
if you want i can send to you txt log with miner's data. if it ll be useful

**Miner Information (If applicable):**
 - pyasic == 0.46

b-rowan, 2024-02-07:
It's possible that the MAC command isn't responding properly, what type, firmware, and firmware version (if not in stock) are you using?  Assuming it's BraiinsOS+, what are the versions that aren't getting the MAC?

tydal-borge, 2024-07-12:
I'm also struggling with this, mac (and hostname for get_data) is set to None in both the response of get_data and get_mac.
I have this issue for all AntMiner models running stock firmware. I have set the correct password, and all other data is coming through fine.

I looked through the source code to try to understand how it gets the mac address internally, and from my understanding it should use the web api for this?

The web api call to "get_network_info" is the only source of the mac address afaik, and it internally calls ifconfig | grep eth0 and splits the result. (It's ugly, but that's another story)

But in the _get_mac function there is a dict called web_get_system_info that I couldn't quite understand.
I tried debugging it by printing what this variable is, and it prints twice:

```
web_get_system_info:  {}
web_get_system_info:  None
```

tydal-borge, 2024-07-29:
@b-rowan 

I _think_ I solved this issue:
In the readme, the way you set a custom AntMiner password is like this:

`settings.update("default_antminer_password", "my_pwd")`

however, the correct way should be 
`settings.update("default_antminer_web_password", "my_pwd")`

which caused me a big headache when trying to find out what went wrong.

@Awote

> And when i check debug i see that some miners don't have mac. Also i set custom parameters for pyasic
> 
> ```
> update("network_ping_timeout", 5)
> update("factory_get_retries", 3)
> update("get_data_retries", 6)
> update("factory_get_timeout", 4)
> update("api_function_timeout", 7)
> update("socket_linger_time", 4000)
> ```
> 

You didn't mention setting your password here, and a password is required to get hostname and mac from miners, as these can only be retrieved in the web API. Even if you set a password, please see above to make sure you didn't make my mistake.

b-rowan, 2024-07-30:
> @b-rowan
>
>
>
> I _think_ I solved this issue:
>
> In the readme, the way you set a custom AntMiner password is like this:
>
>
>
> `settings.update("default_antminer_password", "my_pwd")`
>
>
>
> however, the [...]

Nice find.  I've been pretty busy for the past few weeks, so I haven't touched much here, but this seems like part of the problem for sure.  Never did get a response on the initial problem, I'm not sure if it was with stock or BOS+, but I have updated the documentation for settings so hopefully it is more clear.

closed 2024-08-20 by b-rowan