UpstreamData/pyasic #107: KD box ii hash rate mode change

mahines2290, 2024-02-09:
can you please add support for the gold shell KDbox pro and the KDboxii? Thank you!

UpstreamData, 2024-02-09:
I can try, can you confirm if they are identified as `Goldshell` by `get_miner`?

If you can send the result of this code for both the miners, just make sure to set IP at the top - 

```python
import pyasic
import asyncio

IP = ""

async def main():
    miner = await pyasic.get_miner(IP)
    print(miner)
    print(await pyasic.miner_factory.get_miner_model_goldshell(IP))
    print(await miner.api.send_command("devs"))

asyncio.run(main())
```

mahines2290, 2024-02-10:
Unknown: xxx.xxx.xxx.xxx (redacted)
Goldshell KDBoxⅡ
{'STATUS': [{'STATUS': 'S', 'When': 1707526459, 'Code': 9, 'Msg': '1
PGA(s)', 'Description': 'kdaminer 5.4.2'}], 'DEVS': [{'PGA': 0, 'Name':
'INCS', 'ID': 0, 'Enabled': 'Y', 'Status': 'Alive', 'Device Elapsed':
13502, 'MHS av': 5064851.372, 'MHS 20s': 4986841.201, 'MHS rolling':
4986841.201, 'Nonced': 497734, 'Accepted': 1488, 'Rejected': 11, 'Hardware
Errors': 177, 'Utility': 6.613, 'Stale': 0, 'Last Share Pool': 0, 'Last
Share Time': 1707526440, 'Total MH': 68383438494.7515, 'Diff1 Work':
15921824, 'Work Utility': 70755.441, 'Difficulty Accepted': 16127741,
'Difficulty Rejected': 94103, 'Difficulty Stale': 0, 'Last Share
Difficulty': 17146, 'Last Valid Work': 1707526459, 'Device Hardware%':
0.0356, 'Device Rejected%': 0.591, 'MHS U': 5011866.0, 'voltage': 47.0,
'clock': 790.0, 'fan0': 2820, 'fan1': 2820, 'tstemp-0': 92.0, 'tstemp-1':
92.0, 'tstemp-2': 78.56, 'temp_tv_2': 611793, 'temp_tv': 611793,
'overheat': 0, 'rebootcnt': 0, 'hwerr-ration': 0.000353,
'estimate_hash_rate': 47400000000.0, 'range_scan_time': 0.339792}], 'id': 1}

Process finished with exit code 0

mahines2290, 2024-02-10:
Goldshell KDBoxPro
{'STATUS': [{'STATUS': 'S', 'When': 1707526718, 'Code': 9, 'Msg': '1
PGA(s)', 'Description': 'kdaminer 5.4.2'}], 'DEVS': [{'PGA': 0, 'Name':
'INCS', 'ID': 0, 'Enabled': 'Y', 'Status': 'Alive', 'Device Elapsed':
13763, 'MHS av': 2612890.111, 'MHS 20s': 2594288.93, 'MHS rolling':
2594288.93, 'Nonced': 261648, 'Accepted': 1453, 'Rejected': 6, 'Hardware
Errors': 0, 'Utility': 6.335, 'Stale': 0, 'Last Share Pool': 0, 'Last Share
Time': 1707526706, 'Total MH': 35959940103.2931, 'Diff1 Work': 8372736,
'Work Utility': 36502.351, 'Difficulty Accepted': 8422738, 'Difficulty
Rejected': 44444, 'Difficulty Stale': 0, 'Last Share Difficulty': 7186,
'Last Valid Work': 1707526718, 'Device Hardware%': 0.0, 'Device Rejected%':
0.5308, 'MHS U': 2587107.75, 'voltage': 0.43, 'clock': 915.0, 'fan0': 2160,
'fan1': 2160, 'tstemp-0': 86.0, 'tstemp-1': 86.0, 'tstemp-2': 74.31,
'temp_tv_2': 709368, 'temp_tv': 709368, 'overheat': 0, 'rebootcnt': 0,
'hwerr-ration': 0.0, 'estimate_hash_rate': 54900000000.0,
'range_scan_time': 0.293372}], 'id': 1}

Process finished with exit code 0

mahines2290, 2024-02-10:
I would love it if it is possible to switch the powermodes from hashrate to low-power if possible in HA

b-rowan, 2024-02-10:
Awesome, I'll try to get those added tomorrow, but it's mildly concerning it isn't identifying them initially.  I think I can just check for kdaminer, but I'll maybe have you get the output of the initial web query.

Regarding hass-miner, the issue is that it depends on set_wattage, which relies on a contiguous wattage setpoint, rather than discrete power modes.  I'll see if I can figure anything out for that, but it's kindof complex.

mahines2290, 2024-02-10:
[har files.zip](https://github.com/UpstreamData/pyasic/files/14230336/har.files.zip)
does this help?

b-rowan, 2024-02-10:
It does, yes.  I'll get that power mode thing implemented as well, it seems like its just posting to `/mcb/settings`, which is about what I expect.

Can you also run this code?  This will give me the web response for identification, devdetails for chip counts and board counts, and stats for fan counts and format.  You will want to send this in a text file, the web response will be quite large.

```python
import pyasic
import asyncio
import httpx

IP = ""

async def main():
    m = await pyasic.get_miner(IP)
    print("Model:", await pyasic.miner_factory.get_miner_model_goldshell(IP))
    print("Devdetails:", await m.api.send_command("devdetails"))
    print("Stats:", await m.api.send_command("stats"))
    async with httpx.AsyncClient(transport=pyasic.settings.transport()) as c:
        print("Web Txt:", await pyasic.miner_factory._web_ping(c, f"http://{IP}/"))
    
    
asyncio.run(main())
```

mahines2290, 2024-02-10:
/Users/matthew.hines/PycharmProjects/pythonProject1/.venv/bin/python /Users/matthew.hines/PycharmProjects/pythonProject1/.venv/test.py 
Model: Goldshell KDBoxPro
Devdetails: {}
Stats: {}
Web Txt: ('<!DOCTYPE html><html lang="en"><head><meta charset="utf-8"><meta http-equiv="X-UA-Compatible" content="IE=edge"><meta name="viewport" content="width=device-width,initial-scale=1"><link rel="icon" href="/favicon.ico"><title>cloud-box</title><link href="/css/debug.0a06fc18.css" rel="prefetch"><link href="/css/debug~home~layout~login~setting-miner~setting-system.ba9d0cf7.css" rel="prefetch"><link href="/css/debug~home~login~setting-miner~setting-system.5e46d2ea.css" rel="prefetch"><link href="/css/debug~layout~setting-miner~setting-system.282a7c4e.css" rel="prefetch"><link href="/css/home.5fabb3ac.css" rel="prefetch"><link href="/css/layout.f71ffc8c.css" rel="prefetch"><link href="/css/login~setting-miner~setting-system.01a57b06.css" rel="prefetch"><link href="/css/setting-miner.12979a8f.css" rel="prefetch"><link href="/css/setting-system.82bea42d.css" rel="prefetch"><link href="/js/debug.fb031cac.js" rel="prefetch"><link href="/js/debug~home~layout~login~setting-miner~setting-system.227d0437.js" rel="prefetch"><link href="/js/debug~home~login~setting-miner~setting-system.106cbbf1.js" rel="prefetch"><link href="/js/debug~layout~setting-miner~setting-system.470c03ef.js" rel="prefetch"><link href="/js/home.213e7cde.js" rel="prefetch"><link href="/js/layout.bb684154.js" rel="prefetch"><link href="/js/login.1848b1c8.js" rel="prefetch"><link href="/js/login~setting-miner~setting-system.3bd9d078.js" rel="prefetch"><link href="/js/setting-miner.520a591e.js" rel="prefetch"><link href="/js/setting-system.45c3122f.js" rel="prefetch"><link href="/css/app.3c949e12.css" rel="preload" as="style"><link href="/css/chunk-vendors.ae9f9d43.css" rel="preload" as="style"><link href="/js/app.e568c518.js" rel="preload" as="script"><link href="/js/chunk-vendors.1afa4163.js" rel="preload" as="script"><link href="/css/chunk-vendors.ae9f9d43.css" rel="stylesheet"><link href="/css/app.3c949e12.css" rel="stylesheet"></head><body><noscript><strong>We\'re sorry but cloud-box doesn\'t work properly without JavaScript enabled. Please enable it to continue.</strong></noscript><div id="app"></div><script src="/js/chunk-vendors.1afa4163.js"></script><script src="/js/app.e568c518.js"></script></body></html>', <Response [200 OK]>)

Process finished with exit code 0

mahines2290, 2024-02-10:
Model: Goldshell KDBoxⅡ
Devdetails: {'STATUS': [{'STATUS': 'S', 'When': 1707597153, 'Code': 69, 'Msg': 'Device Details', 'Description': 'kdaminer 5.4.2'}], 'DEVS': [{'DEVDETAILS': 0, 'Name': 'INCS', 'ID': 0, 'Processors': 1, 'Driver': 'intchains_qomo', 'Target Temperature': 89, 'Cutoff Temperature': 95, 'chips-nr': 36, 'core-nr': 60, 'work mode': 'async', 'c1-temp': 0.0, 'c2-temp': 0.0, 'c3-temp': 0.0, 'c4-temp': 0.0, 'c5-temp': 0.0, 'c6-temp': 0.0, 'c7-temp': 0.0, 'c8-temp': 0.0, 'c9-temp': 0.0, 'c10-temp': 0.0, 'c11-temp': 0.0, 'c12-temp': 0.0, 'c13-temp': 0.0, 'c14-temp': 0.0, 'c15-temp': 0.0, 'c16-temp': 0.0, 'c17-temp': 0.0, 'c18-temp': 0.0, 'c19-temp': 0.0, 'c20-temp': 0.0, 'c21-temp': 0.0, 'c22-temp': 0.0, 'c23-temp': 0.0, 'c24-temp': 0.0, 'c25-temp': 0.0, 'c26-temp': 0.0, 'c27-temp': 0.0, 'c28-temp': 0.0, 'c29-temp': 0.0, 'c30-temp': 0.0, 'c31-temp': 0.0, 'c32-temp': 0.0, 'c33-temp': 0.0, 'c34-temp': 0.0, 'c35-temp': 0.0, 'c36-temp': 0.0, 'c1-tempdelt': 550.0, 'c2-tempdelt': 550.0, 'c3-tempdelt': 550.0, 'c4-tempdelt': 550.0, 'c5-tempdelt': 550.0, 'c6-tempdelt': 550.0, 'c7-tempdelt': 550.0, 'c8-tempdelt': 550.0, 'c9-tempdelt': 550.0, 'c10-tempdelt': 550.0, 'c11-tempdelt': 550.0, 'c12-tempdelt': 550.0, 'c13-tempdelt': 550.0, 'c14-tempdelt': 550.0, 'c15-tempdelt': 550.0, 'c16-tempdelt': 550.0, 'c17-tempdelt': 550.0, 'c18-tempdelt': 550.0, 'c19-tempdelt': 550.0, 'c20-tempdelt': 550.0, 'c21-tempdelt': 550.0, 'c22-tempdelt': 550.0, 'c23-tempdelt': 550.0, 'c24-tempdelt': 550.0, 'c25-tempdelt': 550.0, 'c26-tempdelt': 550.0, 'c27-tempdelt': 550.0, 'c28-tempdelt': 550.0, 'c29-tempdelt': 550.0, 'c30-tempdelt': 550.0, 'c31-tempdelt': 550.0, 'c32-tempdelt': 550.0, 'c33-tempdelt': 550.0, 'c34-tempdelt': 550.0, 'c35-tempdelt': 550.0, 'c36-tempdelt': 550.0, 'c1-cores': 90, 'c2-cores': 90, 'c3-cores': 90, 'c4-cores': 90, 'c5-cores': 90, 'c6-cores': 90, 'c7-cores': 90, 'c8-cores': 90, 'c9-cores': 90, 'c10-cores': 90, 'c11-cores': 90, 'c12-cores': 90, 'c13-cores': 90, 'c14-cores': 90, 'c15-cores': 90, 'c16-cores': 90, 'c17-cores': 90, 'c18-cores': 90, 'c19-cores': 90, 'c20-cores': 90, 'c21-cores': 90, 'c22-cores': 90, 'c23-cores': 90, 'c24-cores': 90, 'c25-cores': 90, 'c26-cores': 90, 'c27-cores': 90, 'c28-cores': 90, 'c29-cores': 90, 'c30-cores': 90, 'c31-cores': 90, 'c32-cores': 90, 'c33-cores': 90, 'c34-cores': 90, 'c35-cores': 90, 'c36-cores': 90, 'c1_hwerr': 0, 'c2_hwerr': 247, 'c3_hwerr': 0, 'c4_hwerr': 0, 'c5_hwerr': 0, 'c6_hwerr': 0, 'c7_hwerr': 0, 'c8_hwerr': 0, 'c9_hwerr': 0, 'c10_hwerr': 0, 'c11_hwerr': 284, 'c12_hwerr': 0, 'c13_hwerr': 5, 'c14_hwerr': 432, 'c15_hwerr': 0, 'c16_hwerr': 0, 'c17_hwerr': 0, 'c18_hwerr': 0, 'c19_hwerr': 17, 'c20_hwerr': 0, 'c21_hwerr': 0, 'c22_hwerr': 0, 'c23_hwerr': 0, 'c24_hwerr': 0, 'c25_hwerr': 1, 'c26_hwerr': 1, 'c27_hwerr': 0, 'c28_hwerr': 0, 'c29_hwerr': 0, 'c30_hwerr': 0, 'c31_hwerr': 0, 'c32_hwerr': 0, 'c33_hwerr': 0, 'c34_hwerr': 0, 'c35_hwerr': 0, 'c36_hwerr': 0, 'c1_perf': 85953, 'c2_perf': 85330, 'c3_perf': 86046, 'c4_perf': 86173, 'c5_perf': 85823, 'c6_perf': 85880, 'c7_perf': 86165, 'c8_perf': 85579, 'c9_perf': 86013, 'c10_perf': 85808, 'c11_perf': 85894, 'c12_perf': 85675, 'c13_perf': 85430, 'c14_perf': 85490, 'c15_perf': 85641, 'c16_perf': 85976, 'c17_perf': 85597, 'c18_perf': 85536, 'c19_perf': 85484, 'c20_perf': 85789, 'c21_perf': 85243, 'c22_perf': 85655, 'c23_perf': 85538, 'c24_perf': 85917, 'c25_perf': 85209, 'c26_perf': 85427, 'c27_perf': 85508, 'c28_perf': 85331, 'c29_perf': 85366, 'c30_perf': 85084, 'c31_perf': 85163, 'c32_perf': 85757, 'c33_perf': 85390, 'c34_perf': 85191, 'c35_perf': 85739, 'c36_perf': 85165}], 'id': 1}
Stats: {'STATUS': [{'STATUS': 'S', 'When': 1707597153, 'Code': 70, 'Msg': 'BFGMiner stats', 'Description': 'kdaminer 5.4.2'}], 'STATS': [{'CGMiner': '5.4.2', 'Miner': 'kdaminer 5.4.2-unknown'}, {'STATS': 0, 'ID': 'INCS0', 'Elapsed': 84163, 'Calls': 322011, 'Wait': 16.683353, 'Max': 5.683333, 'Min': 0.0, 'MHS 5s': 3394064.126, 'MHS av': 5032703.339, 'miner_count': 1, 'frequency': 550.0, 'fan_num': 2, 'fan0': 1380, 'fan1': 1380}, {'STATS': 1, 'ID': 'POOL0', 'Elapsed': 84167, 'Calls': 322011, 'Wait': 16.683353, 'Max': 5.683333, 'Min': 0.0, 'Pool Calls': 0, 'Pool Attempts': 0, 'Pool Wait': 0.0, 'Pool Max': 0.0, 'Pool Min': 99999999.0, 'Pool Av': 0.0, 'Work Had Roll Time': False, 'Work Can Roll': False, 'Work Had Expire': False, 'Work Roll Time': 0, 'Work Diff': 18088, 'Min Diff': 3000, 'Max Diff': 85588, 'Min Diff Count': 2050, 'Max Diff Count': 164, 'Times Sent': 9287, 'Bytes Sent': 1020120, 'Times Recv': 20047, 'Bytes Recv': 6242423, 'Net Bytes Sent': 1020120, 'Net Bytes Recv': 6242423}], 'id': 1}
Web Txt: ('<!DOCTYPE html><html lang="en"><head><meta charset="utf-8"><meta http-equiv="X-UA-Compatible" content="IE=edge"><meta name="viewport" content="width=device-width,initial-scale=1"><link rel="icon" href="/favicon.ico"><title>cloud-box</title><link href="/css/debug.0a06fc18.css" rel="prefetch"><link href="/css/debug~home~layout~login~setting-miner~setting-system.ba9d0cf7.css" rel="prefetch"><link href="/css/debug~home~login~setting-miner~setting-system.5e46d2ea.css" rel="prefetch"><link href="/css/debug~layout~setting-miner~setting-system.282a7c4e.css" rel="prefetch"><link href="/css/home.5fabb3ac.css" rel="prefetch"><link href="/css/layout.f71ffc8c.css" rel="prefetch"><link href="/css/login~setting-miner~setting-system.01a57b06.css" rel="prefetch"><link href="/css/setting-miner.12979a8f.css" rel="prefetch"><link href="/css/se [... cut ...] .1afa4163.js" rel="preload" as="script"><link href="/css/chunk-vendors.ae9f9d43.css" rel="stylesheet"><link href="/css/app.3c949e12.css" rel="stylesheet"></head><body><noscript><strong>We\'re sorry but cloud-box doesn\'t work properly without JavaScript enabled. Please enable it to continue.</strong></noscript><div id="app"></div><script src="/js/chunk-vendors.1afa4163.js"></script><script src="/js/app.e568c518.js"></script></body></html>', <Response [200 OK]>)

Process finished with exit code 0

mahines2290, 2024-02-10:
i will be honest i have no idea what i am doing lol. i am using pycharm to run the code but it looks like you need some javascript running which this software doesn't handle. what free program would you recommend to rerun or is the run above enough information?

b-rowan, 2024-02-10:
No stress, thats exactly what I needed.  The way pyasic works when scanning is it checks the result of a few endpoints in different formats, that being one of them, then checks for a string inside the result.  The string im looking for in this case is `"cloud-box"`, so that return is perfect.  Thats what I expect to see from goldshell miners.

I've put out a commit that does the basic support, but its a bit tricky with the mcb settings thing.

Are you able to send pictures of what that settings page (the one where you set the mode) looks like, what the different mode settings are, and the values its sending in the requests from request browser devtools?  The devtools thing can be done by hitting `CTRL+SHIFT+I`, then going to the Network tab, making sure its recording (usually top left of the menu), settign the power mode how you normally would, then looking for `/mcb/settings` in the requests its tracking.  I need the values from the Payload tab, so I can confirm what its actually setting for each power mode.

Heres a picture of what that looks like.
[image]

mahines2290, 2024-02-10:
KDBoxII low power payload

{select: 0, tempcontrol: true, ledcontrol: false, manualPowerplan: "790 MHz 47 V 40 RPM 50 RPM",…}
ledcontrol
: 
false
manual
: 
false
manualPowerplan
: 
"790 MHz 47 V 40 RPM 50 RPM"
name
: 
"28:E2:97:3D:F5:61"
powerplans
: 
[{temp_target: 78, level: 0, info: "790 MHz 47 V 40 RPM 50 RPM", prevoltage: 44.00001,…},…]
select
: 
0
tempcontrol
: 
true
version
: 
"v1.7"

mahines2290, 2024-02-10:
KDBoxII Hashrate Mode 
[image]
{"select":0,"tempcontrol":true,"ledcontrol":false,"manualPowerplan":"790 MHz 47 V 40 RPM 50 RPM","name":"28:E2:97:3D:F5:61","powerplans":[{"temp_target":78,"level":0,"info":"790 MHz 47 V 40 RPM 50 RPM","prevoltage":44.00001,"voltage":47.00001},{"temp_target":78,"level":2,"info":"550 MHz 65 V 40 RPM 64 RPM","prevoltage":63.00001,"voltage":65.00001},{"level":3,"info":"0 MHz 0 V 40 RPM 64 RPM"}],"manual":false,"version":"v1.7"}

mahines2290, 2024-02-10:
KDBoxII HasRate Mode
[image]
{"select":0,"tempcontrol":true,"ledcontrol":false,"manualPowerplan":"790 MHz 47 V 40 RPM 50 RPM","name":"28:E2:97:3D:F5:61","powerplans":[{"temp_target":78,"level":0,"info":"790 MHz 47 V 40 RPM 50 RPM","prevoltage":44.00001,"voltage":47.00001},{"temp_target":78,"level":2,"info":"550 MHz 65 V 40 RPM 64 RPM","prevoltage":63.00001,"voltage":65.00001},{"level":3,"info":"0 MHz 0 V 40 RPM 64 RPM"}],"manual":false,"version":"v1.7"}

mahines2290, 2024-02-10:
KDBoxII Low Power mode 
[image]
{"manualPowerplan":"790 MHz 47 V 40 RPM 50 RPM","ledcontrol":false,"name":"28:E2:97:3D:F5:61","tempcontrol":true,"select":1,"powerplans":[{"temp_target":78,"level":0,"info":"790 MHz 47 V 40 RPM 50 RPM","prevoltage":44.00001,"voltage":47.00001},{"temp_target":78,"level":2,"info":"550 MHz 65 V 40 RPM 64 RPM","prevoltage":63.00001,"voltage":65.00001},{"level":3,"info":"0 MHz 0 V 40 RPM 64 RPM"}],"manual":false,"version":"v1.7"}

mahines2290, 2024-02-10:
KDBoxII Idle mode 
[image]
{"manualPowerplan":"790 MHz 47 V 40 RPM 50 RPM","ledcontrol":false,"name":"2A:CC:F9:1C:6C:70","tempcontrol":true,"select":2,"powerplans":[{"info":"790 MHz 47 V 40 RPM 50 RPM","level":0,"temp_target":78,"prevoltage":44.00001,"voltage":47.00001},{"info":"550 MHz 65 V 40 RPM 64 RPM","level":2,"temp_target":78,"prevoltage":63.00001,"voltage":65.00001},{"info":"0 MHz 0 V 40 RPM 64 RPM","level":3}],"manual":false,"version":"v1.7"}

mahines2290, 2024-02-10:
[image]
thats formatted better

mahines2290, 2024-02-10:
KDBox Pro idle mode
[image]
{"ledcontrol":false,"manualPowerplan":"915 MHz 0.43 V 50 RPM 50 RPM","tempcontrol":true,"select":1,"name":"60:FB:00:90:EF:4C","powerplans":[{"level":0,"info":"915 MHz 0.43 V 50 RPM 50 RPM"},{"level":3,"info":"0 MHz 0 V 50 RPM 50 RPM"}],"manual":false,"temp_targets":[70,80],"version":"v1.2","temp_target":75}

mahines2290, 2024-02-10:
kd box pro hashrate mode
[image]
{"ledcontrol":false,"manualPowerplan":"915 MHz 0.43 V 50 RPM 50 RPM","tempcontrol":true,"select":0,"name":"60:FB:00:90:EF:4C","powerplans":[{"level":0,"info":"915 MHz 0.43 V 50 RPM 50 RPM"},{"level":3,"info":"0 MHz 0 V 50 RPM 50 RPM"}],"manual":false,"temp_targets":[70,80],"version":"v1.2","temp_target":75}

b-rowan, 2024-02-10:
Can you test with v0.52.0?  You should be able to get the miner and call `await miner.stop_mining()` and `await miner.resume_mining()` in that version.

mahines2290, 2024-02-10:
[image]
[image]
i am probably doing something wrong. i loaded new version .52.0

UpstreamData, 2024-02-10:
```python
import pyasic
import asyncio

IP = "192.168.1.20"

async def main():
    miner = await pyasic.get_miner(IP)
    print("Shut down:", await miner.stop_mining())

if __name__ == "__main__":
    asyncio.run(main())
```

mahines2290, 2024-02-10:
it works on the kdboxii! but not the kdbox pro which i am sure you know that. switching modes is likely first test to adding all the other ones. please give me a btc tip address for your help

b-rowan, 2024-02-10:
It should work on the pro, that's interesting.  Make sure you switch the IP at the top of the file to the IP of the Pro?

You can also add a `print(miner)` in there after, so you can make sure its getting the type properly.

```python
import pyasic
import asyncio

IP = "192.168.1.20"

async def main():
    miner = await pyasic.get_miner(IP)
    print(miner)
    print("Shut down:", await miner.stop_mining())

if __name__ == "__main__":
    asyncio.run(main())
```

mahines2290, 2024-02-10:
verified the correct ip and ran it a few times it comes back true however machine doesn't change. wonder if maybe it is because the KDbox pro only has 2 options where the kdboxii has 3 options (much saught after low power mode) not a huge deal as those are getting phased out anyways. it would be cool if we had all 3 options for the kdboxii though if possible. you have already done so much though. thank you so much. 
[image]

b-rowan, 2024-02-10:
They did something really dumb lol, I think I figured it out now.  Please try `0.52.1`.

mahines2290, 2024-02-10:
i am trying to keep up to you lol. i tried running the same above with nothing for either miner but i see a lot of the code is different. what should i try to run this time instead of this 

import pyasic
import asyncio

IP = "192.168.1.20"

async def main():
    miner = await pyasic.get_miner(IP)
    print(miner)
    print("Shut down:", await miner.stop_mining())

if __name__ == "__main__":
    asyncio.run(main())

b-rowan, 2024-02-10:
That's the correct code.  It's not changing the values on either miner?

mahines2290, 2024-02-10:
/Users/matthew.hines/PycharmProjects/pythonProject1/.venv/bin/python /Users/matthew.hines/PycharmProjects/pythonProject1/.venv/test.py 
Traceback (most recent call last):
  File "/Users/matthew.hines/PycharmProjects/pythonProject1/.venv/test.py", line 13, in <module>
    asyncio.run(main())
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/asyncio/runners.py", line 194, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/asyncio/base_events.py", line 664, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/Users/matthew.hines/PycharmProjects/pythonProject1/.venv/test.py", line 9, in main
    print("sleep:", await miner.level(1))
                          ^^^^^^^^^^^
AttributeError: 'GoldshellKDBoxPro' object has no attribute 'level'
[image]

Process finished with exit code 1

b-rowan, 2024-02-10:
Can you try this - 
```python
import pyasic
import asyncio

from pyasic.config import MiningModeConfig

IP = "192.168.1.20"

async def main():
    miner = await pyasic.get_miner(IP)
    settings = await miner.web.setting()
    mode = MiningModeConfig.normal()
    cfg = mode.as_goldshell()
    level = cfg["settings"]["level"]
    for idx, plan in settings["powerplans"]:
        print(plan)
        if plan["level"] == level:
            print("found:", plan)
            settings["select"] = idx
    print(settings)

if __name__ == "__main__":
    asyncio.run(main())
```

mahines2290, 2024-02-10:
/Users/matthew.hines/PycharmProjects/pythonProject1/.venv/bin/python /Users/matthew.hines/Library/Application Support/JetBrains/PyCharm2023.3/scratches/shutdown.py 
Traceback (most recent call last):
  File "/Users/matthew.hines/Library/Application Support/JetBrains/PyCharm2023.3/scratches/shutdown.py", line 24, in <module>
    asyncio.run(main())
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/asyncio/runners.py", line 194, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/asyncio/base_events.py", line 664, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/Users/matthew.hines/Library/Application Support/JetBrains/PyCharm2023.3/scratches/shutdown.py", line 15, in main
    for idx, plan in settings["powerplans"]:
        ^^^^^^^^^
ValueError: too many values to unpack (expected 2)

Process finished with exit code 1
[image]

[image]

b-rowan, 2024-02-10:
Ok, `0.52.2` because I did something dumb.  Try this script again on that version.

```python
import pyasic
import asyncio

IP = "192.168.1.20"

async def main():
    miner = await pyasic.get_miner(IP)
    print(miner)
    print("Shut down:", await miner.stop_mining())

if __name__ == "__main__":
    asyncio.run(main())
```

mahines2290, 2024-02-10:
[image]
[image]
those were the errors from the previous run

mahines2290, 2024-02-10:
Works on both!

closed 2024-02-11 by b-rowan