UpstreamData/pyasic #109: LuxOS S19 Compatibility

jiskiw, 2024-02-16:
Would be great to have LuxOS S19 Compatibility.

b-rowan, 2024-02-16:
Can you confirm the model name from calling `pyasic.miners.factory.miner_factory.get_miner_model_luxos(ip)`?

jiskiw, 2024-02-18:
LUXOS

Miner at 192.168.10.49: Antminer S19
Miner at 192.168.10.50: Antminer S19
Miner at 192.168.10.112: Antminer S19
Miner at 192.168.10.136: Antminer S19
Miner at 192.168.10.192: Antminer S19
Miner at 192.168.10.200: Antminer S19
Miner at 192.168.10.233: Antminer S19

VNish

Miner at 192.168.10.188: Antminer L7
Miner at 192.168.10.231: L3+ (Shown as None on the app)
Miner at 192.168.10.235: L3+ (Shown as None on the app)

b-rowan, 2024-02-18:
Perfect.  I don't have my computer on me right now, but I'll get this added on Tuesday.

jiskiw, 2024-02-18:
Aewsome thank you!

b-rowan, 2024-02-20:
Added in the latest commit, but before I release a version, can you confirm the vnish version on those L3+, and also send the result of `pyasic.miners.factory.miner_factory.get_miner_model_vnish(ip)` for those?  I want to confirm that the model result is actually correct for those ones, as technically pyasic already has L3+ vnish support.

closed 2024-04-07 by b-rowan