UpstreamData/pyasic #114: Dashboard for visualizing data

wilfredallyn, 2024-02-26:
**Is your feature request related to a problem? Please describe.**
It would be useful to have a dashboard for visualizing the miner data. This would make it easier to analyze miner performance and enable experimentation with different settings.

**Describe the solution you'd like**
An interactive python dashboard (e.g., [plotly dash](https://dash.plotly.com/)) would be customizable and let people dive into the data. I think a good first version would be to provide the basic functionality of the [Braiins OS+ dashboard](https://youtu.be/-RiUFCiVPEs?si=D6ttU9vM8B5t3V5I&t=131) showing overall status, hashrate, and temperature. Later, we could add more visualizations.

**Additional context**
I've started working on this and plan to split it into 2 PRs
1. save miner data to SQLite db
2. create plotly dash app with visualizations

b-rowan, 2024-02-26:
I like this idea a lot, however I'm not certain it falls fully into the realm of the library, but rather seems like something the user should implement.

For example - 

https://github.com/UpstreamData/pyasic_web

https://github.com/b-rowan/pyasic_to_influx

That being said, I don't mind having the conversion functions included here, such as formatting for SQLite directly in the miner data class, as long as it doesn't introduce additional dependencies.  Happy to look over PRs for this in any case, but just some general comments for you to consider.

wilfredallyn, 2024-02-26:
After thinking about it, I agree that it's better to keep the dashboard separate to avoid making this repo too complicated.

I'll make a separate repo, and if it makes sense in the future, we can merge it in to pyasic_web or somewhere else

closed 2024-02-26 by wilfredallyn

b-rowan, 2024-02-26:
> After thinking about it, I agree that it's better to keep the dashboard separate to avoid making this repo too complicated.
> 
> 
> 
> I'll make a separate repo, and if it makes sense in the future, we can merge it in to pyasic_web or somewhere else

Agreed.

Let me know if you need any help, happy to provide reference implementations if you need them 👍

wilfredallyn, 2024-02-29:
@b-rowan I created a repo and opened a PR for saving the data. Can you take a look and let me know if you have any suggestions for improving it? Thanks

https://github.com/wilfredallyn/pyasic_dashboard/pull/1

wilfredallyn, 2024-03-05:
@b-rowan, I created a dashboard with some basic figs from the BOS dashboard. I also created a demo site for it so it's easier to play around with it.
https://github.com/wilfredallyn/pyasic_tools/pull/2
https://pyasic.tools/

I added ideas for next steps to the issues. Which one do you think would be good to tackle next?
https://github.com/wilfredallyn/pyasic_tools/issues