UpstreamData/pyasic #116: Can't increase the power target BraiinsOS 24.02 Plus

kamordi, 2024-03-09:
Since the update to Braiins OS 24.02 I am not able to increase the power target. Decreasing is working fine. 

I am using 
```
async def set_power_target(miner_ip: str, power):

    miner = await get_miner(miner_ip)
    answer = await miner.web.set_power_target(int(power))
```
and get no Exception, but also no reaction on the machine.

So, when I am at 2000W and want to increase to 2760W it doesn't work.
When I am at 2760W and want to decrease to 2000W i get it to work.

Used machine: Antminer S19k pro NoPic
Used firmware: Braiins OS
Used OS: 2024-02-27-0-b218886f-24.02-plus
Used pyasic: 0.54.3

As far I can see the new 2024.02 already has included 
https://github.com/braiins/bos-plus-api/releases/tag/1.0.0-beta.6

closed 2024-03-09 by kamordi

kamordi, 2024-03-11:
For information, 
was able to use the api via comandline and received the same behavior -> no issue from pyasic

UpstreamData, 2024-03-11:
Makes sense.  Thanks for the clarification.  I wonder if pyasic was able to identify functionality even with the API not responding, do you know if it found a type, or just `Unknown`?