UpstreamData/pyasic #119: BOS Antminer S19 XP Miner is not returning data instead throwing error .

vijendtra, 2024-04-05:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
 Trying to get data from BraiinsOS firmware Antminer S19XP
 using this code `miner_data = await miner.get_data()`
 it return 
`Failed to fetch data for miner 10.XX.XX.XXX: LUCI web command failed: command=bos/info`

**Expected behavior**
It should return data without having any error. 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Macbook
 - Version: Sanoma 14.4.1

**Miner Information (If applicable):**
 - Manufacturer: Antminer
 - Type: S19 XP
 - Firmware Type:  BraiinsOS
 - Firmware Version: Version: 2023-12-15-0-f43dca21-23.12-plus

**Additional context**
Add any other context about the problem here.

UpstreamData, 2024-04-05:
Can you confirm the result of get_miner for that IP?  Want to make sure it's actually being identified.  In theory that miner shouldn't send a LUCI command, so it's likely failing the identification.

vijendtra, 2024-04-05:
This is what i am getting `Unknown (BOS+): 10.XX.XX.171`

UpstreamData, 2024-04-05:
Ok yep, failing to identify.  Let me make sure it's supported.

vijendtra, 2024-04-05:
it's look like it has  really old version of Braiins OS+,I suspect they don't have the new GRPC based API @UpstreamData

UpstreamData, 2024-04-05:
XP should be supported.  Can you send the output of `await pyasic.miner_factory.get_miner_model_braiins_os(IP)`?  It seems like the XP string might be a bit strange in this case, like something has been changed with it...

vijendtra, 2024-04-06:
This is what i am getting `Unknown (BOS+)` @UpstreamData

UpstreamData, 2024-04-06:
> This is what i am getting `Unknown (BOS+)` @UpstreamData 

That shouldn't be the case for that last function I sent, it should return a string with the model name and no suffix (`ANTMINER S19XP` is what it's supposed to be getting, but it's not or it would work properly).

UpstreamData, 2024-04-06:
`await pyasic.miner_factory.get_miner_model_braiins_os(IP)`

vijendtra, 2024-04-06:
Sorry my bad , you are correct it is returning `Antminer S19 XP`

vijendtra, 2024-04-06:
Howver when i do this with same IP 

```
miner_ip= await get_miner(ip)

print(f"Miner Ip: {miner_ip}")print(f"Miner IP: {ip}, Firmware: {getattr(miner, 'firmware')}, Model:{getattr(miner, 'model')}",
```

it return me 

`Miner IP: 10.XX.XX.171, Firmware: BOS+, Model:Unknown (BOS+) Make: None`

vijendtra, 2024-04-06:
@UpstreamData

UpstreamData, 2024-04-06:
Fixed in 0.54.10.

closed 2024-04-06 by UpstreamData