UpstreamData/pyasic #120: Does not support Braiins OS+ Prior to version 23.03

cryptographicturk, 2024-04-09:
When there are miners on the network that are running on versions prior to the release of Braiins OS+ 23.03, which introduced the GRPC API, pyasic can't connect to port 50051.

**To Reproduce**
Steps to reproduce the behavior:
1. run get_data on a miner running Braiins OS+ 22.02.4

**Expected behavior**
miner data in the returned dict

**Screenshots**
[Errno 111] Connect call failed ('<IP>', 50051)

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04 LTS

**Miner Information (If applicable):**
 Bitmain S19J Pro 104 Braiins OS+ 2022-04-29-0-5eb2c5fe-22.02.4-plus

b-rowan, 2024-04-09:
I'm aware of this, and I don't really plan to be backwards compatible for it, since you should really just update your X19s to the latest version.

That being said, this is the latest version for X17s, so all X17 variants use commands that would be fully comparable with this miner.  If you really wanted to hack something together, you could just check if miner version is <23.03 and replace it with the S17 base type, which setting the model and chip count information.

That being said, you should really just update the miner if possible.  Firmware needs to be kept up to date for security and bug fixes, and although I can't control how you run your stuff, I do highly recommend updating, as you also get more cool functionality by upgrading (such as setting wattage within a certain range of current wattage not requiring a retune).

b-rowan, 2024-04-09:
I'm not sure if there is a good fix here that revolves around better error handling, if that's the case I can implement that. I can likely just raise a user warning to update things.

b-rowan, 2024-04-09:
One more thing I suppose I can add.

I am not firmly opposed to being backwards compatible in this case, thinking about it there might be a way to in place replace the instance of the miner with one that would work on a GRPC error, but I'm not sure how elegant that would be.  I have to think about it more.

I understand there are cases where developers may have no control over the machines and end user is using, and what versions of things they have.  I am working on implementing a firmware update framework for this exact reason, as a dev if you can tell people their stuff is incompatible and out of date, and that they should update it, you should also provide the means to do so.

All that being said, if you have ideas here I am open to them, but I do not want to get stuck in backwards compatibility hell with this project.  If there is no simple and elegant solution, I would rather just support the most recent versions of stuff, since people should realistically always use the latest.  I get the possible standpoints this can be coming from though.

Please let me know if you have any suggestions here.

cryptographicturk, 2024-04-09:
I think the idea of not supporting older firmwares like this is reasonable.. the only reason I have miners with this firmware is for software testing (maintaining our own backwards compatibility). I'll have some conversations with a few miners and see if there is any reason to push this.

b-rowan, 2024-04-09:
Sounds good.  Let me know, like I said, I'm ok doing this as long as there is a clean and reasonable way to do it.  I get the situation of having miners you don't control.