UpstreamData/pyasic #123: MinerData's as_json() method raises error 'Object of type TunerAlgo is not JSON serializable'

martyfay1, 2024-04-12:
**Describe the bug**
Given some instances of MinerData, retrieved from a miner.get_data() call, calling as_json() on said instance will sometimes raise a TypeError with the message 'Object of type TunerAlgo is not JSON serializable'. The included mining_mode seems to be the cause. 

mining_mode=MiningModeHashrateTune(mode='hashrate_tuning', hashrate=88, algo=<TunerAlgo.voltage_optimizer: <class 'pyasic.config.mining.VOptAlgo'>>

**To Reproduce**
Steps to reproduce the behavior:
1. get_data() from a miner that has a TunerAlgo as the algo for mining_mode.
2. call as_json() on the returned data.

**Expected behavior**
The MinerData as_json() method should work when there is a TunerAlgo present.

**Desktop (please complete the following information):**
 - OS: macOS 13.6.6

b-rowan, 2024-04-12:
Ill see if I can get a fix out for this, good find.

b-rowan, 2024-04-12:
Fixed in `v0.54.16`.