UpstreamData/pyasic #129: I can't set power target on Braiins OS+ Miner

cryptographicturk, 2024-04-24:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
I've tried writing the code as:
```
                async with self.semaphore:
                    pyasic_miner = await get_miner(miner['ip'])
                    await pyasic_miner.set_power_limit(power_target)
```

and: 
```
                async with self.semaphore:
                    pyasic_miner = await get_miner(miner['ip'])
                    new_cfg = await pyasic_miner.get_config()
                    new_cfg.mining_mode.power = power_target
                    await pyasic_miner.send_config(new_cfg)
```
In both cases I get variations on 'NoneType' object has no attribute 'replace':
```
.venv/lib/python3.10/site-packages/pyasic/miners/backends/braiins_os.py:218 in set_power_limit
[WARNING][04/23/24 18:10:05](root) - Unknown (BOS+): xx.xx.xx.xx - Failed to set power limit: 'NoneType' object has no attribute 'replace'
```

**Expected behavior**
Power tuning target to change

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04, Python 3.10

**Miner Information (If applicable):**
 - Manufacturer: Bitmain AntMiner S19J Pro Plus Braiins OS+ Version: 2024-04-02-0-0de0f2f0-24.03.1-plus

b-rowan, 2024-04-24:
Seems like the miner isn't being identified properly, it's likely not supported.

Can you send over the result of the `devdetails` rpc command (`await miner.rpc.devdetails()`)?

b-rowan, 2024-04-24:
Seems like the SQoJ is supported with BOS+, but something isn't identifying properly in any case.

cryptographicturk, 2024-04-24:
`{'STATUS': [{'STATUS': 'S', 'When': 1713974038, 'Code': 69, 'Msg': 'Device Details', 'Description': 'BOSer boser-buildroot 0.1.0-0de0f2f0'}], 'DEVDETAILS': [{'Chips': 120, 'Cores': 61680, 'DEVDETAILS': 0, 'Device Path': '', 'Driver': '', 'Frequency': 650, 'ID': 1, 'Kernel': '', 'Model': 'Antminer S19J Pro Plus NoPic', 'Name': 'HashchainManager 1', 'Voltage': 13.191326530612244}, {'Chips': 120, 'Cores': 61680, 'DEVDETAILS': 1, 'Device Path': '', 'Driver': '', 'Frequency': 663, 'ID': 2, 'Kernel': '', 'Model': 'Antminer S19J Pro Plus NoPic', 'Name': 'HashchainManager 2', 'Voltage': 13.191326530612244}, {'Chips': 120, 'Cores': 61680, 'DEVDETAILS': 2, 'Device Path': '', 'Driver': '', 'Frequency': 647, 'ID': 3, 'Kernel': '', 'Model': 'Antminer S19J Pro Plus NoPic', 'Name': 'HashchainManager 3', 'Voltage': 13.191326530612244}], 'id': 1}`

UpstreamData, 2024-04-24:
Should be fixed in 0.54.19

cryptographicturk, 2024-04-24:
I'm still having the same issue in 0.54.19. I used pip to get the update.

UpstreamData, 2024-04-24:
What's the miner type youre getting (`print(await pyasic.get_miner(ip))`)?

UpstreamData, 2024-04-24:
It was named wrong, fixed in 0.54.20.

closed 2024-04-24 by UpstreamData