UpstreamData/pyasic #132: Can't seem to set pool configuration using send_config ({NameError}NameError("name 'self' is not defined") on send_config)

cryptographicturk, 2024-04-29:
**Describe the bug**
When I attempt to configure the pool for a miner using send_config I get name 'self' is not defined.

**To Reproduce**
Steps to reproduce the behavior:
```
import asyncio

from pyasic import get_miner, MinerData, APIError
from pyasic import settings

new_pools = [
    {
      "url": "stratum.url1.io:3333",
      "user": "worker2024.48d605acbec9",
      "password": "123"
    },
    {
      "url": "stratum.url2.io:3333",
      "user": "worker2024.48d605acbec9",
      "password": "123"
    },
    {
      "url": "stratum.url3.io:3333",
      "user": "worker2024.48d605acbec9",
      "password": "123"
    }
]

settings.update("factory_get_retries", 3)
settings.update("factory_get_timeout", 3)
settings.update("get_data_retries", 10)
settings.update("network_ping_retries", 2)
settings.update("network_ping_timeout", 5)

async def set_pools():
    try:
        miner = await get_miner("10.10.73.55")
        new_cfg = await miner.get_config()
        index = 0
        for pool in new_cfg.pools.groups[0].pools:
            pool.url = f"stratum+tcp://{new_pools[index]['url']}"
            pool.user = new_pools[index]['user']
            pool.password = new_pools[index]['password']
            index += 1

        await miner.send_config(new_cfg)
    except NameError as ne:
        print(f"Error: {type(ne).__name__} - {ne}")
    except Exception as e:
        print(f"Error: {type(e).__name__} - {e}")

asyncio.run(set_pools())
```

**Expected behavior**
Pool configuration changes

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04 LTS

**Miner Information (If applicable):**
 - Manufacturer: Bitmain
 - Type: S19 Pro, S19J Pro, S19J Pro Plus, S19 XP
 - Firmware Type: Stock, Braiins OS Plus
 - Firmware Version: multiple

b-rowan, 2024-04-29:
Not sure where this error is coming from, my guess is trying to re-assign to the pool fields is causing an issue.

Can you try with this function instead?

```from pyasic import get_miner, config

async def set_pools(new_pools: list[dict]):
    try:
        miner = await get_miner("10.10.73.55")
        new_cfg = await miner.get_config()
        pools = [
            config.pools.Pool(
                url=f"stratum+tcp://{new_pool['url']}",
                user=new_pool["user"],
                password=new_pool["password"],
            )
            for new_pool in new_pools
        ]
        new_cfg.pools.groups[0].pools = pools

        await miner.send_config(new_cfg)
    except NameError as ne:
        print(f"Error: {type(ne).__name__} - {ne}")
    except Exception as e:
        print(f"Error: {type(e).__name__} - {e}")
```

b-rowan, 2024-04-29:
Honestly the easiest thing might be to remove the error handling for now, and if the error occurs send the whole traceback, im curious where the error itself is getting raised.

cryptographicturk, 2024-04-29:
I only included the attempted error handling to see if we could get any insight as to why the pools don't change. I realize now that I left that out of the bug report as well. I've run this across a large cross section of equipment and the pools don't change. I detect the error in production by checking the config again after expecting a succesful change, and discovering that it isn't working.

b-rowan, 2024-04-29:
Yeah that name error is strange, not sure where that could be coming from, only place I could think of is a static method somewhere.  In any case if you can try it and see where that name error comes up with a full trace back, it will be a lot easier for me to fix.

cryptographicturk, 2024-04-30:
I can't even catch that exception, so I can't get a full traceback on it. I did a stack trace if you're curious, but nothing here. In the meantime, this name error might be a red herring related to using visual step through debugging on async python. The real issue is that I can't change pools.

`python3 ./walp.py
  File "/home/adrianglover/Source/helm/sandbox/async-scan/./walp.py", line 49, in <module>
    anyio.run(set_pools, backend='asyncio')
  File "/home/adrianglover/Source/helm/sandbox/async-scan/.venv/lib/python3.10/site-packages/anyio/_core/_eventloop.py", line 66, in run
    return async_backend.run(func, args, {}, backend_options)
  File "/home/adrianglover/Source/helm/sandbox/async-scan/.venv/lib/python3.10/site-packages/anyio/_backends/_asyncio.py", line 1968, in run
    return native_run(wrapper(), debug=debug)
  File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.10/asyncio/base_events.py", line 636, in run_until_complete
    self.run_forever()
  File "/usr/lib/python3.10/asyncio/base_events.py", line 603, in run_forever
    self._run_once()
  File "/usr/lib/python3.10/asyncio/base_events.py", line 1909, in _run_once
    handle._run()
  File "/usr/lib/python3.10/asyncio/events.py", line 80, in _run
    self._context.run(self._callback, *self._args)
  File "/home/adrianglover/Source/helm/sandbox/async-scan/.venv/lib/python3.10/site-packages/anyio/_backends/_asyncio.py", line 1961, in wrapper
    return await func(*args)
  File "/home/adrianglover/Source/helm/sandbox/async-scan/./walp.py", line 46, in set_pools
    traceback.print_stack()
`

b-rowan, 2024-04-30:
Ah.  I just remembered, I'm waiting on Braiins to fix their gRPC API so that setting pools isn't a 300 step process.  Right now you have to query the groups, get the IDs, delete all the groups, then create a group, add each of the pools 1 by 1, and repeat for each group.  Massively stupid, so I was waiting for them to just let me send a bunch of pool groups in one request so that the pools can be updated in one step.

b-rowan, 2024-04-30:
Reference issue - https://github.com/braiins/bos-plus-api/issues/16

b-rowan, 2024-07-30:
Fixed in 0.59.0.

closed 2024-07-30 by b-rowan