UpstreamData/pyasic #133: Add firmware vendor to MinerData

cryptographicturk, 2024-04-30:
We run a wide variety of equipment under different firmware's for different reasons. As such part of what we want to know about each miner is which specific firmware is running on it. I would like there to be a field added to the MinerData, such as fw_vendor, that we can use to differentiate what equipment we're looking at.

b-rowan, 2024-04-30:
When you say fw_vendor, you mean the `miner.firmware` attribute, such as here,  right?

https://github.com/UpstreamData/pyasic/blob/05a4ae6f04d6db8f6db72de1b756f53b78fe0a90/pyasic/miners/backends/braiins_os.py#L108

b-rowan, 2024-04-30:
As of right now, if you need this information, you can parse it out of the model by using a regex with a capture group for everything inside the brackets eg `S9 (BOS+)`.

b-rowan, 2024-04-30:
If doing this, it might be a good idea to make the `firmware` attribute of each miner a class or enum value, such that it can be string formatted.

cryptographicturk, 2024-04-30:
Thank you for the workaround.. I will try it. And yes, I like the design of a firmware attribute with the different strings within.

cryptographicturk, 2024-04-30:
This might be a bug.. currently that is not in the model string. This is what I'm getting for both BOS+ and bitmain firmware on my test miners:

e.g. 'S19j Pro No PIC', 'S19 Pro', etc

I only get the firmware when it's Unknown

b-rowan, 2024-04-30:
> This might be a bug.. currently that is not in the model string. This is what I'm getting for both BOS+ and bitmain firmware on my test miners:
> 
> e.g. 'S19j Pro No PIC', 'S19 Pro', etc
> 
> I only get the firmware when it's Unknown

Yeah bug, BOS didnt have the firmware string set.  I've fixed it now, just waiting to push out a version until we can test mara firmware stuff.

b-rowan, 2024-05-01:
> > This might be a bug.. currently that is not in the model string. This is what I'm getting for both BOS+ and bitmain firmware on my test miners:
> > e.g. 'S19j Pro No PIC', 'S19 Pro', etc
> > I only get the firmware when it's Unknown
> 
> Yeah bug, BOSer didnt have the firmware string set. I've fixed it now, just waiting to push out a version until we can test mara firmware stuff.

Fixed with 0.55.0.

closed 2024-05-15 by b-rowan