UpstreamData/pyasic #138: Add in functionality to limit the number of ips being pinged when scanning.

martyfay1, 2024-05-06:
When using `network.scan()` to scan a subnet, it is noticed that without modifying the code, `ping_and_get_miner` will eventually result in a `[Errno 24] Too many open files` on linux and result in some miner IPs being missed by the scan. I know I can change the ulimit, but it would also be nice to have the control of setting a cap for the number of `ping_and_get_miner` calls to be executed concurrently.

I thought this may be the reason for the `network_scan_threads` setting but I don't see that used anywhere while examining the code.

**Additional context**
This was a problem I encountered when scanning on Ubuntu.

b-rowan, 2024-05-06:
Yeah, that setting was sort of removed when I refactored the way the network works.  Shouldn't be too hard to add back.

UpstreamData, 2024-05-06:
Added back in with 0.55.3.  By default the value is None, which will not bound the scan, but now you can set a value with `pyasic.settings.update("network_scan_semaphore", 1000)`

closed 2024-05-06 by UpstreamData