UpstreamData/pyasic #140: Doesnt get Miner Modell BraiinsOS 24.04 Plus

kamordi, 2024-05-10:
**Describe the bug**
Since Upgrade BraiinsOS to 24.04 Plus, pyasic doenst receive the correct modell of the miner -> all other functions also doesnt work properly

**To Reproduce**
Calling 
`await miner.web.set_power_target(int(power))` 
I receive: 
`AttributeError: 'BOSMinerWebAPI' object has no attribute 'set_power_target'`
After some testing I looked for
`await miner.get_data()`
And received:
`MinerData(ip='MinerIP', datetime=datetime.datetime(2024, 5, 10, 9, 15, 38, 140165, tzinfo=datetime.timezone(datetime.timedelta(seconds=7200), 'CEST')), uptime=2150, mac='MinerMAC', 
model='Unknown (BOS+)', 
make=None, 
api_ver='3.7', 
fw_ver='24.04', 
hostname='Antminer', ...
`

**Desktop (please complete the following information):**
 - OS: raspianOS
 - Version [e.g. 22]
 - pyasic 0.55.3

**Miner Information (If applicable):**
 - Manufacturer: Antminer
 - Type: S19k
 - Firmware Type: BraiinsOS
 - Firmware Version: 24.04 Plus

I would asume pyasic doenst recognise the miner modell correct so it fails to use the prefered function.
I will downgrade back to 24.03.01 due to my need for automatic power control. When it is necessary for testing, I would upgrade again.

b-rowan, 2024-05-10:
Yeah seems like a model identification issue.  Can you try calling `pyasic.miner_factory. get_miner_model_braiins_os(ip)` and sending the result of that?  That should help me make sure I have the right model string.

b-rowan, 2024-05-10:
Additionally, you should use `miner.set_power_limit()` instead of calling the web commands directly if possible, https://docs.pyasic.org/en/latest/miners/functions/#pyasic.miners.base.MinerProtocol.set_power_limit

kamordi, 2024-05-10:
Thanks for the advice, I will try to use it as designed. 

As result I get 
`Antminer S19k Pro`
Not as in my origin attempt in 
[https://github.com/UpstreamData/pyasic/issues/78](url)

b-rowan, 2024-05-10:
Should be fixed with 0.55.4.  I really wish they would just pick a naming scheme and stick to it, but whatever.

closed 2024-05-10 by b-rowan