UpstreamData/pyasic #148: Unable to modify power limits from inside home assistant on S19K pro miner

CataDRC, 2024-05-22:
Unable to modify power limits from inside HA on S19K pro. Getting a error. Attached screenshot.

Core 2024.5.2
Supervisor 2024.05.1
Operating System 12.3 
Frontend 20240501.1

Antminer S19K pro running Braiins OS Version: 2024-05-06-0-fb624381-24.04-plus

thank you

[image]

b-rowan, 2024-05-22:
Can you send the result of this script on windows? If you don't have a windows machine I can translate this to bash, but most people do.

To use this, make a new text file, paste this data in, and change the "server" variable at the top to match the IP of your miner. Once that's done, change the file extension of the file to .ps1 (if you cant see this, its in file explore top menu, view, then a checkbox that says file name extensions, then you can rename the file). Right click, select "Run with powershell", then send the result in a text file here.

`'`
$server = "192.168.1.5"
$port = 4028
$commands = @("devs", "stats", "devdetails", "version")

try {
    foreach ($cmd in $commands) {
        $client = New-Object System.Net.Sockets.TcpClient
        $client.Connect($server, $port)

        $stream = $client.GetStream()

        $command = '{"command": "' + $cmd + '"}'
        $writer = [System.Text.Encoding]::UTF8.GetBytes($command)

        $stream.Write($writer, 0, $writer.Length)
        $stream.Flush()

        # Read response
        $reader = New-Object System.IO.StreamReader($stream)
        $response = $reader.ReadToEnd()
        $reader.Close()

        # Display the response
        Write-Host "Response for command '$cmd':"
        Write-Host "-------------------------------------------------------------------------------------"
        Write-Host $response
        Write-Host ""

        # Clear the stream for the next command
        $stream.Flush()
        # Close the connection
        $client.Close()
    }

    # Wait for a key press
    Read-Host "Press Enter to exit..."
}
catch {
    Write-Host "An error occurred: $_"
    Read-Host "Press Enter to exit..."
}
```

CataDRC, 2024-05-22:
[New Text Document.txt](https://github.com/UpstreamData/pyasic/files/15405738/New.Text.Document.txt)

thank you

b-rowan, 2024-05-22:
The main problem is here - `{"STATUS":[{"STATUS":"E","When":1716392576,"Code":303,"Msg":"Unavailable","Description":"BOSer boser-buildroot 0.1.0-fb624381"}],"id":1}
`

I hate that they keep doing this, this is like the only place to source model information on most models, these commands should ALWAYS return.  Ill make a complaint to the devs.

b-rowan, 2024-05-22:
I might be able to source it from a LUCI endpoint, let me look into it.

CataDRC, 2024-07-26:
Hi b-rowan, any updates on this issue? Thanks

928driver, 2024-07-30:
@b-rowan thank you for all the great work you are doing! I have been using HASS-Miner on a S19 for quite many months and it is working perfectly. I recently bought a S19 Pro+ Hyd and I am experiencing the same issue as @CataDRC , I cannot change the power limit. Is it possibly related to the same issue? What information can I provide to try helping out to resolve the issue?

928driver, 2024-07-30:
Here is the results from running the script: 
[Antminer S19 Pro+ Hydro BOS+.txt](https://github.com/user-attachments/files/16432492/Antminer.S19.Pro%2B.Hydro.BOS%2B.txt)

I realized that I have a different type of miner than what this thread was about, therefore I created a new feature request for my miner. To avoid double posting, do you want to handle both miner types in this thread or separately?

b-rowan, 2024-07-30:
> Here is the results from running the script: [Antminer S19 Pro+ Hydro BOS+.txt](https://github.com/user-attachments/files/16432492/Antminer.S19.Pro%2B.Hydro.BOS%2B.txt)
>
> I realized that I have a [...]

I can do both here, this seems to be ongoing for multiple unsupported miner types.  Ill just get it fixed rq.

b-rowan, 2024-07-30:
Fixed in 0.59.0.

closed 2024-07-30 by b-rowan