UpstreamData/pyasic #165: Bitaxe ASIC Miner Support (esp-miner)

pixeldoc2000, 2024-06-28:
Please add (basic) Support for Bitaxe Miner (esp-miner firmware).

The bitaxe is a fully open source ASIC bitcoin miner.

Project: https://www.bitaxe.org/
Firmware: https://github.com/skot/ESP-Miner/

Firmware API:  https://github.com/skot/ESP-Miner?tab=readme-ov-file#api

b-rowan, 2024-06-28:
This is a great idea, I wasn't even aware that the bitaxe had much in the way of API, but its nice to see some docs for it.  Not sure if you have one, but if you do, would you mind hitting those web endpoints in a browser and sending some results here?

jpcomps, 2024-06-28:
This was on my wish list to find time to add as well, glad someone pushed a issue for this. Going to tag @skot here if I can in case we need some additional help, but sure we can find many people willing to dump anything we need. I am looking to have one ordered as well so can also help once we put the framework in

b-rowan, 2024-06-28:
> This was on my wish list to find time to add as well, glad someone pushed a issue for this. Going to tag @skot here if I can in case we need some additional help, but sure we can find many people [...]

NGL, im in the middle of doing this right now.  My biggest issue is always that I dont have one haha.  Ill push up something to `dev_bitaxe` shortly once I finish the factory implementation.

b-rowan, 2024-06-28:
Done.  I've done very basic support for it one the `dev_bitaxe` branch, now it just needs testing.

skot, 2024-06-28:
Dang you guys are quick! I'll test this out asap.

b-rowan, 2024-06-28:
> Dang you guys are quick! I'll test this out asap.

Having source code makes things much easier.  Thanks for all your contributions to open source in the space!

jpcomps, 2024-06-28:
{
    "power":    10.869999885559082,
    "voltage":    5042.5,
    "current":    2156.25,
    "fanSpeedRpm":    3827,
    "temp":    55,
    "hashRate":    435.77643357785956,
    "bestDiff":    "274M",
    "bestSessionDiff":    "1.71M",
    "freeHeap":    164292,
    "coreVoltage":    1200,
    "coreVoltageActual":    1234,
    "frequency":    485,
    "ssid":    "KH",
    "hostname":    "espressif",
    "wifiStatus":    "Connected!",
    "sharesAccepted":    63,
    "sharesRejected":    1,
    "uptimeSeconds":    3033,
    "ASICModel":    "BM1366",
    "stratumURL":    "public-pool.io",
    "stratumPort":    21496,
    "stratumUser":    "<btcAddr>.office",
    "version":    "v2.1.8",
    "boardVersion":    "0.11",
    "runningPartition":    "ota_1",
    "flipscreen":    1,
    "invertscreen":    0,
    "invertfanpolarity":    1,
    "autofanspeed":    1,
    "fanspeed":    100
}

api/system/info^^

Thanks to gary_mara in discord!

pixeldoc2000, 2024-06-30:
First test with `dev_bitaxe` branche

Source: https://github.com/UpstreamData/pyasic#data-gathering

```
MinerData(ip='192.168.0.202', datetime='2024-06-30T22:13:36.714119+02:00', timestamp=1719778416, device_info=DeviceInfo(make=None, model=None, firmware=None, algo=SHA256Algo), make=None, model=None, firmware=None, algo='SHA256', mac=None, api_ver='v2.1.7', fw_ver='v2.1.7', hostname='xxx', hashrate=0.595494584964798 TH/s, expected_hashrate=None, expected_hashboards=3, expected_chips=0, expected_fans=2, percent_expected_chips=None, percent_expected_hashrate=None, percent_expected_wattage=None, temperature_avg=None, env_temp=None, wattage=19, wattage_limit=None, voltage=None, fans=[], fan_psu=None, hashboards=[HashBoard(slot=0, hashrate=0.595494584964798 TH/s, temp=None, chip_temp=51, chips=None, expected_chips=None, serial_number=None, missing=False, tuned=None, active=True, voltage=5027.5)], total_chips=None, nominal=None, config=MinerConfig(pools=PoolConfig(groups=[PoolGroup(pools=[Pool(url='stratum+tcp://xxx:3333', user='BTC:bc1qxxx', password='x')], quota=1, name='N2B4L3')]), fan_mode=FanModeNormal(mode='normal', minimum_fans=1, minimum_speed=0), temperature=TemperatureConfig(target=None, hot=None, danger=None), mining_mode=MiningModeNormal(mode='normal')), fault_light=None, errors=[], is_mining=True, uptime=3537, efficiency=32, pools=[])
0.595494584964798 TH/s
```
Not all MinerData is populated, but query some data with `miner_data` and `miner_data.hashrate` is working.

b-rowan, 2024-06-30:
For reference, the model is parsed in `MinerFactory` via the `MINER_CLASSES` dict, the keys are miner type, then model, here is the bitaxe models - https://github.com/UpstreamData/pyasic/blob/ee1eece181798e9ee47e6f23495532b9b6a4eae5/pyasic/miners/factory.py#L444-L449

That uses the result of this function, converted into uppercase - https://github.com/UpstreamData/pyasic/blob/ee1eece181798e9ee47e6f23495532b9b6a4eae5/pyasic/miners/factory.py#L1023-L1033

This means I expect `web_system_info["devicemodel'].upper()` to be one of the values above, which would parse it directly into one of those miner types.  On the bright side, it seems to know that it is a `BitAxe` model of some kind, so the initial layer of discovery is working, but it seems this one is failing.  The problem in this case seems to be that your device contains no `devicemodel` key, which from what I can see has to be set in the inital configuration.  Ill make changes to have it use the asicmodel instead, and that should fix it, although I could actually have it check for either or if needed.

Also, regarding the restart function, the function is not called restart, there are 2 different functions for "restarting" in pyasic, `restart_backend`, which usually is used to retsart the mining process, such as cgminer, and not the whole device, and `reboot`, which is used to restart the whole device (https://docs.pyasic.org/en/latest/#miner-control).

b-rowan, 2024-06-30:
Should be fixed on master, just pushed a commit that uses asic model instead, see https://github.com/UpstreamData/pyasic/commit/7a9c9237a3688ad3b9bd4c2a3ec3a1acba38691c

pixeldoc2000, 2024-07-01:
> This means I expect `web_system_info["devicemodel'].upper()` to be one of the values above, which would parse it directly into one of those miner types. On the bright side, it seems to know that it [...]

That could be my bad, because I have a Lucky Miner LV06, which seem to be a chinese "reference" Bitaxe Ultra 202 Design with BM1366? Thats why i did not configure a devicemodel when flashing factory Bitaxe (esp-miner) firmware. 

https://github.com/skot/ESP-Miner/blob/master/config-401.cvs does suggest to configure "devicemodel", but because it is user configureable, it might not be reliable, or contain other devicesmodel like "luckyminer".

> Also, regarding the restart function, the function is not called restart, there are 2 different functions for "restarting" in pyasic, `restart_backend`, which usually is used to retsart the mining [...]

My bad again, it was getting late.... `miner.reboot()` does work 👍 
AFAIK something like `restart_backend` does not exist in esp-miner firmware / axeos API.

b-rowan, 2024-07-01:
> My bad again, it was getting late.... miner.reboot() does work 👍
AFAIK something like restart_backend does not exist in esp-miner firmware / axeos API.

They are both set to do the same thing, they should both do a reboot.

pixeldoc2000, 2024-07-01:
Thats what MinerData looks like with current master:

```
MinerData(ip='192.168.0.202', datetime='2024-07-01T23:28:18.629639+02:00', timestamp=1719869298, device_info=DeviceInfo(make=<MinerMake.BITAXE: 'BitAxe'>, model=<BitAxeModels.BM1366: 'Ultra'>, firmware=None, algo=SHA256Algo), make='BitAxe', model='Ultra', firmware=None, algo='SHA256', mac=None, api_ver='v2.1.7', fw_ver='v2.1.7', hostname='xxx', hashrate=0.6828209940622305 TH/s, expected_hashrate=None, expected_hashboards=1, expected_chips=1, expected_fans=1, percent_expected_chips=None, percent_expected_hashrate=None, percent_expected_wattage=None, temperature_avg=None, env_temp=None, wattage=19, wattage_limit=None, voltage=None, fans=[], fan_psu=None, hashboards=[HashBoard(slot=0, hashrate=0.6828209940622305 TH/s, temp=None, chip_temp=49, chips=None, expected_chips=1, serial_number=None, missing=False, tuned=None, active=True, voltage=5040)], total_chips=None, nominal=None, config=MinerConfig(pools=PoolConfig(groups=[PoolGroup(pools=[Pool(url='stratum+tcp://xxxx', user='BTC:bc1qxxxx', password='x')], quota=1, name='XK8G9A')]), fan_mode=FanModeNormal(mode='normal', minimum_fans=1, minimum_speed=0), temperature=TemperatureConfig(target=None, hot=None, danger=None), mining_mode=MiningModeNormal(mode='normal')), fault_light=None, errors=[], is_mining=True, uptime=15, efficiency=28, pools=[])
0.6828209940622305 TH/s
```
 `expected_hashboards=1, expected_chips=1, expected_fans=1`is correct now.

pixeldoc2000, 2024-07-01:
`miner.restart_backend()` does nothing.  Searching the pyasic source, it does not look like it is definded for bitaxe either.

I am fine with with only `miner.reboot()` implemented, since thats what is supported by the axeos api.

b-rowan, 2024-07-01:
I suppose that makes sense.

Regarding data, I see it's still not filling out chip count, I'll try to fix that.

pixeldoc2000, 2024-07-01:
Maybe returning some sane `temperature=TemperatureConfig(target=None, hot=None, danger=None)` would be a good idea too.

I suppose tartget=80, hot=85, danger=90 . Thats somewhat related to suggested Temps from Antminer S9 Specs.
@skot any opinions on temps?

b-rowan, 2024-07-01:
I use none for those, as it is assumed that the device should set sensible defaults for itself if it doesn't receive a value for them.  EG BraiinsOS+, which will use those values for S9s, but may use different values for X19s (as they are more temperature sensitive).

Sort of tangential, but I would love to find some time to chat with you @skot about configuration and stuff, love the stuff that you're doing, and I think I could contribute knowledge on the API/control side, although my C skills are quite limited.

b-rowan, 2024-07-01:
> I suppose that makes sense.
> 
> Regarding data, I see it's still not filling out chip count, I'll try to fix that.

Fixed chip count, using a default of 1 now.

pixeldoc2000, 2024-07-02:
MinerData with current master:
```
MinerData(ip='192.168.0.202', datetime='2024-07-02T08:36:21.741810+02:00', timestamp=1719902181, device_info=DeviceInfo(make=<MinerMake.BITAXE: 'BitAxe'>, model=<BitAxeModels.BM1366: 'Ultra'>, firmware=<MinerFirmware.STOCK: 'Stock'>, algo=SHA256Algo), make='BitAxe', model='Ultra', firmware='Stock', algo='SHA256', mac=None, api_ver='v2.1.7', fw_ver='v2.1.7', hostname='xxx', hashrate=0.6664160853237453 TH/s, expected_hashrate=None, expected_hashboards=1, expected_chips=1, expected_fans=1, percent_expected_chips=100, percent_expected_hashrate=None, percent_expected_wattage=None, temperature_avg=None, env_temp=None, wattage=19, wattage_limit=None, voltage=None, fans=[], fan_psu=None, hashboards=[HashBoard(slot=0, hashrate=0.6664160853237453 TH/s, temp=None, chip_temp=49, chips=1, expected_chips=1, serial_number=None, missing=False, tuned=None, active=True, voltage=5047.5)], total_chips=1, nominal=True, config=MinerConfig(pools=PoolConfig(groups=[PoolGroup(pools=[Pool(url='stratum+tcp://xxx', user='BTC:bc1q5xxx', password='x')], quota=1, name='7A4RRM')]), fan_mode=FanModeNormal(mode='normal', minimum_fans=1, minimum_speed=0), temperature=TemperatureConfig(target=None, hot=None, danger=None), mining_mode=MiningModeNormal(mode='normal')), fault_light=None, errors=[], is_mining=True, uptime=146, efficiency=29, pools=[])
0.6664160853237453 TH/s
```

Please don't forget to add Bitaxe to supported miners docs: https://github.com/UpstreamData/pyasic/blob/master/docs/miners/supported_types.md

b-rowan, 2024-07-02:
Done!  Should be all good to go for basic support as of v0.58.0, I'm sure we can add more in the future pending more features in ESPMiner.

closed 2024-07-02 by b-rowan

pixeldoc2000, 2024-07-04:
@b-rowan 
Awesome, thank you very much.